luxonis/depthai-ros #426: [BUG] {Title of the bug}

erikg2123, 2023-10-05:
**Describe the bug**
I keep getting a 854x534 resolution for the rgb camera when running any of the launch files.
```
egonzalez@c1:~/Desktop//dai_ws$ ros2 launch depthai_ros_driver rgbd_pcl.launch.py
[INFO] [launch]: All log files can be found below /home/egonzalez/.ros/log/2023-10-05-13-35-08-564200-c17573-28097
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [28112]
[component_container-1] [INFO] [1696530909.076166978] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[component_container-1] [INFO] [1696530909.083813656] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1696530909.083884840] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1696530909.093979444] [oak_state_publisher]: got segment oak
[component_container-1] [INFO] [1696530909.094009062] [oak_state_publisher]: got segment oak-d-base-frame
[component_container-1] [INFO] [1696530909.094016578] [oak_state_publisher]: got segment oak_imu_frame
[component_container-1] [INFO] [1696530909.094021975] [oak_state_publisher]: got segment oak_left_camera_frame
[component_container-1] [INFO] [1696530909.094027598] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[component_container-1] [INFO] [1696530909.094032847] [oak_state_publisher]: got segment oak_model_origin
[component_container-1] [INFO] [1696530909.094037960] [oak_state_publisher]: got segment oak_rgb_camera_frame
[component_container-1] [INFO] [1696530909.094043282] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[component_container-1] [INFO] [1696530909.094048348] [oak_state_publisher]: got segment oak_right_camera_frame
[component_container-1] [INFO] [1696530909.094053332] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-1] [INFO] [1696530909.095611368] [oak_container]: Load Library: /home/egonzalez/Desktop/Repos/amr-perception/dai_ws/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak_state_publisher' in container 'oak_container'
[component_container-1] [INFO] [1696530909.148612049] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1696530909.148650689] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1696530909.154471535] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [INFO] [1696530911.353125087] [oak]: Camera with MXID: 184430102155860E00 and Name: 1.6 connected!
[component_container-1] [INFO] [1696530911.353856943] [oak]: USB SPEED: SUPER
[component_container-1] [INFO] [1696530911.374970905] [oak]: Device type: OAK-D-PRO-W-97
[component_container-1] [INFO] [1696530911.377170734] [oak]: Pipeline type: RGBD
[component_container-1] [INFO] [1696530911.609956062] [oak]: Finished setting up pipeline.
[component_container-1] [INFO] [1696530911.962042269] [oak]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
[component_container-1] [INFO] [1696530911.963082592] [oak_container]: Load Library: /opt/ros/humble/lib/libdepth_image_proc.so
[component_container-1] [INFO] [1696530911.976285891] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-1] [INFO] [1696530911.976385658] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-1] [INFO] [1696530911.976425013] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-1] [INFO] [1696530911.976458558] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-1] [INFO] [1696530911.976492618] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzRadialNode>
[component_container-1] [INFO] [1696530911.976525260] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[component_container-1] [INFO] [1696530911.976556757] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziRadialNode>
[component_container-1] [INFO] [1696530911.976589404] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[component_container-1] [INFO] [1696530911.976624586] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyzrgb_node' in container 'oak_container'
[component_container-1] [ERROR] [1696530912.131242593] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
```

This is my rgbd.yaml file
```
/oak:
  ros__parameters:
    camera:
      i_nn_type: none
    stereo:
      i_subpixel: true
    rgb:
      i_resolution: 800p
```

```
egonzalez@c1:~/Desktop/dai_ws$ ros2 topic echo /oak/rgb/image_raw --no-arr
header:
  stamp:
    sec: 1696528452
    nanosec: 132337582
  frame_id: oak_rgb_camera_optical_frame
height: 534
width: 854
encoding: bgr8
is_bigendian: 0
step: 2562
data: '<sequence type: uint8, length: 1368108>'
---
```

I am using Ubuntu 22.04 and ROS2 Humble.

Serafadam, 2023-10-06:
Hi, `rgb` node has a paramter `i_set_isp_scale` which resizes the output image, this is used for better alignment of depth in cameras with different sensors. If you set it to false, proper resolution will be output. More information [here](https://github.com/luxonis/depthai-ros#setting-rgb-parameters)

erikg2123, 2023-10-06:
@Serafadam That seemed to work, I am getting the 1280x800 resolution, but I am still unable to align the depth resolution with the RGB resolution. Before setting `i_set_isp_scale` to false, I was getting the following error:

`[component_container-1] [ERROR] [1696530912.131242593] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)`

Now I am seeing This error:
`[component_container-1] [ERROR] [1696603932.537263935] [point_cloud_xyzrgb_node]: Depth resolution (1920x1080) does not match RGB resolution (1920x1080)`

How do I get the the RGB image to align with depth?

Serafadam, 2023-10-09:
Hi, could you provide outputs from the image topics?
- `ros2 topic echo /oak/rgb/image_raw`
- `ros2 topic echo /oak/rgb/camera_info`
- `ros2 topic echo /oak/stereo/image_raw`
- `ros2 topic echo /oak/stereo/camera_info`