luxonis/depthai-ros #427: [BUG] RGBD-PCL examples freezes after some time

dk-teknologisk-lag, 2023-10-09:
Something similar is mentioned in:
https://github.com/luxonis/depthai-ros/issues/104 where the solution was to increase queue size, but I don't think extending it further would help.

**Describe the bug**
Running the ros2 launch depthai_ros_driver rgbd_pcl.launch.py use_rviz:=True in the iron docker launches the rviz and shows a pointcloud fine for some time. But eventually it freezes and doesn't update.

The rgb and depth topic keeps publishing, but the points topic stops publishing.

Its possible to provoke the error by more or less blocking the sensors view as seen in the video. After some time, the depth image gets behind in timestamps and eventually there is too large a gap for the [approximate_time synchronizer](https://github.com/ros2/message_filters/blob/rolling/include/message_filters/sync_policies/approximate_time.h) to synchronize the streams appropriately.
I have tried replacing the time synchronization in [pointcloudxyzrgb nodelet](https://github.com/ros-perception/image_pipeline/blob/noetic/depth_image_proc/src/nodelets/point_cloud_xyzrgb.cpp) with the approximate_epsilon_time and setting it to something like 4.5ms or up to 10ms, makes it work for a longer period of time, but doesn't feel like the best fix.

**Minimal Reproducible Example**
ros2 launch depthai_ros_driver rgbd_pcl.launch.py use_rviz:=True in iron docker

**Expected behavior**
A colored pointcloud shown all the time - even if the "view" is "difficult".

**Screenshots**
[Link to video file](https://dtioffice365-my.sharepoint.com/:v:/g/personal/lag_teknologisk_dk/EV548ZYYH6pAr1JUFZ6PTNMB6jPMbL8xLS3oyeSldkca1A?nav=eyJyZWZlcnJhbEluZm8iOnsicmVmZXJyYWxBcHAiOiJPbmVEcml2ZUZvckJ1c2luZXNzIiwicmVmZXJyYWxBcHBQbGF0Zm9ybSI6IldlYiIsInJlZmVycmFsTW9kZSI6InZpZXciLCJyZWZlcnJhbFZpZXciOiJNeUZpbGVzTGlua0RpcmVjdCJ9fQ&e=k4WwkV)

**Attach system log**
 - Which OS/OS version are you using?
 - Host: Ubuntu 23.04 runinng docker 
 - Which ROS version are you using?
 - Ros2
 - Which ROS distribution are you using ?
 - Rolling and Iron

Serafadam, 2023-10-12:
Hi @dk-teknologisk-lag, I assume this is the same issue as in https://github.com/luxonis/depthai-ros/issues/429, could you try testing the fix branches there to see if that also fixes issue for you? 
https://github.com/luxonis/depthai-ros/tree/frequency_fix_iron
https://github.com/luxonis/depthai-ros/tree/frequency_fix_humble

dk-teknologisk-lag, 2023-10-13:
Will test it asap.

dk-teknologisk-lag, 2023-10-23:
I have tested the latest docker images and it seems to work better, in the sense that it doesn't fall behind and freezes. It however, still runs with low fps, about 4-5. I have seen #409, so I'll try find some other cables to test with. Those I have is relatively long, but the device is seen as a SUPER usb device.

The colored point cloud is also published at unstable frame rate, where there are several hickups. So its not steady 4-5 fps, but jacked at times.

dk-teknologisk-lag, 2023-11-06:
I have noticed, that if I subscribe to the colored point cloud topic from another docker instance and it actually do some network traffic, the FPS goes back to about 20 Hz.
Seems some congestion on the network, makes it choke.. Guess its more a ROS / middelware issue, so I'll close this one as it seems fixed.
I couldn't reproduce it with a recently iron-image at least.

closed 2023-11-06 by dk-teknologisk-lag