luxonis/depthai-ros #429: [BUG] Camera frequency not respected since 2.8.1

BriceRenaudeau, 2023-10-09:
Branch: ROS Iron
Camera: OAK-D-Pro W

**Describe the bug**
Since release 2.8.1 the camera topic frequency has been wrong and unstable (should be 15Hz):
[image]

**Expected behavior**
It was working great on 2.8.0:
[image]

Serafadam, 2023-10-10:
Hi, thanks for the report few minor questions:
-  Could you post some additional information according to the [BUG report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)? 
At the moment I can't pinpoint the exact change that could influence publishing frequency, so I would also like to check if that could be coming from some other source. 
- Are those frequency plots generated in rqt?
- Is it possible for you to check if that also happens in Humble version?

BriceRenaudeau, 2023-10-11:
We are using:
- Ubuntu 22.04
- Depth-ai from sources
- ros-iron-depthai = 2.22.0-1jammy.20230622.191547 amd64 [installed]
- The plots are made with `plotjuggler`

**..._pipeline.json**:
```json
{"connections":[{"node1Id":7,"node1Output":"out","node1OutputGroup":"","node2Id":8,"node2Input":"in","node2InputGroup":""},{"node1Id":3,"node1Output":"out","node1OutputGroup":"","node2Id":5,"node2Input":"right","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":5,"node2Input":"left","node2InputGroup":""},{"node1Id":5,"node1Output":"depth","node1OutputGroup":"","node2Id":6,"node2Input":"in","node2InputGroup":""},{"node1Id":4,"node1Output":"out","node1OutputGroup":"","node2Id":3,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":2,"node1Output":"out","node1OutputGroup":"","node2Id":0,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":1,"node2Input":"in","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"sippBufferSize":18432,"sippDmaBufferSize":16384,"xlinkChunkSize":-1},"nodes":[[0,{"id":0,"ioInfo":[[["","out"],{"blocking":false,"group":"","id":30,"name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","id":29,"name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","id":28,"name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","id":27,"name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"MonoCamera","properties":{"boardSocket":1,"cameraName":"","fps":15.0,"imageOrientation":0,"initialControl":{"aeLockMode":false,"aeRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"afRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"antiBandingMode":0,"autoFocusMode":3,"awbLockMode":false,"awbMode":0,"brightness":0,"chromaDenoise":0,"cmdMask":0,"contrast":0,"effectMode":0,"expCompensation":0,"expManual":{"exposureTimeUs":0,"frameDurationUs":0,"sensitivityIso":0},"frameSyncMode":0,"lensPosAutoInfinity":0,"lensPosAutoMacro":0,"lensPosition":0,"lowPowerNumFramesBurst":0,"lowPowerNumFramesDiscard":0,"lumaDenoise":0,"saturation":0,"sceneMode":0,"sharpness":0,"strobeConfig":{"activeLevel":0,"enable":0,"gpioNumber":0},"strobeTimings":{"durationUs":0,"exposureBeginOffsetUs":0,"exposureEndOffsetUs":0},"wbColorTemp":0},"isp3aFps":0,"numFramesPool":3,"numFramesPoolRaw":3,"rawPacked":null,"resolution":1}}],[1,{"id":1,"ioInfo":[[["","in"],{"blocking":true,"group":"","id":26,"name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":{"maxFpsLimit":-1.0,"metadataOnly":false,"streamName":"left_mono"}}],[2,{"id":2,"ioInfo":[[["","out"],{"blocking":false,"group":"","id":25,"name":"out","queueSize":8,"type":0,"waitForMessage":false}]],"name":"XLinkIn","properties":{"maxDataSize":5242880,"numFrames":8,"streamName":"left_control"}}],[3,{"id":3,"ioInfo":[[["","out"],{"blocking":false,"group":"","id":24,"name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","id":23,"name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","id":22,"name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","id":21,"name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"MonoCamera","properties":{"boardSocket":2,"cameraName":"","fps":15.0,"imageOrientation":0,"initialControl":{"aeLockMode":false,"aeRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"afRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"antiBandingMode":0,"autoFocusMode":3,"awbLockMode":false,"awbMode":0,"brightness":0,"chromaDenoise":0,"cmdMask":0,"contrast":0,"effectMode":0,"expCompensation":0,"expManual":{"exposureTimeUs":0,"frameDurationUs":0,"sensitivityIso":0},"frameSyncMode":0,"lensPosAutoInfinity":0,"lensPosAutoMacro":0,"lensPosition":0,"lowPowerNumFramesBurst":0,"lowPowerNumFramesDiscard":0,"lumaDenoise":0,"saturation":0,"sceneMode":0,"sharpness":0,"strobeConfig":{"activeLevel":0,"enable":0,"gpioNumber":0},"strobeTimings":{"durationUs":0,"exposureBeginOffsetUs":0,"exposureEndOffsetUs":0},"wbColorTemp":0},"isp3aFps":0,"numFramesPool":3,"numFramesPoolRaw":3,"rawPacked":null,"resolution":1}}],[4,{"id":4,"ioInfo":[[["","out"],{"blocking":false,"group":"","id":20,"name":"out","queueSize":8,"type":0,"waitForMessage":false}]],"name":"XLinkIn","properties":{"maxDataSize":5242880,"numFrames":8,"streamName":"right_control"}}],[5,{"id":5,"ioInfo":[[["","depth"],{"blocking":false,"group":"","id":19,"name":"depth","queueSize":8,"type":0,"waitForMessage":false}],[["","rectifiedLeft"],{"blocking":false,"group":"","id":17,"name":"rectifiedLeft","queueSize":8,"type":0,"waitForMessage":false}],[["","syncedLeft"],{"blocking":false,"group":"","id":18,"name":"syncedLeft","queueSize":8,"type":0,"waitForMessage":false}],[["","syncedRight"],{"blocking":false,"group":"","id":14,"name":"syncedRight","queueSize":8,"type":0,"waitForMessage":false}],[["","rectifiedRight"],{"blocking":false,"group":"","id":13,"name":"rectifiedRight","queueSize":8,"type":0,"waitForMessage":false}],[["","outConfig"],{"blocking":false,"group":"","id":11, [... cut ...]
        "ufoLib2==0.13.1",
        "unicodedata2==14.0.0",
        "unique-identifier-msgs==2.3.2",
        "urllib3==1.26.16",
        "vcstool==0.3.0",
        "vision-msgs==4.1.0",
        "visualization-msgs==5.0.0",
        "wheel==0.37.1",
        "xacro==2.0.9",
        "zipp==1.0.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux 000351f 6.1.35 #1 SMP PREEMPT_DYNAMIC Thu Jun 22 18:03:13 UTC 2023 x86_64 x86_64"
    ]
}
```

I also tried the Humble branch and I have the same issue

BriceRenaudeau, 2023-10-11:
Debug logs:
```bash
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-10-11-08-26-51-115683-000351f-291598
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [291613]
[component_container-1] [INFO] [1697012811.466670258] [front_depth_container]: Load Library: /src/robocc-ros/.build/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[component_container-1] [INFO] [1697012811.513448152] [front_depth_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1697012811.513484956] [front_depth_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [2023-10-11 08:26:51.515] [depthai] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[component_container-1] [2023-10-11 08:26:51.515] [depthai] [debug] Library information - version: 2.22.0, commit:  from , build: 2023-06-22 19:15:47 +0000
[component_container-1] [2023-10-11 08:26:51.517] [depthai] [debug] Initialize - finished
[component_container-1] [2023-10-11 08:26:51.585] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.26.tar.xz' open: 2ms, archive read: 67ms
[component_container-1] [2023-10-11 08:26:51.886] [depthai] [debug] Resources - Archive 'depthai-device-fwp-f033fd9c7eb0b3578d12f90302e87759c78cfb36.tar.xz' open: 1ms, archive read: 369ms
[component_container-1] [INFO] [1697012812.024717887] [cameras.oak_front]: Connecting to the camera using USB ID: 3.5
[component_container-1] [2023-10-11 08:26:53.454] [depthai] [debug] Searching for booted device: DeviceInfo(name=3.5, mxid=1844301021C1AF0E00, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[component_container-1] [1844301021C1AF0E00] [3.5] [1.133] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304418 should be: 0x5E7D3A77
[component_container-1] GPIO boot mode 0x3, interface SPI_MASTER_EFF
[component_container-1] Setting aons(0..4) back to boot from flash (offset = 0)
[component_container-1] ====ENABLE WATCHDOG====1
[component_container-1] initial keepalive, countdown: 10
[component_container-1] PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
[component_container-1] Board init ret 3
[component_container-1] DepthAI Firmware - version: f033fd9c7eb0b3578d12f90302e87759c78cfb36
[component_container-1] eeprom configuration version: 55AA0007
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] eeprom configuration load from user area, status: 0
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] PMIC read  reg 0x19, val 0xb8
[component_container-1] PMIC read  reg 0x23, val 0x10
[component_container-1] PMIC read  reg 0x31, val 0x08
[component_container-1] PMIC read  reg 0x33, val 0x28
[component_container-1] Found a device/board entry matching the eeprom data - Board: OAK-D S2/Pro (R6 - generic) Device: 
[component_container-1] Board options: 00000004
[component_container-1] --> brdInit ...
[component_container-1] brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]
[component_container-1] 
[component_container-1] spi_N25Q_init: Flash JEDEC ID: c2 25 3a
[component_container-1] SR CR: 40 07
[component_container-1] QUAD mode already enabled. Dummy cfg: 0x00
[component_container-1] Initialized pin: 56, mode: 7, level: 1, pad: 0x00000010
[component_container-1] inited hal ---
[component_container-1] def: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
[component_container-1] val: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
[component_container-1] Opening bus for IR driver: 1
[component_container-1] Failed to probe IR driver LM3644
[component_container-1] Opening bus for IR driver: 2
[component_container-1] LM3644 detected, ID = 0x02
[component_container-1] ===== IR write bus 2: 0x07 = 0x89
[component_container-1] ===== IR write bus 2: 0x02 = 0x01
[component_container-1] ===== IR write bus 2: 0x03 = 0x00
[component_container-1] ===== IR write bus 2: 0x04 = 0x00
[component_container-1] ===== IR write bus 2: 0x05 = 0x00
[component_container-1] ===== IR write bus 2: 0x06 = 0x00
[component_container-1] ===== IR write bus 2: 0x07 = 0x09
[component_container-1] ===== IR write bus 2: 0x08 = 0x1a
[component_container-1] ===== IR write bus 2: 0x09 = 0x08
[component_container-1] ===== IR write bus 2: 0x01 = 0x24
[component_container-1] Opening bus for IR driver: 3
[component_container-1] Failed to probe IR driver LM3644
[component_container-1] Closing EEPROm!
[component_container-1] MyriaX board configuration
[component_container-1]         skipped OV7251
[component_container-1]         skipped AR0234
[component_container-1]         skipped S5K33D
[component_container-1]   >> Registered camera A12N02A (imx378) as /dev/Camera_0
[component_container-1] camera socket: 0, name: color
[component_container-1]         config - w: 1920, h: 1080, type: COLOR
[component_container-1]         config - w: 3840, h: 2160, type: COLOR
[component_container-1]         config - w: 1280, h: 720, type: COLOR
[component_container-1]         config - w: 1280, h: 800, type: MONO
[... 41 lines cut ...]
[component_container-1] [E] lens_move_to_def_pos_do():251: Error executing af_move_to_pos()!
[... cut ...]
[component_container-1] [1844301021C1AF0E00] [3.5] [29.164] [system] [info] Temperatures - Average: 40.04C, CSS: 40.97C, MSS 39.35C, UPA: 40.04C, DSS: 39.81C
[component_container-1] [1844301021C1AF0E00] [3.5] [29.164] [system] [info] Cpu Usage - LeonOS 17.77%, LeonRT: 3.65%
[component_container-1] [1844301021C1AF0E00] [3.5] [30.166] [system] [info] Cpu Usage - LeonOS 17.72%, LeonRT: 3.49%
[component_container-1] [1844301021C1AF0E00] [3.5] [31.167] [system] [info] Memory Usage - DDR: 114.94 / 337.18 MiB, CMX: 2.10 / 2.50 MiB, LeonOS Heap: 36.12 / 80.03 MiB, LeonRT Heap: 4.14 / 41.06 MiB
[exited]
```

BriceRenaudeau, 2023-10-11:
I reduced the search to the commit "update socket logic": #400

Serafadam, 2023-10-12:
Hi @BriceRenaudeau, I've found the culprit, default resolution for stereo cameras has been switched to 800P which caused desynchronization. It's been reverted in https://github.com/luxonis/depthai-ros/tree/frequency_fix_iron and https://github.com/luxonis/depthai-ros/tree/frequency_fix_humble, you can test them and see if that works for you as well.

BriceRenaudeau, 2023-10-13:
Hi, I tested the Iron branch.

It's better, the mean frequency is 15hz. But it"s still less stable than before:
[image]

Serafadam, 2023-10-13:
Could you show the debug logs now? One "bugfix" that was also introduced was to make the resolution naming the same, so instead of "400" it should be "400P" for the stereo cameras as well.

BriceRenaudeau, 2023-10-20:
On `Iron` branch using "720P" i_resolution parameters. The frequency issue is still here:
[image]

Here are the debug Logs:
```
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-10-20-09-50-19-602927-000351f-66933
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [66948]
[component_container-1] [INFO] [1697795419.985660887] [front_depth_container]: Load Library: /src/robocc-ros/.build/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[component_container-1] [INFO] [1697795420.070049494] [front_depth_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1697795420.070095280] [front_depth_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [2023-10-20 09:50:20.071] [depthai] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[component_container-1] [2023-10-20 09:50:20.071] [depthai] [debug] Library information - version: 2.22.0, commit:  from , build: 2023-06-22 19:15:47 +0000
[component_container-1] [2023-10-20 09:50:20.072] [depthai] [debug] Initialize - finished
[component_container-1] [2023-10-20 09:50:20.140] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.26.tar.xz' open: 1ms, archive read: 66ms
[component_container-1] [2023-10-20 09:50:20.442] [depthai] [debug] Resources - Archive 'depthai-device-fwp-f033fd9c7eb0b3578d12f90302e87759c78cfb36.tar.xz' open: 1ms, archive read: 369ms
[component_container-1] [INFO] [1697795420.578794613] [cameras.oak_front]: Connecting to the camera using USB ID: 3.5
[component_container-1] [2023-10-20 09:50:21.997] [depthai] [debug] Searching for booted device: DeviceInfo(name=3.5, mxid=1844301021C1AF0E00, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[component_container-1] [INFO] [1697795423.152981489] [cameras.oak_front]: Ignoring device info: MXID: 19443010B13C481300, Name: 3.6
[component_container-1] [1844301021C1AF0E00] [3.5] [1.139] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304418 should be: 0x5E7D3A77
[component_container-1] GPIO boot mode 0x3, interface SPI_MASTER_EFF
[component_container-1] Setting aons(0..4) back to boot from flash (offset = 0)
[component_container-1] ====ENABLE WATCHDOG====1
[component_container-1] initial keepalive, countdown: 10
[component_container-1] PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
[component_container-1] Board init ret 3
[component_container-1] DepthAI Firmware - version: f033fd9c7eb0b3578d12f90302e87759c78cfb36
[component_container-1] eeprom configuration version: 55AA0007
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] eeprom configuration load from user area, status: 0
[component_container-1] Reading VERSION 7 --- ->
[component_container-1] PMIC read  reg 0x19, val 0xb8
[component_container-1] PMIC read  reg 0x23, val 0x10
[component_container-1] PMIC read  reg 0x31, val 0x08
[component_container-1] PMIC read  reg 0x33, val 0x28
[component_container-1] Found a device/board entry matching the eeprom data - Board: OAK-D S2/Pro (R6 - generic) Device: 
[component_container-1] Board options: 00000004
[component_container-1] --> brdInit ...
[component_container-1] brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]
[component_container-1] 
[component_container-1] spi_N25Q_init: Flash JEDEC ID: c2 25 3a
[component_container-1] SR CR: 40 07
[component_container-1] QUAD mode already enabled. Dummy cfg: 0x00
[component_container-1] Initialized pin: 56, mode: 7, level: 1, pad: 0x00000010
[component_container-1] inited hal ---
[component_container-1] def: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
[component_container-1] val: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
[component_container-1] Opening bus for IR driver: 1
[component_container-1] Failed to probe IR driver LM3644
[component_container-1] Opening bus for IR driver: 2
[component_container-1] LM3644 detected, ID = 0x02
[component_container-1] ===== IR write bus 2: 0x07 = 0x89
[component_container-1] ===== IR write bus 2: 0x02 = 0x01
[component_container-1] ===== IR write bus 2: 0x03 = 0x00
[component_container-1] ===== IR write bus 2: 0x04 = 0x00
[component_container-1] ===== IR write bus 2: 0x05 = 0x00
[component_container-1] ===== IR write bus 2: 0x06 = 0x00
[component_container-1] ===== IR write bus 2: 0x07 = 0x09
[component_container-1] ===== IR write bus 2: 0x08 = 0x1a
[component_container-1] ===== IR write bus 2: 0x09 = 0x08
[component_container-1] ===== IR write bus 2: 0x01 = 0x24
[component_container-1] Opening bus for IR driver: 3
[component_container-1] Failed to probe IR driver LM3644
[component_container-1] Closing EEPROm!
[component_container-1] MyriaX board configuration
[component_container-1] pll0 frequency: 700000, ref0 frequency: 24000
[component_container-1] Is booted from flash by bootloader: 0
[component_container-1] Networking not available...
[component_container-1] === Enumerating on socket: Cam_A / RGB / Center
[component_container-1]         skipped OV7251
[... 136 lines cut ...]
[component_container-1] [INFO] [1697795423.606017325] [cameras.oak_front]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/cameras/oak_front' in container '/front_depth_container'
[... cut ...]
[component_container-1] [1844301021C1AF0E00] [3.5] [4.146] [system] [info] Cpu Usage - LeonOS 17.86%, LeonRT: 3.81%
[component_container-1] [1844301021C1AF0E00] [3.5] [5.147] [system] [info] Memory Usage - DDR: 111.46 / 337.18 MiB, CMX: 2.10 / 2.50 MiB, LeonOS Heap: 36.12 / 80.03 MiB, LeonRT Heap: 4.14 / 41.06 MiB
[component_container-1] [1844301021C1AF0E00] [3.5] [5.147] [system] [info] Temperatures - Average: 31.59C, CSS: 32.73C, MSS 31.77C, UPA: 31.04C, DSS: 30.80C
[component_container-1] [1844301021C1AF0E00] [3.5] [5.147] [system] [info] Cpu Usage - LeonOS 17.85%, LeonRT: 4.10%
```

BriceRenaudeau, 2023-10-20:
I was able to get a stable frequency by turning ON only the stereo topic.

Whenever I turn on one of the camera (left or right) `i_publish_topic: true`. The frequency becomes unstable.

Serafadam, 2023-10-20:
Hi:
- Could you also set up the log level to debug for the ROS node (similarly to what is done [here]?(https://github.com/luxonis/depthai-ros/blob/4795034f539dbbef7e95168a8579b170fca1538c/depthai_ros_driver/launch/camera.launch.py#L110)
- I can see that your USB speed is high, which indicates USB2.0, is it possible for you to use USB3.0?
- Could you verify this issue also persists when using the tagged Docker images?
- Could you share the current saved pipeline.json?

BriceRenaudeau, 2023-10-20:
Here is the full debug logs with frequency issues: 
[oak_full_debug_log.txt](https://github.com/luxonis/depthai-ros/files/13055581/oak_full_debug_log.txt)

Here is the pipeline:
[camera_pipeline.json](https://github.com/luxonis/depthai-ros/files/13055600/camera_pipeline.json)

I tried using `i_usb_speed: SUPER_PLUS` but the frequency issue is still here.

Unfortunately, I cannot test using the tagged Docker images today.

Serafadam, 2023-11-06:
Hi @BriceRenaudeau, have you managed to change the USB speed? It seems it is the culprit in this case, as USB2.0's bandwidth can cause issues with message transfer.

BriceRenaudeau, 2023-11-06:
Hi, yes, I changed it to `i_usb_speed: SUPER_PLUS ` and I got the same result.

It was working great before 2.8.1 using USB 2.0.

Serafadam, 2023-11-06:
It seems that for some reason the device can't use faster USB speed, are you using USB3.0 port? And is the cable the one that came with the camera? From parameters it seems current resolution for stereo pair is 720P, whereas before it was set to 400P, could you change the params file and check if the results improve?

BriceRenaudeau, 2023-11-06:
By changing the resolution back to 400P, the frequency is stable even with the left topic published.

BriceRenaudeau, 2023-11-06:
I think there were several changes mixed together.

- I can set `i_usb_speed: SUPER_PLUS` in the config file, but the logs still say `[cameras.oak_back]: USB SPEED: HIGH`
- It was working on the 2.7.4 tag because I was using 400P
- we are using another cable because of the length, and that may be the source of the problem.

Serafadam, 2023-11-07:
Glad it got solved, sorry for the issues. Regarding the USB speed/cable, using a powered USB hub could improve the performance here.

closed 2023-11-07 by Serafadam