luxonis/depthai-ros #431: How to set OAK-1 fixed focus.

rahulsharma11, 2023-10-11:
Hi.

Any suggestion on how to set the fixed focus in OAK-1 camera ? I want to use this camera for my drone so i need fixed focus.
I tried following methods but no success:
inside depthai_ros_driver/src/param_handlers/sensor_param_handler.cpp
Line 140:
size_t focus = declareAndLogParam("r_focus", 130, getRangedIntDescriptor(0, 255));
    if(declareAndLogParam("r_set_man_focus", true)) {
        colorCam->initialControl.setManualFocus(focus);
    }

Any help?
Thanks.

Serafadam, 2023-10-11:
Hi, you can set fixed focus using parameters, either in yaml file or by using dynamic reconfigure as shown in the [Readme](https://github.com/luxonis/depthai-ros#setting-rgb-parameters)

rahulsharma11, 2023-10-13:
Thanks @Serafadam

closed 2023-10-13 by rahulsharma11