luxonis/depthai-ros #437: Synchronizing Timestamps between the OAK Camera and Computer for Unified Data Acquisition

Samaleke, 2023-10-13:
I am encountering an issue with timestamp synchronization between the OAK camera and a connected computer. The camera generates its own timestamp, which differs from that of the computer. Consequently, all nodes utilizing the camera data adhere to its timestamp, while other nodes follow the computer's timestamp. This discrepancy presents a challenge in simultaneously obtaining data from the LiDAR and camera within the same frame, due to the differing timestamps. Is there a way to synchronize the camera's timestamp with that of the hosting computer?

Serafadam, 2023-10-13:
Hi, could you record a rosbag containing lidar data and OAK camera data for comparison?

BriceRenaudeau, 2023-10-19:
I got the same issue, it ended up with the navigation rejecting the camera info because the timestamp was too different or even in the future.

Serafadam, 2023-10-19:
Hi @BriceRenaudeau, could you share more details? Is this something that's started happening recently?

BriceRenaudeau, 2023-10-19:
It happened on the Iron 2.7.4 tag.
It happened after several hours of use (> 10h) and we have seen it only one time.
I will do some experiments and bags on the camera tomorrow but it will be nearly impossible to reproduce as the occurrence is very low.
[image]

Serafadam, 2023-10-19:
It sound an issue that was reported and fixed in [ROS Noetic](https://github.com/luxonis/depthai-ros/pull/332) a while ago, since there were no report for ROS2 I assumed that this wasn't an issue with ROS2, nevertheless the same mechanism has been added in ROS2 releases, you can try checking if enabling `i_update_ros_base_time_on_ros_msg` parameter for given nodes makes a change in this case.

BriceRenaudeau, 2023-10-20:
Here is the graph of the time stamps of the left/right/stereo images without `i_update_ros_base_time_on_ros_msg`. There not sync.
[image]

BriceRenaudeau, 2023-10-20:
Here is the time stamp graph comparison with /odom /laser_scan topics. And using the `i_update_ros_base_time_on_ros_msg: true` :

[image]