luxonis/depthai-ros #445: [BUG] {DepthAi Bridge CMakeFile cannot find tf2_geometry_msgs}

Scoeerg, 2023-10-17:
**Describe the bug**
```
colcon build
Starting >>> depthai_ros_msgs
Starting >>> depthai_descriptions
Finished <<< depthai_descriptions [0.25s]                                                                  
Finished <<< depthai_ros_msgs [1.01s]                     
Starting >>> depthai_bridge
Starting >>> depthai_filters
Finished <<< depthai_filters [0.38s]                                                                   
--- stderr: depthai_bridge                             
~/ros2humble_ws/src/depthai-ros/depthai_bridge/src/TFPublisher.cpp:19:10: fatal error: 
tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp: No such file or directory
19 | #include "tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp"

compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_bridge.dir/build.make:146: CMakeFiles/depthai_bridge.dir/src/TFPublisher.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/depthai_bridge.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< depthai_bridge [0.77s, exited with code 2]

Summary: 3 packages finished [2.53s]
  1 package failed: depthai_bridge
  1 package had stderr output: depthai_bridge
  3 packages not processed
```

**Minimal Reproducible Example**

Assume there is a Humble workspace (e.g. ~/ros_ws/ where ~/ros_ws/src exists), then 

```
cd  ~/ros_ws/src
sudo sh ./installscript.sh
```

where `installscript.sh` is just a compact script to follow the [github.com/luxonis/depthai-ros#install-from-source](https://github.com/luxonis/depthai-ros#install-from-source) instructions

```
#!/bin/bash

## PREPARING
	#Setup Directory
	SourceDirectory="$(dirname "$0")"
	SourceDirectory_FullPath=$(cd "$(dirname ${SourceDirectory})" && pwd)/$(basename ${SourceDirectory})
	echo "Installing DepthAI ROS code into ${SourceDirectory_FullPath}"
	sudo git config --global --add safe.directory ${SourceDirectory}

## ROS2 Depth AI 
	# Pull Git Repository for the Depth AI 
			echo "Pulling and installing Depth AI ROS2 from https://github.com/luxonis/depthai-ros.git"
			sudo rm -r depthai-ros
			git clone humble https://github.com/luxonis/depthai-ros.git
			depthai_directory="$(readlink -f ${SourceDirectory_FullPath})/depthai-ros/"
			echo "depthai_Directory is $(readlink -f ${depthai_directory})"
			sudo chown -R ${SUDO_USER} ${depthai_directory}
	# Install Dependencies
			workspace_directory="$(readlink -f ${SourceDirectory_FullPath})/.."
			cd ${workspace_directory}
			rosdep install --from-paths src --ignore-src -r -y
			sudo apt install ros-humble-rtabmap-ros
			sudo apt install ros-humble-camera-info-manager
			sudo apt install ros-humble-vision-msgs
	# Prepare System UDEV Rules & install depth-Ai-Core
			echo "Installing Depth AI core library"
			echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
			sudo udevadm control --reload-rules && sudo udevadm trigger
			mkdir tmp
			temp_directory="$(readlink -f ${SourceDirectory_FullPath})/tmp/"
			cd ${temp_directory}
			git clone --recursive https://github.com/luxonis/depthai-core.git --branch main
			cmake -Hdepthai-core -Bdepthai-core/build -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=/usr/local
			cmake --build depthai-core/build --target install
			cd ..
			sudo rm -r ${temp_directory}
```

then 

```
cd ~/ros_ws
colcon build
```

which results ins:
```
Starting >>> depthai_ros_msgs
Starting >>> depthai_descriptions
Finished <<< depthai_descriptions [0.25s]                                                                  
Finished <<< depthai_ros_msgs [0.99s]                     
Starting >>> depthai_bridge
Starting >>> depthai_filters                 
Finished <<< depthai_filters [0.32s]                                                                
--- stderr: depthai_bridge                             
/home/user/ros_ws/src/depthai-ros/depthai_bridge/src/TFPublisher.cpp:19:10: fatal error: tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp: No such file or directory
   19 | #include "tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_bridge.dir/build.make:146: CMakeFiles/depthai_bridge.dir/src/TFPublisher.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/depthai_bridge.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< depthai_bridge [0.75s, exited with code 2]

Summary: 3 packages finished [2.48s]
  1 package failed: depthai_bridge
  1 package had stderr output: depthai_bridge
  3 packages not processed
```

Of course I made sure the `tf2_geometry_msgs` was installed, by

```
sudo apt install ros-humble-tf2-geometry-msgs
```

might as well add it to the install script. Already did the same with `ros-humble-rtabmap-ros`, `ros-humble-camera-info-manager` and `ros-humble-vision-msgs` which were not installed with `rosdep` for whatever reason. 

**Expected behavior**
Builds without Error.

Serafadam, 2023-10-17:
Hi, I would guess that `rosdep` part is not working properly for some unknown reason, could you share the logs from that step? For source installation steps you can also refer to the [Dockerfile](https://github.com/luxonis/depthai-ros/blob/humble/Dockerfile)

Scoeerg, 2023-10-17:
Hi. By hand in `~/ros_ws/` `rosdep install --from-paths src --ignore-src -r -y` results in:

```
ERROR: your rosdep installation has not been initialized yet.  Please run:
    rosdep update
```

so I did `rosdep update` and repeated `rosdep install --from-paths src --ignore-src -r -y` which then:

```
executing command [sudo -H apt-get install -y ros-humble-xacro]
[sudo] password for user: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  dctrl-tools dkms gconf-service gconf-service-backend gconf2 gconf2-common libflashrom1 libftdi1-2 libgconf-2-4 libllvm13 libllvm13:i386 libnvidia-cfg1-515 libnvidia-common-515 libnvidia-compute-515 libnvidia-compute-515:i386 libnvidia-decode-515
  libnvidia-decode-515:i386 libnvidia-encode-515 libnvidia-encode-515:i386 libnvidia-extra-515 libnvidia-fbc1-515 libnvidia-fbc1-515:i386 libnvidia-gl-515 libnvidia-gl-515:i386 libvulkan1:i386 libwayland-client0:i386 mercurial mercurial-common
  mesa-vulkan-drivers:i386 nvidia-compute-utils-515 nvidia-kernel-common-515 nvidia-utils-515 python3-vcstool python3-vcstools ros-build-essential xserver-xorg-video-nvidia-515
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  ros-humble-xacro
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 36.8 kB of archives.
After this operation, 154 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-xacro amd64 2.0.8-1jammy.20230919.194745 [36.8 kB]
Fetched 36.8 kB in 0s (103 kB/s)            
Selecting previously unselected package ros-humble-xacro.
(Reading database ... 284119 files and directories currently installed.)
Preparing to unpack .../ros-humble-xacro_2.0.8-1jammy.20230919.194745_amd64.deb ...
Unpacking ros-humble-xacro (2.0.8-1jammy.20230919.194745) ...
Setting up ros-humble-xacro (2.0.8-1jammy.20230919.194745) ...
executing command [sudo -H apt-get install -y ros-humble-depthai]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  dctrl-tools dkms gconf-service gconf-service-backend gconf2 gconf2-common libflashrom1 libftdi1-2 libgconf-2-4 libllvm13 libllvm13:i386 libnvidia-cfg1-515 libnvidia-common-515 libnvidia-compute-515 libnvidia-compute-515:i386 libnvidia-decode-515
  libnvidia-decode-515:i386 libnvidia-encode-515 libnvidia-encode-515:i386 libnvidia-extra-515 libnvidia-fbc1-515 libnvidia-fbc1-515:i386 libnvidia-gl-515 libnvidia-gl-515:i386 libvulkan1:i386 libwayland-client0:i386 mercurial mercurial-common
  mesa-vulkan-drivers:i386 nvidia-compute-utils-515 nvidia-kernel-common-515 nvidia-utils-515 python3-vcstool python3-vcstools ros-build-essential xserver-xorg-video-nvidia-515
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  ros-humble-depthai
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 10.5 MB of archives.
After this operation, 20.7 MB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-depthai amd64 2.22.0-1jammy.20230623.022440 [10.5 MB]
Fetched 10.5 MB in 9s (1,208 kB/s)                                                                                                                                                                                                                                         
Selecting previously unselected package ros-humble-depthai.
(Reading database ... 284159 files and directories currently installed.)
Preparing to unpack .../ros-humble-depthai_2.22.0-1jammy.20230623.022440_amd64.deb ...
Unpacking ros-humble-depthai (2.22.0-1jammy.20230623.022440) ...
Setting up ros-humble-depthai (2.22.0-1jammy.20230623.022440) ...
Processing triggers for libc-bin (2.35-0ubuntu3.4) ...
executing command [sudo -H apt-get install -y ros-humble-camera-calibration]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  dctrl-tools dkms gconf-service gconf-service-backend gconf2 gconf2-common libflashrom1 libftdi1-2 libgconf-2-4 libllvm13 libllvm13:i386 libnvidia-cfg1-515 libnvidia-common-515 libnvidia-compute-515 libnvidia-compute-515:i386 libnvidia-decode-515
  libnvidia-decode-515:i386 libnvidia-encode-515 libnvidia-encode-515:i386 libnvidia-extra-515 libnvidia-fbc1-515 libnvidia-fbc1-515:i386 libnvidia-gl-515 libnvidia-gl-515:i386 libvulkan1:i386 libwayland-client0:i386 mercurial mercurial-common
  mesa-vulkan-drivers:i386 nvidia-compute-utils-515 nvidia-kernel-common-515 nvidia-utils-515 python3-vcstool python3-vcstools ros-build-essential xserver-xorg-video-nvidia-515
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  ros-humble-camera-calibration
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 53.9 kB of archives.
After this operation, 217 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-camera-calibration amd64 3.0.0-1jammy.20230919.224259 [53.9 kB]
Fetched 53.9 kB in 0s (127 kB/s)                         
[... 136 lines cut ...]
Need to get 169 kB of archives.
[... cut ...]
  libnvidia-decode-515:i386 libnvidia-encode-515 libnvidia-encode-515:i386 libnvidia-extra-515 libnvidia-fbc1-515 libnvidia-fbc1-515:i386 libnvidia-gl-515 libnvidia-gl-515:i386 libvulkan1:i386 libwayland-client0:i386 mercurial mercurial-common
  mesa-vulkan-drivers:i386 nvidia-compute-utils-515 nvidia-kernel-common-515 nvidia-utils-515 python3-vcstool python3-vcstools ros-build-essential xserver-xorg-video-nvidia-515
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
#All required rosdeps installed successfully
```

essentially it says it did not find any more required packages to install.

Scoeerg, 2023-10-17:
Update: I just checked `/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp` exists.

Scoeerg, 2023-10-17:
Based on the docker-file (does essentially what I did), ran the `ros_ws/src/depthai-ros/build.sh`

```
sh ./src/depthai-ros/build.sh 
./src/depthai-ros/build.sh: 35: [: 0: unexpected operator
./src/depthai-ros/build.sh: 40: [: 0: unexpected operator
Build type: Release, Install_type: symlink-install
./src/depthai-ros/build.sh: 45: [: 1: unexpected operator
Parallel build
Starting >>> depthai_ros_msgs
Starting >>> depthai_descriptions
--- stderr: depthai_descriptions                                                                       
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_SHARED_LIBS

---
Finished <<< depthai_descriptions [0.68s]
--- stderr: depthai_ros_msgs                             
failed to create symbolic link 'ros_ws/build/depthai_ros_msgs/ament_cmake_python/depthai_ros_msgs/depthai_ros_msgs' because existing path cannot be removed: Is a directory
gmake[2]: *** [CMakeFiles/ament_cmake_python_symlink_depthai_ros_msgs.dir/build.make:70: CMakeFiles/ament_cmake_python_symlink_depthai_ros_msgs] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:421: CMakeFiles/ament_cmake_python_symlink_depthai_ros_msgs.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< depthai_ros_msgs [2.02s, exited with code 2]

Summary: 1 package finished [2.78s]
  1 package failed: depthai_ros_msgs
  2 packages had stderr output: depthai_descriptions depthai_ros_msgs
  5 packages not processed
```

removed `/build /install /log` and ran `colcon build` again. works. Whatever this was...

closed 2023-10-17 by Scoeerg