luxonis/depthai-ros #446: [BUG] {Unable to get IMU data rate > 100 Hz when rotation is enabled}

ehong-tl, 2023-10-25:
**Describe the bug**
Unable to get IMU data rate of more than 100 Hz when rotation is enabled.

**Expected behavior**
To get IMU data rate as intended in camera.yaml config file.

**Screenshots**
camera.yaml
`/oak:
  ros__parameters:
    camera:
      i_calibration_dump: false
      i_enable_imu: true
      i_enable_ir: true
      i_external_calibration_path: ''
      i_floodlight_brightness: 0
      i_ip: ''
      i_laser_dot_brightness: 500
      i_mx_id: ''
      i_nn_type: none
      i_pipeline_dump: false
      i_pipeline_type: RGBD
      i_publish_tf_from_calibration: true
      i_tf_base_frame: oak
      i_tf_cam_pitch: '0.0'
      i_tf_cam_pos_x: '0.0'
      i_tf_cam_pos_y: '0.0'
      i_tf_cam_pos_z: '0.0'
      i_tf_cam_roll: '0.0'
      i_tf_cam_yaw: '0.0'
      i_tf_camera_model: ''
      i_tf_camera_name: oak
      i_tf_custom_urdf_location: ''
      i_tf_custom_xacro_args: ''
      i_tf_imu_from_descr: 'true'
      i_tf_parent_frame: oak-d-base-frame
      i_usb_port_id: ''
      i_usb_speed: SUPER_PLUS
    diagnostic_updater:
      period: 1.0
    imu:
      i_acc_cov: 0.0
      i_acc_freq: 250
      i_batch_report_threshold: 5
      i_enable_rotation: true
      i_get_base_device_timestamp: false
      i_gyro_cov: 0.0
      i_gyro_freq: 200
      i_mag_cov: 0.0
      i_max_batch_reports: 10
      i_message_type: IMU
      i_rot_cov: 0.0
      i_rot_freq: 200
      i_sync_method: LINEAR_INTERPOLATE_ACCEL
      i_update_ros_base_time_on_ros_msg: false
    left:
      i_add_exposure_offset: false
      i_board_socket_id: 1
      i_calibration_file: 'package://depthai_ros_driver/config/calibration/left_640x400.yaml'
      i_disable_node: false
      i_enable_feature_tracker: false
      i_enable_lazy_publisher: true
      i_exposure_offset: 0
      i_fps: 30.0
      i_fsync_continuous: false
      i_fsync_trigger: false
      i_get_base_device_timestamp: false
      i_low_bandwidth: false
      i_low_bandwidth_quality: 50
      i_max_q_size: 30
[... 356 lines cut ...]
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        }
    ],
    "uname": [
        "Linux orinnano1-desktop 5.10.120-tegra #1 SMP PREEMPT Tue Aug 1 12:32:50 PDT 2023 aarch64 aarch64"
    ]
}`
 - Which OS/OS version are you using?
 Ubuntu 20.04 (Jetpack 5.1.2)
 - Which ROS version are you using?
 ROS2
 - Which ROS distribution are you using ?
 ROS2 Humble
 - Is `depthai-ros` built from source or installed from apt?
 From source
 - Is `depthai/depthai-core` library installed from rosdep or manually? For rosdep install, check if `ros-<rosdistro>-depthai` package is installed, manual install can be checked with `ldconfig -p | grep depthai`
 Installed manually

**Additional context**
Any reason not to output rotation vector when in COPY sync mode?
https://github.com/luxonis/depthai-ros/blob/4795034f539dbbef7e95168a8579b170fca1538c/depthai_bridge/src/ImuConverter.cpp#L90C6-L90C18

danilo-pejovic, 2023-10-27:
Hey thanks for reporting the issue, does this issue appear on any other device or only on jetpack?

ehong-tl, 2023-10-27:
Hey @danilo-pejovic

I've only tried it on Nvidia Jetson Orin Nano with ROS2 running in [jetson docker container](https://github.com/dusty-nv/jetson-containers).

With this camera config, I am able to get IMU data rate of ~200 Hz in COPY mode and ~185 Hz in LINEAR_INTERPOLATE_ACCEL mode when rotation is disabled.
I'm unable to get IMU data rate of more than 100 Hz when rotation is enabled regardless of the settings.

Haven't try it on PC yet, I shall find one day to try it.

ehong-tl, 2023-11-01:
Hey @danilo-pejovic 

I've tested on my laptop running on Ubuntu 22.04 ROS2 Humble and got the same result, IMU rate will not get over 100 Hz when rotation is enabled.

Serafadam, 2023-11-06:
Hi @ehong-tl, could you test[ this branch ](https://github.com/luxonis/depthai-ros/tree/separate_mag_humble)and see if that improves the readings?

ehong-tl, 2023-11-07:
> Hi @ehong-tl, could you test[ this branch ](https://github.com/luxonis/depthai-ros/tree/separate_mag_humble)and see if that improves the readings?

Hi @Serafadam 

I'm able to get more than 100 Hz with this branch, thanks for the fix.