luxonis/depthai-ros #448: [BUG] Cannot locate node of type [stereo_mimic_node] in package [depthai_examples]

zippyzoo77, 2023-11-03:
Hello,

I am trying to get separate stereo camera feeds so that I can run various SLAM repositories that need the stereo in two different ros topics (one for left and one for right), and I saw the stereo_mimicer example, and thought that might be just the kind of node I need, but it doesn't seem to have all of the required files to make it work. (Other examples like the stereo inertial node work fine, they just don't have the outputs I need.) So I guess two questions here, first does that example do what I am looking for? Second, how do I fix the error I keep getting? It looks like the file is just missing? I just updated both the standalone depthai stuff and the depthai-ros, and I can't find the needed file in the git repository either.

error: `ERROR: cannot launch node of type [depthai_examples/stereo_mimic_node]: Cannot locate node of type [stereo_mimic_node] in package [depthai_examples]. Make sure file exists in package path and permission is set to executable (chmod +x)`

ros version: noetic
ubuntu version: 20.04

code to reproduce: `roslaunch depthai_examples stereo_mimicer.launch`

Serafadam, 2023-11-06:
Hi, first I'd like to clarify the desired outputs, do you want to output `left` and `right` sensor streams from the stereo node? If yes, this can be enabled in `depthai_ros_driver` using [parameters](https://github.com/luxonis/depthai-ros#depth-alignment).

The `stereo_mimicer` has been deprecated some time ago, although it seems that the launch file was missed for the deprecation in noetic branch, you can find the [node code here](https://github.com/luxonis/depthai-ros/blob/f00038a3d5b5fe9a6c155d8f9b45cbecfa897e9e/depthai_examples/src/stereo_mimic_node.cpp).

zippyzoo77, 2023-11-06:
Yes the sensor streams is what I am looking for, I will try that thanks!

Serafadam, 2023-11-07:
Glad I could help. Closing the issue, please reopen or make a new one if you have further questions.

closed 2023-11-07 by Serafadam