luxonis/depthai-ros #452: [BUG] Error launching custom spatial Yolo model

mkarklins, 2023-11-09:
**Describe the bug**
Unabe to launch custom spatial Yolo NN, however it works from plain pipeline without ROS.

The error message seems to be:

```
[component_container-2] [INFO] [1699512205.069016888] [oak]: Camera with MXID: 18443010316CAB0F00 and Name: 3.2 connected!
[component_container-2] [INFO] [1699512205.073139504] [oak]: USB SPEED: SUPER
[component_container-2] [INFO] [1699512205.096143270] [oak]: Device type: OAK-D-PRO-FF
[component_container-2] [INFO] [1699512205.107098481] [oak]: Pipeline type: RGBD
[component_container-2] [INFO] [1699512205.149628352] [oak]: NN Family: YOLO
[component_container-2] [ERROR] [1699512209.924899599] [oak_container]: Component constructor threw an exception: [json.exception.type_error.302] type must be string, but is null
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: [json.exception.type_error.302] type must be string, but is null
```

**Minimal Reproducible Example**

Here are the relevant files:

<details>

<summary>main.py - this is minimally adjusted segment.launch.py</summary>

```
import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription, OpaqueFunction
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration

def launch_setup(context, *args, **kwargs):
    name = LaunchConfiguration('name').perform(context)
    depthai_prefix = get_package_share_directory("depthai_ros_driver")
    rviz_config = os.path.join(depthai_prefix, "config", "rviz", "segmentation.rviz")

    params_file = LaunchConfiguration("params_file")

    return [
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(
                os.path.join(depthai_prefix, 'launch', 'camera.launch.py')),
            launch_arguments={"name": name,
                              "params_file": params_file,
                              "use_rviz": LaunchConfiguration("use_rviz"),
                              "rviz_config": rviz_config}.items())
    ]

def generate_launch_description():
    declared_arguments = [
        DeclareLaunchArgument("name", default_value="oak"),
        DeclareLaunchArgument("params_file", default_value="./params.yaml"),
        DeclareLaunchArgument("use_rviz", default_value="True"),
    ]

    return LaunchDescription(
        declared_arguments + [OpaqueFunction(function=launch_setup)]
    )

```

</details>

<details>
<summary>params.yaml</summary>

```
/oak:
  ros__parameters:
[... 623 lines cut ...]
Version: 2.8.1-2jammy.20230919.232227
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 1 723 kB
Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.3.1), libpython3.10 (>= 3.10.0), libstdc++6 (>= 11), ros-humble-fastcdr, ros-humble-builtin-interfaces, ros-humble-geometry-msgs, ros-humble-rclcpp, ros-humble-rosidl-default-generators, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 129 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: Package to keep interface independent of the driver

Package: ros-humble-depthai-ros-driver
Version: 2.8.1-2jammy.20230920.001359
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 2 782 kB
Depends: libc6 (>= 2.34), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 12), libtinyxml2-9 (>= 8.0.0), ros-humble-ament-cmake-auto, ros-humble-camera-calibration, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-depthai-descriptions, ros-humble-depthai-examples, ros-humble-depthai-ros-msgs, ros-humble-diagnostic-msgs, ros-humble-diagnostic-updater, ros-humble-image-pipeline, ros-humble-image-transport, ros-humble-image-transport-plugins, ros-humble-pluginlib, ros-humble-rclcpp, ros-humble-rclcpp-components, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-std-srvs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 616 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: Depthai ROS Monolithic node.
```

</details>

mkarklins, 2023-11-09:
<details>
<summary> Verbose logs with DEPTHAI_DEBUG=1</summary>

```
mkarklins@mkarklins:~/code/temp$ ros2 launch main.py
[INFO] [launch]: All log files can be found below /home/mkarklins/.ros/log/2023-11-09-08-22-45-322020-mkarklins-20400
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [20414]
[INFO] [component_container-2]: process started with pid [20416]
[component_container-2] [DEBUG] [1699510965.612146880] [rclcpp]: signal handler installed
[component_container-2] [DEBUG] [1699510965.612234864] [rclcpp]: deferred_signal_handler(): waiting for SIGINT/SIGTERM or uninstall
[component_container-2] [DEBUG] [1699510965.612249610] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '0' services
[component_container-2] [DEBUG] [1699510965.612341210] [rcl]: Initializing node 'ComponentManager' in namespace ''
[component_container-2] [DEBUG] [1699510965.612355561] [rcl]: Using domain ID of '0'
[component_container-2] [DEBUG] [1699510965.620387694] [rcl]: Initializing publisher for topic name '/rosout'
[component_container-2] [DEBUG] [1699510965.620425027] [rcl]: Expanded and remapped topic name '/rosout'
[component_container-2] [DEBUG] [1699510965.621559125] [rcl]: Publisher initialized
[component_container-2] [DEBUG] [1699510965.621587858] [rcl]: Node initialized
[component_container-2] [DEBUG] [1699510965.621692811] [rcl]: Initializing subscription for topic name '/parameter_events'
[component_container-2] [DEBUG] [1699510965.621701443] [rcl]: Expanded and remapped topic name '/parameter_events'
[component_container-2] [DEBUG] [1699510965.622847562] [rcl]: Subscription initialized
[component_container-2] [DEBUG] [1699510965.623001390] [rcl]: Initializing service for service name '~/_container/load_node'
[component_container-2] [DEBUG] [1699510965.623014285] [rcl]: Expanded and remapped service name '/oak_container/_container/load_node'
[component_container-2] [DEBUG] [1699510965.623748624] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container-2] [DEBUG] [1699510965.623761636] [rmw_fastrtps_cpp]: Sub Topic rq/oak_container/_container/load_nodeRequest
[component_container-2] [DEBUG] [1699510965.623766050] [rmw_fastrtps_cpp]: Pub Topic rr/oak_container/_container/load_nodeReply
[component_container-2] [DEBUG] [1699510965.623769553] [rmw_fastrtps_cpp]: ***********
[component_container-2] [DEBUG] [1699510965.623833957] [rcl]: Service initialized
[component_container-2] [DEBUG] [1699510965.623861612] [rcl]: Initializing service for service name '~/_container/unload_node'
[component_container-2] [DEBUG] [1699510965.623869799] [rcl]: Expanded and remapped service name '/oak_container/_container/unload_node'
[component_container-2] [DEBUG] [1699510965.624204507] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container-2] [DEBUG] [1699510965.624216438] [rmw_fastrtps_cpp]: Sub Topic rq/oak_container/_container/unload_nodeRequest
[component_container-2] [DEBUG] [1699510965.624220683] [rmw_fastrtps_cpp]: Pub Topic rr/oak_container/_container/unload_nodeReply
[component_container-2] [DEBUG] [1699510965.624224087] [rmw_fastrtps_cpp]: ***********
[component_container-2] [DEBUG] [1699510965.624285473] [rcl]: Service initialized
[component_container-2] [DEBUG] [1699510965.624300613] [rcl]: Initializing service for service name '~/_container/list_nodes'
[component_container-2] [DEBUG] [1699510965.624308012] [rcl]: Expanded and remapped service name '/oak_container/_container/list_nodes'
[component_container-2] [DEBUG] [1699510965.624560654] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container-2] [DEBUG] [1699510965.624570278] [rmw_fastrtps_cpp]: Sub Topic rq/oak_container/_container/list_nodesRequest
[component_container-2] [DEBUG] [1699510965.624573934] [rmw_fastrtps_cpp]: Pub Topic rr/oak_container/_container/list_nodesReply
[component_container-2] [DEBUG] [1699510965.624577237] [rmw_fastrtps_cpp]: ***********
[component_container-2] [DEBUG] [1699510965.624656912] [rcl]: Service initialized
[component_container-2] [DEBUG] [1699510965.675478741] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1699510965.675540710] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1699510965.676059552] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1699510965.676082639] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1699510965.676099605] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1699510965.676133183] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1699510965.676146183] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1699510965.676152477] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1699510965.676160536] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1699510965.676166388] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1699510965.846962031] [rcl]: Service server taking service request
[component_container-2] [DEBUG] [1699510965.847038002] [rcl]: Service take request succeeded: true
[component_container-2] [INFO] [1699510965.847131149] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[... cut ...]
[component_container-2] [DEBUG] [1699510976.917933646] [rcl]: Node finalized
[component_container-2] [DEBUG] [1699510977.018695183] [rclcpp]: SignalHandler::uninstall(): notifying deferred signal handler
[component_container-2] [DEBUG] [1699510977.018757812] [rclcpp]: deferred_signal_handler(): woken up due to SIGINT/SIGTERM or uninstall
[component_container-2] [DEBUG] [1699510977.018772499] [rclcpp]: deferred_signal_handler(): signal handling uninstalled
[component_container-2] [DEBUG] [1699510977.018936166] [rclcpp]: signal handler uninstalled

```

</details>

Serafadam, 2023-11-09:
Hi, to load custom YOLO model, you would need to set the "zoo" parameter to "path" and then provide the path to the `.blob` file, similarly to how the [ current yolo example](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/nn/yolo.json) is set up.

mkarklins, 2023-11-09:
Thank you very much - works as expected!

I just didn't know where to look for the problem based on the error log.

closed 2023-11-09 by mkarklins