luxonis/depthai-ros #453: how to run stereo_inertial_node.launch with another IMU data

Ghada09, 2023-11-10:
I have oak d lite camera but it does not have the IMU built in, and I was planning to put this camera on my drone, so I thought about using the drone's IMU in the camera's example files as they don't run without IMU, so can you help me on what to change in these files maybe the urdf? so that it takes the IMU data from the drone not from the camera

Serafadam, 2023-11-13:
Hi, if the drone's driver supports it, then you should be able to get the IMU readings from it. To get the transforms, you can just publish robot's description and use camera's launch file arguments to mount it to the drone.

Ghada09, 2023-11-17:
can you explain in detail please

Serafadam, 2023-11-21:
Hi, I'm not sure what software your drone is running, but if it has a ROS driver, then it should output IMU messages after running it. It should also publish it's URDF on description, after you set the mount point of the camera you should be able to get those transforms. An example on how to "mount" the camera can be seen for example [here](https://github.com/Serafadam/interbotix_depthai_demo/blob/main/interbotix_depthai_demo/launch/bringup.launch.py)

slowrunner, 2023-12-23:
The “use another IMU” requires a ROS IMU node in your system that can directly access the drone’s IMU.  That is the complexity hidden in this.