luxonis/depthai-ros #456: [BUG] Resolution 720P not supported by sensor IMX378.

fwarmuth, 2023-11-13:
Hey,
i am try to setup a OAK-D Pro W (POE).
In the docs it stated (https://github.com/luxonis/depthai-ros/blob/humble/README.md/README.md?plain=305)
```
##### **OAK D PRO W**
To properly align with depth, you need to set `rgb.i_resolution` parameter to `720` (see `config/oak_d_w_pro.yaml`).
```
But during setup of the pipeline, i get:
```
Resolution 720P not supported by sensor IMX378. Using default resolution 1080P
```
Is this the correct intended behavior?
Are there any other ways to align depth with RGB on IMX378?

regards

Serafadam, 2023-11-13:
Hi, this was targeted towards OAK-D-PRO with OV9782 sensor, we will update the docs in the future, in case of IMX378 resolution change is not needed.

fwarmuth, 2023-11-13:
Thanks for the fast reply!

closed 2023-11-13 by fwarmuth