luxonis/depthai-ros #457: [Feature-Request] Rolling release?

MatthijsBurgh, 2023-11-15:
### Start with the `why`:
I am running ROS2 `rolling`. I need `moveit_py`, which is not on `Humble`. So therefore I need `rolling`, as I want to use U22. U23 is not desired as it isn't LTS.

### Move to the `what`:
A release on `rolling`

### Move to the `how`:
Release on `rolling` 😉

Serafadam, 2023-11-28:
Hi @MatthijsBurgh , at this moment we only support latest stable releases (Humble, Iron, Noetic). We could create a separate branch for rolling, but for now it would have to be community-supported.

MatthijsBurgh, 2023-11-28:
I don't see a reason to not release the current main branch, which you release to Iron, to release it to rolling. As that is kind of the idea of the rolling distro.

ruffsl, 2024-05-12:
> We could create a separate branch for rolling, but for now it would have to be community-supported.

@Serafadam , I would think there'd be a financial incentive to officially support rolling as well, given how it'd help users up to date with the latest supported LTS. The ROS 2 Jazzy release is open for public testing, yet I could not find this package available through the public build farm. It's rather painful when folks are held back from critical security platform upgrades and ROS bug fixes because a component of the stack has not been maintained for the latest LTS. If rolling was actively maintained, then binary package releases for latest LTS releases would be an automated by product, unblocking many of end users and business partners.

> at this moment we only support latest stable releases (Humble, Iron, Noetic)

Speaking of LTS releases, hows support for Jazzy coming along? I got this repo to compile locally, but not after patching a large number of  (deprecated?) OpenCV header paths, and fixing undeclared Cmake depacies for `ament_index_cpp`. Any ETA for Jazzy given it's release is almost upon us? I could open some drive by PRs, but this repo has no CONTRIBUTING file as a guide. Thanks!

G-Levine, 2024-05-19:
@ruffsl Do you have your changes documented anywhere? I'm currently trying to get it to build on rolling, and it'd be incredibly helpful to see how you did it.

ruffsl, 2024-05-24:
@G-Levine , I didn't notice at first that there was an Iron branch here at first, given the number of branches here and the default branch still being pinned to Humble. So the changes for building are even fewer:
- https://github.com/luxonis/depthai-ros/pull/533

Haven't tested my Jazzy build of this just yet, as I'm still just getting my CI rolled over to Ubuntu Noble.