luxonis/depthai-ros #460: calibration error

FPSychotic, 2023-11-22:
20.04, Noetic, OAK-d-pro-DEV, apt installation, trying calibrate the camera after sensor got unglued.

```
 roslaunch depthai_ros_driver calibration.launch 
... logging to /home/nrover/.ros/log/fa242cbc-8933-11ee-aba9-24ee9a5f388b/roslaunch-ubuntu-411254.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:33105/

SUMMARY
========

PARAMETERS
 * /oak/camera_i_nn_type: none
 * /oak/camera_i_pipeline_type: RGBDStereo
 * /oak/left_i_publish_topic: True
 * /oak/right_i_publish_topic: True
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    calibrator (camera_calibration/cameracalibrator.py)
    oak (nodelet/nodelet)
    oak_nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[oak_nodelet_manager-1]: started with pid [411275]
process[oak-2]: started with pid [411276]
process[calibrator-3]: started with pid [411277]
[ INFO] [1700661136.011894966]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1700661136.015587130]: waitForService: Service [/oak_nodelet_manager/load_nodelet] could not connect to host [ubuntu:46141], waiting...
[ INFO] [1700661136.084101950]: Initializing nodelet with 8 worker threads.
[ INFO] [1700661136.103713074]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[ INFO] [1700661136.310688675]: No ip/mxid/usb_id specified, connecting to the next available device.
[ INFO] [1700661138.979708533]: Camera with MXID: 14442C1071EB3FD700 and Name: 1.2.2 connected!
[ INFO] [1700661138.981372067]: USB SPEED: HIGH
[ INFO] [1700661139.002729542]: Device type: OAK-D-PRO-W
[ INFO] [1700661139.007340803]: Pipeline type: RGBDStereo
[ERROR] [1700661139.008777194]: Unable to find name RGBDSTEREO in map.
Map values:
CAMARRAY
DEPTH
STEREO
RGBSTEREO
RGBD
RGB

[FATAL] [1700661141.757061092]: Failed to load nodelet '/oak` of type `depthai_ros_driver/Camera` to manager `oak_nodelet_manager'
[oak_nodelet_manager-1] process has died [pid 411275, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager /nodelet_manager/load_nodelet:=oak/nodelet_manager/load_nodelet /nodelet_manager/unload_nodelet:=oak/nodelet_manager/unload_nodelet /nodelet_manager/list:=oak/nodelet_manager/list __name:=oak_nodelet_manager __log:=/home/nrover/.ros/log/fa242cbc-8933-11ee-aba9-24ee9a5f388b/oak_nodelet_manager-1.log].
log file: /home/nrover/.ros/log/fa242cbc-8933-11ee-aba9-24ee9a5f388b/oak_nodelet_manager-1*.log
================================================================================REQUIRED process [oak-2] has died!
process has died [pid 411276, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load depthai_ros_driver/Camera oak_nodelet_manager __name:=oak __log:=/home/nrover/.ros/log/fa242cbc-8933-11ee-aba9-24ee9a5f388b/oak-2.log].
log file: /home/nrover/.ros/log/fa242cbc-8933-11ee-aba9-24ee9a5f388b/oak-2*.log
Initiating shutdown!
================================================================================
[calibrator-3] killing on exit
[oak-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

Serafadam, 2023-11-23:
Hi, there is a minor typo [in the launch file](https://github.com/luxonis/depthai-ros/blob/c0d4f8111e05d05a8ab68082d15f39455c9088a5/depthai_ros_driver/launch/calibration.launch#L14C10-L14C10) (will be fixed soon), the correct value should be `rgbstereo`

FPSychotic, 2023-11-29:
> Contributor
Hi, I'm reopening the issue as after calibration , there is not calibration printed in the terminal and no .yalm file is saved. So what I do after calibration to have the calibration values printed, the yalm files or both?