luxonis/depthai-ros #463: [BUG] OAK D PRO POE can only get 720P. v2.8.2-humble

Victorsoeby, 2023-11-23:
Expected result: Able to change resolution to "12MP", "4K", "1080P"
Result: Can only get 720P
Branch: v2.8.2-humble-12-g48fd970.
Launchfile: camera.launch.py (depthai_ros_driver)

I am using this with a OAK D Pro POE camera with the IMX378 sensor, and can for the life of me not get it to output the RGB image in 1080P. I have cloned the V2.8.2, changed the camera.yaml file to have the following:

```
/oak:
  ros__parameters:
    camera:
      i_enable_imu: true
      i_enable_ir: true
      i_pipeline_type: RGBD
      i_nn_type: ''
    rgb: 
      i_resolution: '1080P'
```
When launching the camera.launch.py: 

```
root@agx-orin2:/workspaces# ros2 launch depthai_ros_driver camera.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-11-23-14-02-21-228962-agx-orin2-46251
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [46265]
[component_container-1] [INFO] [1700748142.526463430] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[component_container-1] [INFO] [1700748142.532228582] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1700748142.532365673] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1700748142.557751327] [oak_state_publisher]: got segment ctis_camera_link
[component_container-1] [INFO] [1700748142.557883842] [oak_state_publisher]: got segment oak
[component_container-1] [INFO] [1700748142.557931811] [oak_state_publisher]: got segment oak_imu_frame
[component_container-1] [INFO] [1700748142.557974340] [oak_state_publisher]: got segment oak_left_camera_frame
[component_container-1] [INFO] [1700748142.558015269] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[component_container-1] [INFO] [1700748142.558053254] [oak_state_publisher]: got segment oak_model_origin
[component_container-1] [INFO] [1700748142.558089990] [oak_state_publisher]: got segment oak_rgb_camera_frame
[component_container-1] [INFO] [1700748142.558125543] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[component_container-1] [INFO] [1700748142.558162888] [oak_state_publisher]: got segment oak_right_camera_frame
[component_container-1] [INFO] [1700748142.558199529] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak_state_publisher' in container 'oak_container'
[component_container-1] [INFO] [1700748142.562600043] [oak_container]: Load Library: /workspaces/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[component_container-1] [INFO] [1700748142.664661165] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1700748142.664867602] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1700748142.680244008] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [INFO] [1700748149.563710927] [oak]: Camera with MXID: 1844301071F88E0F00 and Name: 10.46.28.215 connected!
[component_container-1] [INFO] [1700748149.564009910] [oak]: PoE camera detected. Consider enabling low bandwidth for specific image topics (see readme).
[component_container-1] [INFO] [1700748149.588918081] [oak]: Device type: OAK-D-PRO-POE-FF
[component_container-1] [INFO] [1700748149.597741125] [oak]: Pipeline type: RGBD
[component_container-1] [INFO] [1700748149.883668521] [oak]: Finished setting up pipeline.
[component_container-1] [INFO] [1700748150.994838049] [oak]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
```
All seems to go well, but the image that is published is 1280x720.
Here is me doing a ros2 topic echo /oak/rgb/image_raw
[image]

When i try to change the resolution to 4MP in the camera.yaml config file i get this: 

root@agx-orin2:/workspaces# ros2 launch depthai_ros_driver camera.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-11-23-14-14-20-489431-agx-orin2-50833
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [50847]
[component_container-1] [INFO] [1700748861.767367651] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[component_container-1] [INFO] [1700748861.771652002] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1700748861.771843334] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1700748861.788308182] [oak_state_publisher]: got segment oak
[component_container-1] [INFO] [1700748861.788446137] [oak_state_publisher]: got segment oak-d-base-frame
[component_container-1] [INFO] [1700748861.788497114] [oak_state_publisher]: got segment oak_imu_frame
[component_container-1] [INFO] [1700748861.788539579] [oak_state_publisher]: got segment oak_left_camera_frame
[component_container-1] [INFO] [1700748861.788579388] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[component_container-1] [INFO] [1700748861.788617853] [oak_state_publisher]: got segment oak_model_origin
[... 144 lines cut ...]
        DeclareLaunchArgument("camera_model", default_value="OAK-D-Pro"),
        DeclareLaunchArgument("cam_pos_x", default_value="0.0"),
        DeclareLaunchArgument("cam_pos_y", default_value="0.0"),
        DeclareLaunchArgument("cam_pos_z", default_value="0.0"),
        DeclareLaunchArgument("cam_roll", default_value="0.0"),
        DeclareLaunchArgument("cam_pitch", default_value="0.0"),
        DeclareLaunchArgument("cam_yaw", default_value="0.0"),
        DeclareLaunchArgument("params_file", default_value=os.path.join(depthai_prefix, 'config', 'camera.yaml')),
        DeclareLaunchArgument("use_rviz", default_value='false'),
        DeclareLaunchArgument("rviz_config", default_value=os.path.join(depthai_prefix, "config", "rviz", "rgbd.rviz")),
        DeclareLaunchArgument("rsp_use_composition", default_value='true'),
        DeclareLaunchArgument("pass_tf_args_as_params", default_value='false', description='Enables TF publishing from camera calibration file.'),
        DeclareLaunchArgument("imu_from_descr", default_value='false', description='Enables IMU publishing from URDF.'),
        DeclareLaunchArgument("override_cam_model", default_value='false', description='Overrides camera model from calibration file.'),
        DeclareLaunchArgument("use_gdb", default_value='false'),
        DeclareLaunchArgument("use_valgrind", default_value='false'),
        DeclareLaunchArgument("use_perf", default_value='false')
    ]

    return LaunchDescription(
        declared_arguments + [OpaqueFunction(function=launch_setup)]
    )

```
I know it is not hardware related, as the drivers have been working fine when i used the Humble 2.7.1 branch earlier. Reason i updated was that i found a bug where the timestamps of the images were wrong, which seems to have been fixed in this version.

Serafadam, 2023-11-23:
Hi, by default [ISP output is scaled when running with default parameters](https://github.com/luxonis/depthai-ros#setting-rgb-parameters), you can set `rgb.i_set_isp_scale: false` to turn that off.

RMonica, 2024-06-07:
Hi,

I ran into this problem yesterday. It was confusing also because the camera info changes resolution according to `i_height` and `i_width`, but the actual image doesn't.

The link is broken, by the way.

Serafadam, 2024-06-07:
Hi, the output image depends on DAI node output type, you can have `isp` or `video`, by default it is ISP which will get scaled, setting i_width and i_height sets video outputs size. Updated link [here](https://docs.luxonis.com/software/ros/depthai-ros/driver/#DepthAI%20ROS%20Driver-Setting%20RGB%20parameters)