luxonis/depthai-ros #464: Publish compressed color image

metanav, 2023-11-26:
I am using the example `yolov4_spatial_publisher.cpp `.  How can we publish compressed  (encoded jpeg) image on the /color/image topic?

GustavBroend, 2023-12-22:
I'm working in ROS Noetic, and for me the JPEG compression happens automatically - you can see the code attached. I assume the publisher setup uses the image_transport somehow?
I also have a request/question regarding the topic of compressed images. I would like to have a lossless datastream from the camera to ROS, so either PNG or lossless JPEG. 

When I run "rosparam list" with my node, I get these parameters:

/oak/left/image_raw/compressed/format
/oak/left/image_raw/compressed/jpeg_optimize
/oak/left/image_raw/compressed/jpeg_progressive
/oak/left/image_raw/compressed/jpeg_quality
/oak/left/image_raw/compressed/jpeg_restart_interval
/oak/left/image_raw/compressed/png_level
/oak/left/image_raw/compressedDepth/depth_max
/oak/left/image_raw/compressedDepth/depth_quantization
/oak/left/image_raw/compressedDepth/format
/oak/left/image_raw/compressedDepth/png_level

/oak/right/image_raw/compressed/format
/oak/right/image_raw/compressed/jpeg_optimize
/oak/right/image_raw/compressed/jpeg_progressive
/oak/right/image_raw/compressed/jpeg_quality
/oak/right/image_raw/compressed/jpeg_restart_interval
/oak/right/image_raw/compressed/png_level

However, when i try to set the format to "png" or change the jpeg quality, nothing changes on the image output. I have another node which saves the compressed images directly to my disk, so I can tell whether the ouptut formatting has changed. Is there a way to modify these values? Does the JPEG formatting come from the OAK camera or my own system/ROS? And is there a way to configure it?

Best Regards,
Gustav

[stereo_inertial_publisher.zip](https://github.com/luxonis/depthai-ros/files/13752809/stereo_inertial_publisher.zip)

Serafadam, 2023-12-28:
Hi, compressed topics are published with the use of [image_transport](https://wiki.ros.org/image_transport). According to the documentation, you should be able to change the format on the fly, although if that is not the case, maybe setting format parameter in node params file or resetting the camera can help here.

GustavBroend, 2024-02-14:
Thanks for the insight. I tried to set the parameters in my c++ code, but with no luck. I managed to change the formats by manually compiling the "image_transport_plugins" library in catkin_ws and changing defaults in the file "CompressedPublisher.cfg". The image_transport png compression is very slow however, so perhaps i will be better off using the lossless mjpeg or other encoded frames that i see the Depthai c++ api also supports.