luxonis/depthai-ros #466: Camera not detected on socket:0 Error when running stereo_inertial_node.launch using OAK-D-SR.

Wayne-sketch, 2023-11-28:
process[oak_state_publisher-1]: started with pid [64176]
process[stereo_inertial_publisher-2]: started with pid [64177]
process[nodelet_manager-3]: started with pid [64178]
process[depth_image_convertion_nodelet-4]: started with pid [64179]
process[depth_image_to_rgb_pointcloud-5]: started with pid [64184]
process[rviz-6]: started with pid [64187]
[ INFO] [1701150803.552768848]: Initializing nodelet with 8 worker threads.
 nnPath ,, /home/roma/dai_ws/src/depthai-ros/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
1280 720 1280 720
**Listing available devices...
Device Mx ID: 19443010315B782700
[19443010315B782700] [3.5.2] [1.769] [ColorCamera(7)] [error] Camera not detected on socket: 0
terminate called after throwing an instance of 'std::runtime_error'
  what():  IMU(5) - IMU invalid settings!: IMU not detected. Your board doesn't have IMU or requires firmware update (Set imu.enableFirmwareUpdate(True) explicitly.). 1, 321**
================================================================================REQUIRED process [stereo_inertial_publisher-2] has died!
process has died [pid 64177, exit code -6, cmd /home/roma/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/roma/.ros/log/d36ee638-8d12-11ee-93f2-69a72f819408/stereo_inertial_publisher-2.log].
log file: /home/roma/.ros/log/d36ee638-8d12-11ee-93f2-69a72f819408/stereo_inertial_publisher-2*.log
Initiating shutdown!
================================================================================
[rviz-6] killing on exit
[depth_image_to_rgb_pointcloud-5] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
[nodelet_manager-3] killing on exit
[stereo_inertial_publisher-2] killing on exit
[oak_state_publisher-1] killing on exit
QFileSystemWatcher::removePaths: list is empty
QFileSystemWatcher::removePaths: list is empty
shutting down processing monitor...
... shutting down processing monitor complete
done

Wayne-sketch, 2023-11-29:
So OAK-D-SR doesn't have an IMU sensor, does it?

Serafadam, 2023-12-11:
Hi @Wayne-sketch, I think that depends on when it was bought [according to the store page](https://shop.luxonis.com/collections/oak-cameras-1/products/oak-d-sr), do you have order information?

heebahsaleem, 2024-01-23:
@Serafadam 
I am using OAK-1 and I am also getting the same error:

`[ WARN] [1705985304.054897606]: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/GLX/OgreGLXContext.cpp (line 61)
rviz::RenderSystem: error creating render window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/GLX/OgreGLXContext.cpp (line 61)
[ERROR] [1705985304.054922172]: Unable to create the rendering window after 100 tries.
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Listing available devices...
Device Mx ID: 184430107165680F00
[rviz-7] process has died [pid 52467, exit code -6, cmd /opt/ros/noetic/lib/rviz/rviz -d /home/sonnet/dai_ws/src/depthai-ros/depthai_examples/rviz/stereoInertialDepthAlign.rviz __name:=rviz __log:=/home/sonnet/.ros/log/a32ec8ac-b9aa-11ee-8ab4-ed2d9ffad481/rviz-7.log].
log file: /home/sonnet/.ros/log/a32ec8ac-b9aa-11ee-8ab4-ed2d9ffad481/rviz-7*.log
[184430107165680F00] [3.3] [1.571] [MonoCamera(2)] [error] Camera not detected on socket: 2
[184430107165680F00] [3.3] [1.571] [MonoCamera(1)] [error] Camera not detected on socket: 1
terminate called after throwing an instance of 'std::runtime_error'
  what():  IMU(5) - IMU invalid settings!: IMU not detected. Your board doesn't have IMU or requires firmware update (Set imu.enableFirmwareUpdate(True) explicitly.). 1, 321
================================================================================REQUIRED process [stereo_inertial_publisher-3] has died!
process has died [pid 52457, exit code -6, cmd /home/sonnet/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/sonnet/.ros/log/a32ec8ac-b9aa-11ee-8ab4-ed2d9ffad481/stereo_inertial_publisher-3.log].
log file: /home/sonnet/.ros/log/a32ec8ac-b9aa-11ee-8ab4-ed2d9ffad481/stereo_inertial_publisher-3*.log
Initiating shutdown!
================================================================================
[depth_image_convertion_nodelet-5] killing on exit
[depth_image_to_rgb_pointcloud-6] killing on exit
[nodelet_manager-4] killing on exit
[stereo_inertial_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`