luxonis/depthai-ros #470: OAK-D-PRO-W Camera Data Stream Issues in ROS Environment

RP-ND, 2023-12-06:
I am experiencing a recurrent issue with the OAK-D-PRO-W camera while operating it within a ROS environment. This problem has persisted for the last few days, specifically manifesting when I attempt to launch the camera using camera.launch. The core of the issue lies in the absence of a data stream from the camera in the ROS setup.

Notably, the camera functions as intended in non-ROS environments, which suggests that the hardware is not at fault. This discrepancy leads me to believe that the issue might be related to the ROS integration or configuration.

I have been unable to locate any existing documentation, issues, or discussions that address or resolve this particular problem. If I have missed something, I would be glad to be pointed to it.

PARAMETERS
 * /oak/camera_i_base_frame: oak-d_frame
 * /oak/camera_i_cam_pitch: 0.0
 * /oak/camera_i_cam_pos_x: 0.0
 * /oak/camera_i_cam_pos_y: 0.0
 * /oak/camera_i_cam_pos_z: 0.0
 * /oak/camera_i_cam_roll: 0.0
 * /oak/camera_i_cam_yaw: 0.0
 * /oak/camera_i_enable_imu: True
 * /oak/camera_i_enable_ir: True
 * /oak/camera_i_floodlight_brightness: 0.0
 * /oak/camera_i_imu_from_descr: False
 * /oak/camera_i_ip:
 * /oak/camera_i_laser_dot_brightness: 800.0
 * /oak/camera_i_mx_id: 
 * /oak/camera_i_nn_type: spatial
 * /oak/camera_i_parent_frame: oak-d-base-frame
 * /oak/camera_i_pipeline_type: rgbd
 * /oak/camera_i_publish_tf_from_calibration: False
 * /oak/camera_i_usb_port_id: 
 * /oak/camera_i_usb_speed: SUPER_PLUS
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 
NODES
  /
    oak (nodelet/nodelet)
    oak_nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
 
ROS_MASTER_URI=http://localhost:11311
 
process[oak_state_publisher-1]: started with pid [17254]
process[oak_nodelet_manager-2]: started with pid [17255]
process[oak-3]: started with pid [17256]
[ INFO] [1701905531.742241699]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1701905531.750574315]: waitForService: Service [/oak_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1701905531.765855955]: Initializing nodelet with 6 worker threads.
[ INFO] [1701905531.772846966]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[ INFO] [1701905532.155187183]: No ip/mxid/usb_id specified, connecting to the next available device.
[ INFO] [1701905539.676089107]: Camera with MXID: 18443010810A970F00 and Name: 169.254.1.222 connected!
[ INFO] [1701905539.676312473]: PoE camera detected. Consider enabling low bandwidth for specific image topics (see readme).
[ INFO] [1701905539.703325316]: Device type: OAK-D-PRO-W-POE-97
[ INFO] [1701905539.716544946]: Pipeline type: rgbd
[ INFO] [1701905540.010408174]: NN Family: mobilenet
[ INFO] [1701905540.116766056]: NN input size: 300 x 300. Resizing input image in case of different dimensions.
[ INFO] [1701905540.576985780]: Finished setting up pipeline.
[18443010810A970F00] [169.254.1.222] [7.466] [StereoDepth(8)] [error] Custom disparity/depth width must be multiple of 16.
[ INFO] [1701905542.226487428]: Camera ready!
[ERROR] [1701905552.217410249]: Camera diagnostics error: No Data
[ INFO] [1701905552.217533933]: Restarting camera
[ INFO] [1701905552.217609135]: Stopping camera.
[18443010810A970F00] [169.254.1.222] [1701905555.657] [host] [warning] Monitor thread (device: 18443010810A970F00 [169.254.1.222]) - ping was missed, closing the device connection
[ INFO] [1701905557.698913315]: Starting camera.
[ INFO] [1701905557.701144736]: No ip/mxid/usb_id specified, connecting to the next available device.
[ INFO] [1701905564.586619342]: Camera with MXID: 18443010810A970F00 and Name: 169.254.1.222 connected!
[ INFO] [1701905564.586758961]: PoE camera detected. Consider enabling low bandwidth for specific image topics (see readme).
[ INFO] [1701905564.613161537]: Device type: OAK-D-PRO-W-POE-97
[ INFO] [1701905564.621372993]: Pipeline type: rgbd
[ INFO] [1701905564.792966853]: NN Family: mobilenet
[ INFO] [1701905564.894732859]: NN input size: 300 x 300. Resizing input image in case of different dimensions.
[ INFO] [1701905565.235650275]: Finished setting up pipeline.
[18443010810A970F00] [169.254.1.222] [6.769] [StereoDepth(8)] [error] Custom disparity/depth width must be multiple of 16.
[ INFO] [1701905566.926362394]: Camera ready!

Serafadam, 2023-12-07:
Hi, could you post more information according to the[ bug template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)? I can see that the camera restarts due to lack of diagnostic data, are other streams stopping here as well, or after the restart? Regarding similar issues, [there was one a bit similar](https://github.com/luxonis/depthai-ros/issues/352) that was related to ROS master setup.

yh725k, 2024-01-16:
Hi, Did you found the solution?  I am facing the same issue here when i run to a 'camera.launch.py'.

yh725k, 2024-01-24:
Hi @Serafadam, I already solved the problem. thank you. 
But I have more questions. I would like to use sorted images in ROS2(humble) like rgb_undistort.py(https://github.com/luxonis/depthai-python/blob/main/examples/ColorCamera/rgb_undistort.py). Is there any guidance regarding this?

RP-ND, 2024-01-24:
Sincere apologies for a delayed response, University was on holiday. @yh725k you said you solved it? Could you for propensity's sake (and for everyone else who may have this issue in the future) describe what you did to solve it. Did you simply use the new branch?

@Serafadam I will try and post according to the bug template after I try a few more things, especially the new branch.

RP-ND, 2024-01-28:
@yh725k No it didn't, but it did lead me on the right path.

@Serafadam Building from the newest source worked perfectly, thanks for the help!

closed 2024-01-28 by RP-ND