luxonis/depthai-ros #471: [custom] ROS 1 Noetic Support for OAK-D LR

TFM99102, 2023-12-13:
Is the OAK-D LR currently supported by the ROS code? 
I have followed the readme file but it does not work for the OAK-D LR. 
If there is currently no support, are there any plans to support the device in the future? 

Kind regards

Serafadam, 2023-12-18:
Hi, could you post some more information as described in[ bug report template?](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)

TFM99102, 2023-12-20:
Hey, I'm trying to work with the OAK-D Long range: 
[Product link:](https://shop.luxonis.com/products/oak-d-lr)

I have installed everything according to the readme. 
I'm trying to launch the stereo_inertial_node 
`roslaunch depthai_examples stereo_inertial_node.launch camera_model:=OAK-D-LR`
For the camera_model i want to set OAK-D-LR but this is failing ( I have looked into the URDF files and there is no file for my camera model) --> that is why I'm thinking the model is currently not supported. 
**Error message when launching with camera_model:=OAK-D-LR**
`... logging to /home/urban/.ros/log/fce0fd42-9f48-11ee-9ef7-48b02de60df0/roslaunch-urban0-5334.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://urban0:41375/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/detectionClassesCount: 80
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableRosBaseTimeUpdate: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/expTime: 20000
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/manualExposure: False
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/nnName: x
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/previewHeight: 416
 * /stereo_inertial_publisher/previewWidth: 416
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/resourceBaseFolder: /opt/ros/noetic/s...
 * /stereo_inertial_publisher/rgbResolution: 1080p
 * /stereo_inertial_publisher/rgbScaleDinominator: 3
 * /stereo_inertial_publisher/rgbScaleNumerator: 2
 * /stereo_inertial_publisher/sensIso: 800
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

auto-starting new master
process[master]: started with pid [5368]
[... 110 lines cut ...]
1280 720 1280 720
Listing available devices...
terminate called after throwing an instance of 'std::runtime_error'
  what():  ros::NodeHandle() from Node "/stereo_inertial_publisher" DepthAI Device with MxId  "" not found.  "
================================================================================REQUIRED process [stereo_inertial_publisher-3] has died!
process has died [pid 6025, exit code -6, cmd /opt/ros/noetic/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/urban/.ros/log/eb9b0418-9f4a-11ee-8a5d-48b02de60df0/stereo_inertial_publisher-3.log].
log file: /home/urban/.ros/log/eb9b0418-9f4a-11ee-8a5d-48b02de60df0/stereo_inertial_publisher-3*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[depth_image_to_rgb_pointcloud-6] killing on exit
[depth_image_convertion_nodelet-5] killing on exit
[nodelet_manager-4] killing on exit
[stereo_inertial_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`
**Attach system log**
- ROS noetic 
- ubuntu 20.04 
- apt install

TFM99102, 2024-01-10:
Hey any updates on this ?

Serafadam, 2024-01-15:
Hi @TFM99102 , sorry for the inconvenience. LR support will be updated this week (hopefully today/tomorrow) by @danilo-pejovic

TFM99102, 2024-01-22:
Thanks, I look forward to the issue being resolved.

danilo-pejovic, 2024-01-22:
I am working on it, we are updating couple of our cameras. I will let you know when I am ready to push changes for LR.