luxonis/depthai-ros #474: [BUG] Dropouts in IMU data publishing on OAK-D PRO W

petrlmat, 2023-12-21:
**Issue description**
We are trying to run a VIO algorithm in ROS using the stereo IR images from OAK-D PRO W with mixed success. While analyzing the IMU data, we have found out that the data is published quite sporadically in batches. This is not ideal for VIO, which preintegrates IMU data between image frames to provide a stable high-frequency stream of state estimation. I'm not sure whether the problem is in the way the IMU messages are published to ROS or if the problem is in the reading of data from the sensor itself. 

I will try to better explain the issue on these plots:
[image]
The OAK-D IMU data in this plot arrives in batches with periods in the tens of ms without any data. Notice the other IMU in the plot, which publishes evenly spaced data points on 100 Hz (OAK-D IMU runs on 400 Hz in this plot). That would be the desired behavior. Can it be achieved or is there something preventing getting evenly-spaced IMU data points from OAK-D?

Let's see the problem in `rqt_bag`:
[image]
The bins are 0.005s long, so each bin should contain 1 IMU message (IMU running at 200 Hz) but there seem to be a window without any message for over ~30 ms.

**Solutions attempted**
We tried various settings of the IMU parameters to improve this behavior but so far have been unsuccessful in eliminating the problem entirely. The parameter having the biggest effect is the `batch_report_threshold` and potentially `max_batch_reports`. If I understand the parameters correctly, OAK waits until `batch_report_threshold` data points (IMU and camera images combined) before sending over USB and this batch can only have `max_batch_reports` IMU data points. So unless `max_batch_reports < batch_report_threshold`  the `max_batch_reports` parameter should have no effect right? I included both parameter values in case I understand it wrong. Let's see how the parameters influence the consistency of IMU data frequency. The plots show the ROS time of message arrival on the X-axis and the message time stamp (from its header) on the Y-axis.

`batch_report_threshold = 5; max_batch_reports = 10 (default)`
[image]
I see periods of ~40 ms without IMU data. Also, the periodicity of the dropout is suspicious. Could something else be blocking the sending of IMU data?

`batch_report_threshold = 20; max_batch_reports = 20`
[image]
Much worse but expected. OAK waits until 20 data points are available before sending. I included this test to show that changing the parameters in this direction won't help. Also, I saw somewhere in code comment that this setting might help in high-throughput applications, so I wanted to try it.

`batch_report_threshold = 1; max_batch_reports = 10`
[image]
This is the best performance achieved so far. The messages are sent much more often but still, there are ~40ms dropouts. Seems that the dropout has a ~100ms period. _Could the communication be blocked by sending camera images at these moments?_ I don't have access to the camera today but I will check if this happens also when running the IMU only.

The dropouts are periodic and seem to happen when images from all cameras arrive at the same time so indeed it seems that the communication is blocked by the images. _Can IMU messages be somehow prioritized?_
[image]

**System info**
- OAK-D PRO W (IMU type: BNO086, firmware version: 3.9.9, embedded firmware version: 3.9.9)
- Recent generation of Intel NUC i7
- Ubuntu 20.04
- ROS Noetic
- `depthai-ros` build in a ros workspace with`-O2` optimization
- `depthai/depthai-core` installed from rosdep
- This is all I can provide ATM (no access to OAK-D today) but happy to provide more information tomorrow upon request.

Serafadam, 2024-01-30:
Hi, sorry for the delay and thanks for the detailed reports, we are currently discussing these issues internally and I will get back to you with the conclusions hopefully this/next week.

petrlmat, 2024-04-08:
Any updates on this issue?

themarpe, 2024-04-17:
@petrlmat 

Hi - just one thing to cross check, are you referring to When the reports arrive or What timestamp they hold? 
Due to comms channel we cannot have eg "realtime" IMU updates, but batched IMU readings should still hold correct individual timestamp per report.
Those reports should be feed into the system with the timestamp component

petrlmat, 2024-04-18:
Hi, I am referring to "When the reports arrive". Realtime IMU updates are necessary for running realtime state estimation algorithms. ~40 ms long period without data is just too much for state estimation of unstable systems such as drones. We would like to use your camera as a "do it all" camera on our drones but due to this issue we have to rely on our custom solution with realtime IMU. 
How can the time between batches be influenced? Would e.g. sending less camera streams through the comm channel increase the batch frequency?

themarpe, 2024-04-18:
@petrlmat 

>  How can the time between batches be influenced? Would e.g. sending less camera streams through the comm channel increase the batch frequency?

Yes, this is likely the biggest factor. You may quickly check it by *only* sending the IMU updates instead of camera stream as well

petrlmat, 2024-10-02:
Any updates on this issue? Is there a plan or possibility to obtain the IMU measurements ASAP instead of getting them in batches or is it infeasible due to hardware limitations?

klaxalk, 2025-09-15:
Hello, Is the problem still occurring? We are considering ordering several POE cameras for our drones. Having fresh and regular IMU data is crucial for realtime state estimation. If would be so good to have a camera with IMU in a single package like this...

But with this issue, we could not use these...