luxonis/depthai-ros #479: [BUG] {Streams have duplicate name 'detections'}

FrederikeDurow, 2024-01-17:
**Describe the bug**
I'd like to use my OAK camera with ros2 for yolo detection, so I tried to run the yolo example launch file. I get the error that Streams have duplicate name 'detections'. 

I installed from ros binaries on humble, and am working with ubuntu 22.04. 

The following work without any issues:
- `stereo_inertial_node` in `depthai_examples`
- `camera` in `depthai_ros_driver`

[image]

slowrunner, 2024-01-21:
I get the same error 
```
[yolov4_spatial_node-2]   what():  Streams have duplicate name 'detections'
```
launching in the luxonis/depthai-ros:humble-latest Docker image

slowrunner, 2024-02-03:
@FrederikeDurow Have you tried the latest build from sources humble branch?  

`ros2 launch depthai_examples yolov4_publisher.launch.py camera_model:=OAK-D-LITE spatial_camera:=false  `

is working for me after building from the humble sources (over Ubuntu 22.04 in Docker on Raspberry Pi 5)

Rodhaaret, 2024-02-06:
It work now! It seems like the newer version of the package solved the issue.

FrederikeDurow, 2024-02-06:
Yes, it works with the newest build from source, thanks!

closed 2024-02-06 by FrederikeDurow