luxonis/depthai-ros #483: [BUG] {There is no orientation when using imu related topics}

abloomyt, 2024-01-24:
Problem description:
My camera's type is oak-d pro, and my depthai-ros's version is Release v2.8.2 Noetic. I want to use imu topic to produce orientation. Whether I use /stereo_inertial_publisher/imu in depthai_examples/launch/stereo_inertial_node.launch, or I use /oak/imu/data in depthai_ros_driver/launch/pointcloud.launch, I can't generate orientation, but can produce angular velocity and linear velocity.
Just like the picture below:
[image]
I don't know what went wrong.I would appreciate it if you could give me an answer.

Serafadam, 2024-01-24:
Hi, you can use depthai_ros_driver with specific IMU parameters to enable rotation, stereo_inertial_node serves as a generic example since not all IMU sensors can have the rotation part enabled. More on IMU configuration [here](https://docs-beta.luxonis.com/develop/ros/depthai-ros-driver/#Depthai%20ROS%20Driver-Setting%20IMU%20parameters)

abloomyt, 2024-01-25:
I really appreciate your help. My problem has been solved. Can I ask you a question again, why is the covariance of rotation set to -1? What is the meaning of -1?

Serafadam, 2024-01-26:
Hi, unfortunately the datasheets are not explicit about covariance but since the orientation is a fused estimate I don't think it matters in that case. More information on the IMU sensors can be found [here](https://docs.luxonis.com/projects/api/en/latest/components/nodes/imu/).

abloomyt, 2024-01-26:
Thank you very much for your reply.

closed 2024-01-26 by abloomyt