luxonis/depthai-ros #487: [Feature-Request] {ROS2 Humble support for OAK D SR PoE}

spalande2023, 2024-01-25:
I have got the OAK D SR PoE camera and the RGB and depth data quality seems to be very good. I believe there is no ROS2 support for this model yet. Can you please add ROS2 support for that since our application is highly dependent on ROS2. Also, please let me know the timeline for this development.

Erol444, 2024-01-26:
@Serafadam , if this isn't yet supported, could we provide some guesstimation, so the client can better plan their project?
Thanks!

Serafadam, 2024-01-26:
Hi @spalande2023, SR PoE is not officially supported yet but you can test [it on this branch](https://github.com/luxonis/depthai-ros/tree/tof_support_humble), the one change you currently need to make to enable it is to set parameter 
`camera.i_pipeline_type: StereoToF`.

spalande2023, 2024-01-29:
Hello @Serafadam ,
As suggested, I changed the camera.i_pipeline_type to StereoToF and launched the camera using camera.launch.py file.
I am getting the stereo image topics and the tof image topic so it seems to work fine. 

However, when I launch the camera using sr_rgbd_pcl.launch.py to visualize the point cloud data and rgb image, there is a grayscale image published on the rgb color image topic (oak/right/image_raw). 
[image]

When I visualize the point cloud data in RViz, it shows a weird point cloud. Any reason why this is happening? 
[image]

Please let me know the nodes that I need to launch to get all the rgb, depth and point cloud data topics in ROS2. Is there a single launch file for this yet?

Serafadam, 2024-02-13:
Hi, sorry for the delay, we pushed some small modifications which should allow publishing rgb data.

dmn-seasony, 2024-03-15:
Hi @Serafadam, like @spalande2023 I'm having problems with the point cloud. The depth and RGB images look ok but the point cloud is all wrong:
[image]

I'm also having issues with parameters. If I call `ros2 param list /oak` I get an exception and the camera node prints the following:
```
[component_container-2] [WARN] [1710514197.384784851] [oak.rclcpp]: failed to send response to /oak/list_parameters (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314
```

dmn-seasony, 2024-03-19:
@Serafadam So the sr_rgbd_pcl launch file was generating the point cloud using the stereo image instead of the ToF image. The stereo image seems to be completely broken (RViz crashes when just trying to view it). I anyway don't plan to use stereo, so I switched it to the ToF depth image topic. But before I can do that, the depth image needs to be registered to the RGB image, but it looks like there is no TF frame defined for the ToF camera. The depth image has frame_id `oak_rgb_camera_optical_frame`, which doesn't seem correct and no transform is published for this frame by default. It is published if I set `camera.i_publish_tf_from_calibration: true`, but the published transform is just identical to the base frame. Can someone help me with this? And is it possible to do the depth registration (and potentially the point cloud transform) on the camera's onboard processor?

I'm also still having the issue with parameters mentioned above.