luxonis/depthai-ros #489: OAK-D LR [error] Maximum supported input image width for stereo is 1280 (Release v2.9.0 Noetic)

hangcui1201, 2024-01-30:
Hi,

I am using **Release v2.9.0 Noetic** with OAK-D LR with USB3 cable (Later I will switch to PoE cable)

When I run the program, for example, "roslaunch depthai_examples stereo_inertial_node.launch camera_model:=OAK-D-LR"

I always showed me the errors as below. I could not get the rgb image or depth image from ROS topics.

[14442C103167C7D600] [1.9] [1.113] [MonoCamera(1)] [error] AR0234 only supports THE_1200_P resolution, defaulting to it
[14442C103167C7D600] [1.9] [1.113] [MonoCamera(0)] [error] AR0234 only supports THE_1200_P resolution, defaulting to it
[14442C103167C7D600] [1.9] [1.437] [StereoDepth(4)] [error] Maximum supported input image width for stereo is 1280. Skipping frame!
[14442C103167C7D600] [1.9] [1.457] [StereoDepth(4)] [error] Maximum supported input image width for stereo is 1280. Skipping frame!
[14442C103167C7D600] [1.9] [1.478] [StereoDepth(4)] [error] Maximum supported input image width for stereo is 1280. Skipping frame!

Any suggestions on above issues?

Thanks!

Best,
Hang

Serafadam, 2024-01-31:
Hi, for OAK-D-LR I would suggest using `depthai_ros_driver` as `stereo_inertial_node` is just an example developed mostly for OAK-D-PRO/Lite cameras and not an actual driver, if you have issues with that please let us know.

hangcui1201, 2024-01-31:
@Serafadam Hi Serafadam, Thanks a lot! I have tried the depthai_ros_driver and have several questions.

(1) **The PoE did not work**. Only one PoE cable was connected to the OAK-D LR, no USB cable. My computer's IP was  169.254.1.10, the OAK-D LR's IP was 169.254.1.222. On my computer, I could ping 169.254.1.222. Then, I run 
"roslaunch depthai_ros_driver camera.launch camera_model:=OAK-D-LR" which uses a config file called "camera.yaml". I had configurations as below in camera.yaml.
/oak:
  camera_i_enable_imu: true
  camera_i_enable_ir: false
  camera_i_floodlight_brightness: 0.0
  camera_i_ip: '169.254.1.222'
  camera_i_laser_dot_brightness: 800.0
  camera_i_mx_id: '14442C103167C7D600'
  camera_i_nn_type: spatial
  camera_i_pipeline_type: rgbd
  camera_i_usb_port_id: ''
  camera_i_usb_speed: SUPER_PLUS
  nn_i_nn_config_path: depthai_ros_driver/mobilenet

It showed **No devices detected!**

ROS_MASTER_URI=http://localhost:11311

setting /run_id to ea00fff4-c060-11ee-bbcd-33b97493a404
process[rosout-1]: started with pid [265589]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [265596]
process[oak_nodelet_manager-3]: started with pid [265597]
process[oak-4]: started with pid [265598]
[ INFO] [1706723297.581947909]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1706723297.582917717]: waitForService: Service [/oak_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1706723297.588556843]: Initializing nodelet with 20 worker threads.
[ INFO] [1706723297.603908115]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[ERROR] [1706723298.189500596]: No devices detected!
[ERROR] [1706723299.197319005]: No devices detected!
^C[ERROR] [1706723300.193395356]: No devices detected!
[oak-4] killing on exit
[oak_nodelet_manager-3] killing on exit
[oak_state_publisher-2] killing on exit

**So, how to do the PoE setup for OAK-D LR using the depthai_ros_driver ?**

(2) Since the PoE did not work for me. I tried the USB cable. Now, only one USB cable was connected to OAK-D LR, no PoE cable. Then, I run 
"roslaunch depthai_ros_driver camera.launch camera_model:=OAK-D-LR"  with the same configurations in camera.yaml file as above. This time it can recognized the OAK-D LR camera. I could get one rgb image and one depth image. However, both of them were 768x480, too small. **How to get a larger rgb and depth images, since OAK-D LR has resolution up to 1920 x 1200?**

(3) OAK-D LR has three cameras. With depthai-viewer, I could set the **center camera** to output rgb images, and depth images were generated by the **left and right cameras**. **How to do such configurations with the ROS interface?** 

Thank you very much for the help!

Best,
Hang

hangcui1201, 2024-02-04:
Hello, Just double check any suggestions on above issues of depthai_ros_driver. Thanks a lot for the help!

Serafadam, 2024-02-05:
Hi, sorry for the delay:
1) Are you able to connect to the camera:
 - without specifying IP?
 - Using a pure-python example like the [demo program](https://github.com/luxonis/depthai)?
 2) By default RGB images are ISP frames downscaled for performance/compatibility reasons, currently maximum stereo resolution is 1280x800 but that could change in near future. You can disable ISP rescale and set video size explicitly via parameters, some more information here:
 - https://docs-beta.luxonis.com/develop/ros/depthai-ros-driver/#Depthai%20ROS%20Driver-Setting%20RGB%20parameters
 - https://docs.luxonis.com/projects/api/en/latest/components/nodes/color_camera/
 3) You can enable streams via ROS parameters (`i_publish_topic`) and[ customize stereo inputs](https://docs-beta.luxonis.com/develop/ros/depthai-ros-driver/#Depthai%20ROS%20Driver-Setting%20Stereo%20parameters). By default stereo is aligned to the `center`/`CAM_A` camera socket.
 4) Well, that depends on the use-case for PoE cameras the bandwidth can be an issue, here is some more information [on bandwidth](https://docs.luxonis.com/projects/api/en/latest/tutorials/low-latency/), regarding the driver outside of reducing FPS/Resolution you can also enable `low_bandwidth_mode` for streams which changes streams to output mjpeg frames which are converted on host to regular messages, although for stereo that option cannot be used with subpixel mode.

hangcui1201, 2024-02-05:
Hi Serafadam,

If the PoE option has potential low bandwidth, we think we will keep using the USB cable.

For above (2): **It shows me the error as below.** Even if I disable the i_output_isp, it does not change the image size.

[image]

What I did was modify the rgbd.yaml file as below and launch "roslaunch depthai_ros_driver rgbd_pcl.launch camera_model:=OAK-D-LR"

/oak:
    camera_i_nn_type: none
    camera_i_mx_id: '14442C103167C7D600'
    right_i_publish_topic: true
    right_i_output_isp: false
    right_i_width: 1200
    right_i_height: 800
    left_i_publish_topic: true
    left_i_output_isp: false
    left_i_width: 1200
    left_i_height: 800
    stereo_i_align_depth: true
    stereo_i_subpixel: true

For above (3)

I don't quite understand that. Do you mean the depth image was generated by the left and right cameras, then align with the center camera?

Can we launch all three cameras with 1200 x 800, depth images generated by the left and right camera, rgb images output by the center camera? How to do the settings in the yaml file?

Thanks a lot for the time and help!

Best,
Hang

Serafadam, 2024-02-06:
2) Setting `i_set_isp_scale: false` should get rid of that error.
3) Yes, by default Stereo is composed of L&R cameras and aligned to the middle one. To the parameters as above should provide you with such publishing, you might also want to set output sizes for `rgb` sensor as well, the stereo node resizes its output to fit the width/height of the aligned node.

hangcui1201, 2024-02-06:
Hi Serafadam,

Thanks a lot! It worked! I set 

i_publish_topic: true
i_output_isp: false
i_set_isp_scale: false
i_width: 1280
i_height: 800

I have one more question. I found the frame rate was directly related to the i_width and i_height. However, if I chose a smaller image size, the FoV would also be cut. **Is there anyway that I can down sampling the image, keep the FoV and increase the frame rate?** For example, the output size of the images are 640 x 400 which keep the FoV of 1280 x 800. At the meantime, the frame rate can be increased? The frame rate of 1280 x 800 is around 21Hz.

Thanks a lot for the help!

Best,
Hang

hangcui1201, 2024-02-09:
Hi Serafadam,

In addition to the previous question, I have another quick question that could you give me some suggestions on the configurations of stereo images? For example, some configurations provide higher resolution on the depth images. 

Thanks a lot for the help!

Best,
Hang

Serafadam, 2024-02-09:
Hi, to keep the same FOV you'll need to use the ISP scaling parameters as mentioned [here](https://docs.luxonis.com/projects/api/en/latest/components/nodes/color_camera/), you can use [this document](https://docs.google.com/spreadsheets/d/153yTstShkJqsPbkPOQjsVRmM8ZO3A6sCqm7uayGF-EE/edit#gid=0) to calculate the correct scaling options, in short if all fields are green then it is OK for stereo (by default, for LR it sets `i_isp_num: 3` & `i_isp_den: 5` which should result in 1152x720 resolution.
Regarding the framerate, here are some options that might help:
- Enabling low_bandwidth
- Setting isp3a algorithm fps, more information [here](https://docs.luxonis.com/projects/api/en/latest/tutorials/debugging/#cpu-usage), this can be set using `i_set_isp3a_fps` and `i_isp3a_fps` parameters
- Using a high quality USB cable, maximum speed that is possible is [SUPER_PLUS](https://docs.luxonis.com/projects/api/en/latest/references/python/?highlight=super_plus#depthai.UsbSpeed) (using Thunderbolt port if you have it might also help here)

Regarding stereo configuration, you can refer to [this page](https://docs.luxonis.com/projects/api/en/latest/components/nodes/stereo_depth/), quality of depth depends on a large number of factors, we currently provide default parameters based on OAK-D-PRO models (also, subpixel mode is disabled by default due to legacy reasons).

hangcui1201, 2024-02-12:
Hi Serafadam,

Thanks a lot! It works! A last question, how did you get 1152 x 720 when i_isp_num: 3 & i_isp_den: 5 from the table of ISP scaling options? Or under which tab I should do such calculation?

Best,
Hang

Serafadam, 2024-02-13:
Hi, just edit `Input` width and height options.

closed 2024-02-27 by Serafadam

hangcui1201, 2024-03-03:
Hi Serafadam,

Thanks a lot! Just one more question.

I found if I used the USB 3.0 cable with the OAK-D LR camera, it would draw a lot of current from the USB port of the computer. It lead to some malfunction to other USB device. In my case, it was the GNSS/INS device which used another USB port. 

**Is there any way to adjust the power consumption of the the OAK-D LR camera? For example, low power mode, or draw less current?**

Thanks!

Best,
Hang

hangcui1201, 2024-03-03:
Hello,

I found reduce the FPS can reduce the power consumption. https://docs.luxonis.com/projects/hardware/en/latest/pages/OAK-D-LR/. **Which parameter of depthai_ros_driver that I can tune for reducing the FPS?**

Thanks a lot!

Best,
Hang

Serafadam, 2024-03-04:
Hi, for fps you can use `i_fps` parameter, for other parameters you can refer to the documentation. You can also supply power separately, via [USB Y adapter](https://shop.luxonis.com/collections/accessories/products/oak-y-adapter)

hangcui1201, 2024-03-04:
Hi Serafadam,

Thanks a lot!

Best,
Hang

zshaikh901, 2025-04-29:
[image]
hello sir, 
i want to run the fire detection example, i am able to run it before by this OAK-D LR camera, but it is keep saying fire to every thing, after i try to run it again but i am getting this error again. is there any thing i am doing wrong or i have to do something? i am new at ros and all these things. 
thank for your guidance.

Serafadam, 2025-05-06:
Hi, it looks like you are not using ROS integration but depthai-experiments which are written in python, please create an issue there or post on our forums to get more general feedback