luxonis/depthai-ros #490: [BUG] RGBStereo fails for OAK-FFC-4P-POE with RGB camera on CAM_D (socket 3)

andermi, 2024-01-30:
RGBStereo fails for OAK-FFC-4P-POE with RGB camera on CAM_D (socket 3) with exception:
```
Failed to load node 'port_oak' of type 'depthai_ros_driver::Camera' in container '/port_oak_container': Component constructor threw an exception: _Map_base::at
```

Although, a second OAK-FFC-4P-POE with RGB camera on CAM_A (socket 0) works just fine.

Serafadam, 2024-01-31:
Hi, yeah, sockets are hardcoded in the pipeline creation although you should be able to change them using ROS parameter `sensor.i_board_socket_id`, please let us know if that doesn't work properly.

Serafadam, 2024-02-27:
Closing due to inactivity, please reopen if you have further questions.

closed 2024-02-27 by Serafadam