luxonis/depthai-ros #496: [BUG] Installation of depthai.git on Jetson Orin is unsuccessful.

MAli7319, 2024-02-12:
When I try to clone depthai.git in my Orin, this error message appears:
_E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-depthai-examples/ros-noetic-depthai-examples_2.8.2-1focal.20231212.082208_arm64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-depthai-filters/ros-noetic-depthai-filters_2.8.2-1focal.20231017.124552_arm64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-depthai-ros-driver/ros-noetic-depthai-ros-driver_2.8.2-1focal.20231212.083424_arm64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-depthai-ros/ros-noetic-depthai-ros_2.8.2-1focal.20231212.084926_arm64.deb  404  Not Found [IP: 64.50.233.100 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?_

This message points out the site http://packages.ros.org/ros/ubuntu/pool/main/r/. It seems there should be a file with version of "**2.9.0-1focal.20240125.204947_arm64.deb**" but in the depthai.git repo, the version is "**2.8.2-1focal.20231212.082208_arm64.deb**". For all of the 4 files inside the repo needs to be updated to the current version I think. Can you fix this issue? Thanks in advance.

Serafadam, 2024-02-13:
Hi does that error occur when cloning or at some other step? Could you try running `sudo apt update` and retrying?

MAli7319, 2024-02-13:
Hi and thanks for the quick response. This error occurred when I tried to clone depthai-ros repo. Your advice solved it. 
But after I tried to setup depthai-demo (https://github.com/luxonis/depthai#depthai-api-demo-program):
`git clone --recursive https://github.com/luxonis/depthai.git` command caused another error:

Cloning into 'depthai'...
remote: Enumerating objects: 21659, done.
remote: Counting objects: 100% (3624/3624), done.
remote: Compressing objects: 100% (436/436), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

Before this error message printed, cloning process continued up to %9 completion. After some time, the upload slowed down and printed the error message at the end. What should I do? Thanks in advance.

Serafadam, 2024-02-14:
Hi, seems like an issue with git, not depthai side, could be caused by some network error or git[ configuration issue](https://stackoverflow.com/questions/15240815/git-fatal-the-remote-end-hung-up-unexpectedly)

Serafadam, 2024-02-27:
Closing due to inactivity, please reopen if you have further questions.

closed 2024-02-27 by Serafadam