luxonis/depthai-ros #497: Cannot colcon build depthai-ros on Jetson Orin Jetpack 6

brianlmerritt, 2024-02-13:
**Check if issue already exists**

https://github.com/luxonis/depthai-ros/issues/419 mentions the issue on Xavier, but has not progressed

**Describe the bug**
Standard Jetpack 6 Ubuntu 22.04 build with ROS 2 humble compiles ok with Orbbec Astra but not with DepthAI  See below

```
Starting >>> depthai_examples
Finished <<< depthai_examples [1.06s]                                                                       
Starting >>> depthai_ros_driver
Finished <<< depthai_ros_driver [3.89s]                                                                       
[Processing: depthai_filters]                               
[Processing: depthai_filters]                                        
--- stderr: depthai_filters                                           
In file included from /home/blm/limo_weston_ws/src/depthai-ros/depthai_filters/src/wls_filter.cpp:1:
/home/blm/limo_weston_ws/src/depthai-ros/depthai_filters/include/depthai_filters/wls_filter.hpp:8:10: fatal error: opencv2/ximgproc/disparity_filter.hpp: No such file or directory
    8 | #include "opencv2/ximgproc/disparity_filter.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_filters.dir/build.make:118: CMakeFiles/depthai_filters.dir/src/wls_filter.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/depthai_filters.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< depthai_filters [1min 27s, exited with code 2]

Summary: 13 packages finished [1min 34s]
  1 package failed: depthai_filters

```

**Minimal Reproducible Example**

Install Jetpack 6 on Jetson (must be Orin + for Jetpack 6
Install ROS 2 humble, install depthai-sdk and ros-humble-depthai
Git clone depthai-ros in src and colcon build --symlink-install  reproduces the above error.

If available launch files don't work in your case, please check if you also get errors while running:
not applicable

**Expected behavior**
colcon build should work or the necessary dependencies should be included (or workarounds mentioned) in depthai-ros

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Pipeline Graph**

Not applicable

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py) - 

```
python3 log_system_information.py 
{
    "architecture": "64bit ELF",
    "machine": "aarch64",
    "platform": "Linux-5.15.122-tegra-aarch64-with-glibc2.35",
    "processor": "aarch64",
    "python_build": "main Nov 20 2023 15:14:05",
    "python_compiler": "GCC 11.4.0",
    "python_implementation": "CPython",
    "python_version": "3.10.12",
    "release": "5.15.122-tegra",
    "system": "Linux",
    "version": "#1 SMP PREEMPT Mon Dec 18 21:24:25 PST 2023",
    "win32_ver": "",
    "packages": [
[... 330 lines cut ...]
        "xdg==5",
        "xkit==0.0.0",
        "zipp==1.0.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux ubuntu 5.15.122-tegra #1 SMP PREEMPT Mon Dec 18 21:24:25 PST 2023 aarch64 aarch64"
    ]
}
```

 - Which OS/OS version are you using?  UBUNTU 22.04
 - Which ROS version are you using? ROS 2 
 - Which ROS distribution are you using ? Humble
 - Is `depthai-ros` built from source or installed from apt - APT but either makes no difference
 - Is `depthai/depthai-core` library installed from rosdep or manually? For rosdep install, check if `ros-<rosdistro>-depthai` package is installed, manual install can be checked with `ldconfig -p | grep depthai`-    Don't know
 - Please include versions of following packages - `apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver`
 - To get additional logs, set `DEPTHAI_DEBUG=1` and paste the logs, either from command line or from latest log in `~/.ros/log`  **Not applicable as it doesn't compile**

**Additional context**
I have seen suggestions go to https://github.com/mdegans/nano_build_opencv/ but even after modifying the code to work with Ubuntu 22.04 and install opencv 4.9 this still fails because the build does not create the necessary headers for `opencv2/ximgproc`  I used `locate *opencv*/*.h` to look for all related files and all I got was duplicates of what was already installed with the nvidia-opencv-dev so it isn't just python or cmake looking at the wrong files.

If there is an opencv build that includes the necessary files, happy to try it, or any work around.

brianlmerritt, 2024-02-13:
Just noticed I should have been going for 

locate *opencv*/*.hpp

closed 2024-02-13 by brianlmerritt

brianlmerritt, 2024-02-14:
Just in case someone finds this closed ticket, first I updated nano_build_opencv for jetpack 6 to reinstall opencv 4.9 from source.

https://github.com/brianlmerritt/nano_build_opencv

I then created a separate workspace `depthai_ws/src/` with depthai-ros there, and ran the build script in the top ws folder.  I was trying to build inside the agilex limo ws but had to run build.sh to get it to work.

jagruth-medavarapu, 2025-08-26:
Hi @Serafadam, the issue is resolved after building and instaaling openCV from source.