luxonis/depthai-ros #499: 'camera.yaml' parameters not being used when launch 'camera.launch.py'

PlayWeird, 2024-02-15:
When I launch "camera.launch.py" it does not seam to be taking into account the camera.yaml config for use with a RGB pipeline.

Here is the config.yaml I am testing with

```/oak:
  ros__parameters:
    camera:
      i_calibration_dump: true
      i_enable_imu: true
      i_enable_ir: false
      i_external_calibration_path: ''
      i_nn_type: rgb
      i_pipeline_dump: false
      i_pipeline_type: RGB
      i_publish_tf_from_calibration: false
      i_usb_speed: SUPER_PLUS
    diagnostic_updater:
      period: 1.0```
 
And here is the launch readout

``` [INFO] [rviz2-1]: process started with pid [20544]
[INFO] [component_container-2]: process started with pid [20546]
[component_container-2] [INFO] [1708034507.593736651] [AR0234_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[component_container-2] [INFO] [1708034507.595357180] [AR0234_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-2] [INFO] [1708034507.595382326] [AR0234_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-2] [INFO] [1708034507.598533221] [AR0234_state_publisher]: got segment AR0234
[component_container-2] [INFO] [1708034507.598552239] [AR0234_state_publisher]: got segment AR0234_left_camera_frame
[component_container-2] [INFO] [1708034507.598556051] [AR0234_state_publisher]: got segment AR0234_left_camera_optical_frame
[component_container-2] [INFO] [1708034507.598558761] [AR0234_state_publisher]: got segment AR0234_model_origin
[component_container-2] [INFO] [1708034507.598561310] [AR0234_state_publisher]: got segment AR0234_rgb_camera_frame
[component_container-2] [INFO] [1708034507.598563873] [AR0234_state_publisher]: got segment AR0234_rgb_camera_optical_frame
[component_container-2] [INFO] [1708034507.598566246] [AR0234_state_publisher]: got segment AR0234_right_camera_frame
[component_container-2] [INFO] [1708034507.598568792] [AR0234_state_publisher]: got segment AR0234_right_camera_optical_frame
[component_container-2] [INFO] [1708034507.598571134] [AR0234_state_publisher]: got segment oak-d-base-frame
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/AR0234_state_publisher' in container 'AR0234_container'
[component_container-2] [INFO] [1708034507.599454186] [AR0234_container]: Load Library: /home/gaetano/depthai_ws/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[component_container-2] [INFO] [1708034507.646183218] [AR0234_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1708034507.646232425] [AR0234_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1708034507.651654267] [AR0234]: No ip/mxid specified, connecting to the next available device.
[rviz2-1] [INFO] [1708034507.679568009] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1708034507.679697066] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1708034507.695637936] [rviz2]: Stereo is NOT SUPPORTED
[component_container-2] [INFO] [1708034512.407257065] [AR0234]: Camera with MXID: 14442C1061BDC8D600 and Name: 3.4.3 connected!
[component_container-2] [INFO] [1708034512.407989836] [AR0234]: USB SPEED: SUPER
[component_container-2] [INFO] [1708034512.428791258] [AR0234]: Device type: OAK-FFC-3P
[component_container-2] [INFO] [1708034512.430686121] [AR0234]: Pipeline type: RGBD
[component_container-2] [ERROR] [1708034512.431521224] [AR0234]: Invalid pipeline chosen for camera as it has only one sensor. Switching to RGB.
[component_container-2] [WARN] [1708034512.455275490] [AR0234]: Spatial NN selected, but configuration is RGB. Please change camera.i_nn_type parameter to RGB.
[component_container-2] [INFO] [1708034512.676463959] [AR0234]: Finished setting up pipeline.
[component_container-2] [ERROR] [1708034512.770546562] [AR0234]: No calibration for socket 0! Publishing empty camera_info.
[component_container-2] [INFO] [1708034512.824545139] [AR0234]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/AR0234' in container '/AR0234_container'
[rviz2-1] [INFO] [1708034543.685646323] [rviz2]: Stereo is NOT SUPPORTED
```

I had this same setup running in ROS1 with no issues, but have been switching to ROS2 and I don't understand why the config file is not changing the pipeline correctly.

Serafadam, 2024-02-16:
Hi, it seems you have changed you node name, you should also change it in yaml config - `/oak` -> `/AR0234`

PlayWeird, 2024-02-16:
Silly mistake on my part. Thank you, that got things up and running again.

PlayWeird, 2024-02-16:
Silly mistake on my part. Thank you, that fixed the issue.

closed 2024-02-16 by PlayWeird