luxonis/depthai-ros #503: [BUG] ROS Documentation link goes to 404 Error page

tweng-bdai, 2024-02-20:
https://docs-beta.luxonis.com/develop/ros/depthai-ros-intro/ returns 404. The link is in the README of this repo. I have accessed this page previously but am unable to access it now. 

Repro'd on several browsers and networks.

Serafadam, 2024-02-21:
Hi, thanks, link has been updated on the main branch.

gestalt73, 2024-02-27:
https://docs-beta.luxonis.com/software/ros/depthai-ros/intro/ is the link on the default branch "humble"

I'm still getting a cloudfare 404 error.

raymondsong00, 2024-03-02:
I was trying build from source but couldn't find the steps anymore. The "build from source" and "driver" options that are linked options on the docs returns "unexpected error occurred" and the hyperlink from the intro tab returns a 404.

Serafadam, 2024-03-04:
Hi, thanks for notifying, this was caused by some error in the docs, those pages should be online now.

closed 2024-03-18 by Serafadam