luxonis/depthai-ros #514: [BUG] {colcon build fail - opencv version stderr}

gsoto-tellez, 2024-03-21:
I am using a Jetson Xavier and Ubuntu 22.04. 

When running colcon build after cloning the depthai files and creating the workspace but got an error related to the opencv version. I have verison 4.5.4. However I get this error:

[image]

the CMakeLists.txt file includes:

[image]

I tried changing the version but still got more errors. Can I remove the find version line altogether?

Serafadam, 2024-03-22:
Hi, as ROS Foxy is EOL it is no longer officially supported, I would recommend switching to Humble if it is possible, either via apt or Docker image. If that is not possible, you could try updating [depthai_filters' CMakeLists.txt](https://github.com/luxonis/depthai-ros/blob/foxy/depthai_filters/CMakeLists.txt) based on [current Humble version](https://github.com/luxonis/depthai-ros/blob/humble/depthai_filters/CMakeLists.txt).

gsoto-tellez, 2024-03-22:
Thank you, doing that solved that issue. However, now I am getting the following issue related to opencv. I have opencv version 4.2.0. Do I need to make a change to the opencv files?

[image]

Serafadam, 2024-03-26:
Hi, I think you do not have modules from opencv_contrib module, you should install it to get them.

gsoto-tellez, 2024-03-29:
I installed the opencv_contrib module but have this error when building depthai_filters. I have verified the ximgproc file is present.

Starting >>> depthai_filters
--- stderr: depthai_filters                         
CMake Error at /usr/share/cmake-3.29/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenCV (missing: ximgproc) (found version "4.5.4")
Call Stack (most recent call first):
  /usr/share/cmake-3.29/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/lib/aarch64-linux-gnu/cmake/opencv4/OpenCVConfig.cmake:354 (find_package_handle_standard_args)
  CMakeLists.txt:4 (find_package)

make: *** [Makefile:348: cmake_check_build_system] Error 1

Serafadam, 2024-04-03:
Hi, I think this might be issue with Jetson as I've seen similar issues on [Nvidia forums](https://forums.developer.nvidia.com/t/installing-opencv-contrib-for-jetson-nano/191453), could you try checking if their solutions work?