luxonis/depthai-ros #515: [BUG] rgbd_pcl inconsistent Resolution configuration

TheMarksniper, 2024-03-22:
Description:
Running the rgbd_pcl.launch.py launcher file from depthai_ros_driver package with custom params_file is yielding an inconsistent resolution.
Specified resolution (1920x1080, 1080P) set and checked against parameters (ros2 param dump /oak
) and camera info (ros2 topic echo /oak/rgb/camera_info) give the right resolution. Inconsistency comes with the oak/rgb/image_raw topic as the output gives resolution of 1280x720. This problem is not present for the stereo topic as it changes properly according to the parameters.

used parameter file:

/oak:
  ros__parameters:
    camera:
      i_nn_type: none
      i_enable_ir: true
      i_laser_dot_brightness: 1200
      i_floodlight_brightness: 1500
      i_ip: 192.168.8.23
    rgb:
      i_resolution: 1080P
      i_width: 1920
      i_height: 1080
      rgb_i_set_isp_scale: true
      i_max_q_size: 1
      i_fps: 10.0

    stereo:
      i_subpixel: false
      i_enable_disparity_shift: true
      i_disparity_shift: 50
      i_max_q_size: 1
      i_align_depth: true

    right:
      i_resolution: 400P
      i_max_q_size: 1
      i_fps: 10.0

    left:
      i_resolution: 400P
      i_max_q_size: 1
      i_fps: 10.0

Error also exists while running  - `camera` in `depthai_ros_driver` with same resolution parameters

Expected behavior:
Image height and width being the same in the params ,camera_info and image_raw topics

Attached log:
OS: Ubuntu 22.04.4 LTS
ROS: ROS2 Humble
Depthai installation: apt install

[image]
[image]
[image]

Serafadam, 2024-03-26:
Hi, by default `rgb` camera is using ISP scaling which changes the image size, more information on that can be found in [documentation](https://docs-beta.luxonis.com/software/ros/depthai-ros/driver/#Depthai%20ROS%20Driver-Setting%20RGB%20parameters)

TheMarksniper, 2024-03-27:
> Hi, by default `rgb` camera is using ISP scaling which changes the image size, more information on that can be found in [documentation](https://docs-beta.luxonis.com/software/ros/depthai-ros/driver/#Depthai%20ROS%20Driver-Setting%20RGB%20parameters)

Sorry i should have read the documentation better (RTFM is allways true).
Thank you for reply

closed 2024-03-27 by TheMarksniper