luxonis/depthai-ros #516: [BUG] Right stereo camera is not detected with OAK-D SR PoE

dmn-seasony, 2024-03-22:
**Describe the bug**
We just received the production units of OAK-D SR PoE, but it seems like one of the RGB camera sensors is not detected. I tried two different units and both have the same problem. This in turn causes the ROS driver (which is what we intend to use) to fail to start because it expects 3 sensors to be present. We did not have these issues with the engineering sample which we tested previously.

**Minimal Reproducible Example**
```
$ python3 -c 'from depthai_sdk import OakCamera; oak = OakCamera(); print(oak.device.getConnectedCameraFeatures())'
[{socket: CAM_A, sensorName: S5K33D, width: 640, height: 480, orientation: ROTATE_180_DEG, supportedTypes: [TOF], hasAutofocus: 0, name: tof}, {socket: CAM_B, sensorName: OV9282, width: 1280, height: 800, orientation: ROTATE_180_DEG, supportedTypes: [COLOR], hasAutofocus: 0, name: left}]
```

**Expected behavior**
One more entry in the list above, corresponding to the right RGB camera.

**Attach system log**
<details><summary>Details</summary>

```
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-6.5.0-25-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "python_build": "main Nov 20 2023 15:14:05",
    "python_compiler": "GCC 11.4.0",
    "python_implementation": "CPython",
    "python_version": "3.10.12",
    "release": "6.5.0-25-generic",
    "system": "Linux",
    "version": "#25~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Feb 20 16:09:15 UTC 2",
    "win32_ver": "",
    "packages": [
        "ackermann-msgs==2.0.2",
        "action-msgs==1.2.1",
        "action-tutorials-interfaces==0.20.3",
        "action-tutorials-py==0.20.3",
        "actionlib-msgs==4.2.3",
        "AHRS==0.3.1",
        "ament-cmake-test==1.3.8",
        "ament-copyright==0.12.10",
        "ament-cppcheck==0.12.10",
        "ament-cpplint==0.12.10",
        "ament-flake8==0.12.10",
        "ament-index-python==1.4.0",
        "ament-lint==0.12.10",
        "ament-lint-cmake==0.12.10",
        "ament-package==0.14.0",
        "ament-pep257==0.12.10",
        "ament-uncrustify==0.12.10",
        "ament-xmllint==0.12.10",
        "angles==1.15.0",
        "blobconverter==1.4.3",
        "bond==3.0.2",
        "builtin-interfaces==1.2.1",
        "camera-calibration==3.0.3",
        "cfgv==3.4.0",
        "comm==0.2.2",
        "composition-interfaces==1.2.1",
        "control-msgs==4.4.0",
        "controller-manager==2.39.1",
        "controller-manager-msgs==2.39.1",
        "cv-bridge==3.2.1",
        "demo-nodes-py==0.20.3",
        "depthai==2.22.0.0",
        "depthai-pipeline-graph==0.0.5",
        "depthai-ros-driver==2.9.0",
        "depthai-ros-msgs==2.9.0",
        "depthai-sdk==1.13.1",
        "depthai-viewer==0.1.8",
        "diagnostic-msgs==4.2.3",
        "diagnostic-updater==3.1.2",
        "distlib==0.3.8",
        "domain-coordinator==0.10.0",
        "dwb-msgs==1.1.13",
        "example-interfaces==0.9.3",
        "examples-rclpy-executors==0.15.1",
        "examples-rclpy-minimal-action-client==0.15.1",
        "examples-rclpy-minimal-action-server==0.15.1",
        "examples-rclpy-minimal-client==0.15.1",
        "examples-rclpy-minimal-publisher==0.15.1",
        "examples-rclpy-minimal-service==0.15.1",
        "examples-rclpy-minimal-subscriber==0.15.1",
        "filelock==3.13.1",
        "foxglove-msgs==2.3.0",
        "gazebo-msgs==3.7.0",
        "generate-parameter-library-py==0.3.7",
        "geographic-msgs==1.0.5",
        "geometry-msgs==4.2.3",
        "identify==2.5.35",
        "image-geometry==3.2.1",
        "interactive-markers==2.3.2",
        "ipympl==0.9.3",
        "ipywidgets==8.1.2",
        "jupyterlab_widgets==3.0.10",
        "laser-geometry==2.4.0",
        "launch==1.0.5",
        "launch-ros==0.19.7",
        "launch-testing==1.0.5",
        "launch-testing-ros==0.19.7",
        "launch-xml==1.0.5",
        "launch-yaml==1.0.5",
        "lifecycle-msgs==1.2.1",
        "logging-demo==0.20.3",
        "map-msgs==2.1.0",
        "marshmallow==3.17.0",
        "message-filters==4.3.3",
        "nav-2d-msgs==1.1.13",
        "nav-msgs==4.2.3",
        "nav2-common==1.1.13",
        "nav2-msgs==1.1.13",
        "nav2-simple-commander==1.0.0",
        "nodeenv==1.8.0",
        "numpy==1.26.4",
        "opencv-contrib-python==4.9.0.80",
        "osrf-pycommon==2.0.2",
        "pcl-msgs==1.0.0",
        "pendulum-msgs==0.20.3",
        "platformdirs==4.2.0",
        "pre-commit==3.6.2",
        "pyarrow==10.0.1",
        "python-qt-binding==1.1.2",
        "pytube==15.0.0",
        "PyTurboJPEG==1.6.4",
        "qt-dotgraph==2.2.3",
        "qt-gui==2.2.3",
        "qt-gui-cpp==2.2.3",
        "qt-gui-py-common==2.2.3",
        "Qt.py==1.3.10",
        "quality-of-service-demo-py==0.20.3",
        "rcl-interfaces==1.2.1",
        "rclpy==3.3.12",
        "rcutils==5.1.5",
        "resource-retriever==3.1.1",
        "rmw-dds-common==1.6.0",
        "robot-localization==3.5.2",
        "ros2action==0.18.9",
        "ros2bag==0.15.9",
        "ros2cli==0.18.9",
        "ros2component==0.18.9",
        "ros2controlcli==2.39.1",
        "ros2doctor==0.18.9",
        "ros2interface==0.18.9",
        "ros2launch==0.19.7",
        "ros2lifecycle==0.18.9",
        "ros2multicast==0.18.9",
        "ros2node==0.18.9",
        "ros2param==0.18.9",
        "ros2pkg==0.18.9",
        "ros2run==0.18.9",
        "ros2service==0.18.9",
        "ros2topic==0.18.9",
        "rosbag2-interfaces==0.15.9",
        "rosbag2-py==0.15.9",
        "rosbags==0.9.22",
        "rosgraph-msgs==1.2.1",
        "rosidl-adapter==3.1.5",
        "rosidl-cli==3.1.5",
        "rosidl-cmake==3.1.5",
        "rosidl-generator-c==3.1.5",
        "rosidl-generator-cpp==3.1.5",
        "rosidl-generator-py==0.14.4",
        "rosidl-parser==3.1.5",
        "rosidl-runtime-py==0.9.3",
        "rosidl-typesupport-c==2.0.1",
        "rosidl-typesupport-cpp==2.0.1",
        "rosidl-typesupport-fastrtps-c==2.2.2",
        "rosidl-typesupport-fastrtps-cpp==2.2.2",
        "rosidl-typesupport-introspection-c==3.1.5",
        "rosidl-typesupport-introspection-cpp==3.1.5",
        "rpyutils==0.2.1",
        "rqt-action==2.0.1",
        "rqt-bag==1.1.4",
        "rqt-bag-plugins==1.1.4",
        "rqt-console==2.0.3",
        "rqt-graph==1.3.0",
        "rqt-gui==1.1.7",
        "rqt-gui-py==1.1.7",
        "rqt-msg==1.2.0",
        "rqt-plot==1.1.2",
        "rqt-publisher==1.5.0",
        "rqt-py-common==1.1.7",
        "rqt-py-console==1.0.2",
        "rqt-reconfigure==1.1.2",
        "rqt-service-caller==1.0.5",
        "rqt-shell==1.0.2",
        "rqt-srv==1.0.3",
        "rqt-topic==1.5.0",
        "ruamel.yaml==0.18.6",
        "ruamel.yaml.clib==0.2.8",
        "scripts==3.7.0",
        "sensor-msgs==4.2.3",
        "sensor-msgs-py==4.2.3",
        "sentry-sdk==1.21.0",
        "shape-msgs==4.2.3",
        "sick-safetyscanners2-interfaces==1.0.0",
        "slam-toolbox==2.6.8",
        "smclib==3.0.2",
        "sros2==0.10.4",
        "statistics-msgs==1.2.1",
        "std-msgs==4.2.3",
        "std-srvs==4.2.3",
        "stereo-msgs==4.2.3",
        "teleop-twist-keyboard==2.3.2",
        "test-msgs==1.2.1",
        "tf2-geometry-msgs==0.25.6",
        "tf2-kdl==0.25.6",
        "tf2-msgs==0.25.6",
        "tf2-py==0.25.6",
        "tf2-ros-py==0.25.6",
        "tf2-tools==0.25.6",
        "theora-image-transport==2.5.0",
        "topic-monitor==0.20.3",
        "trajectory-msgs==4.2.3",
        "turtlesim==1.4.2",
        "types-pyside2==5.15.2.1.7",
        "unique-identifier-msgs==2.2.1",
        "urllib3==2.2.1",
        "virtualenv==20.25.1",
        "vision-msgs==4.1.1",
        "visualization-msgs==4.2.3",
        "widgetsnbextension==4.0.10",
        "wiimote-msgs==3.3.0",
        "xacro==2.0.8",
        "xmltodict==0.13.0",
        "zstandard==0.22.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux daniel-ThinkPad-P15-Gen-2i 6.5.0-25-generic luxonis/depthai#25~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Feb 20 16:09:15 UTC 2 x86_64 x86_64"
    ]
}
```

</details>

Erol444, 2024-03-22:
Moved the ticket here as it's more ROS-related. @dmn-seasony , are you using the latest depthai version (2.25)?

dmn-seasony, 2024-03-25:
@Erol444 It is not a problem specific to ROS, see the MRE. I just updated to the latest version of depthai and the problem is still there.
```
$ pip3 list | grep depthai
depthai                              2.22.0.0
depthai-pipeline-graph               0.0.5
depthai-sdk                          1.13.1
depthai-viewer                       0.1.8
$ python3 -c 'from depthai_sdk import OakCamera; oak = OakCamera(); print(oak.device.getConnectedCameraFeatures())'
[{socket: CAM_A, sensorName: S5K33D, width: 640, height: 480, orientation: ROTATE_180_DEG, supportedTypes: [TOF], hasAutofocus: 0, name: tof}, {socket: CAM_B, sensorName: OV9282, width: 1280, height: 800, orientation: ROTATE_180_DEG, supportedTypes: [COLOR], hasAutofocus: 0, name: left}]
```

closed 2024-03-25 by dmn-seasony