luxonis/depthai-ros #517: How can I compress the camera data ?

Jenanaputra, 2024-03-23:
I launched the camera.launch.py. It looked fine to me and no errors. But I got the lag camera data stream since the default size/resolution of the camera is quite high. Therefore I was wondering to compress the camera data with the size that i want. Is it possible to do it ? If it is, Can you guys tell me what should i do? (For additional information, I used the OAK-D POE with me and access it through wire).

Serafadam, 2024-03-26:
Hi, you can use low_bandwidth_mode parameter to enable MJPEG encoding for data. More information on that can be found in the documentation [here](https://docs-beta.luxonis.com/software/ros/depthai-ros/driver/#Depthai%20ROS%20Driver-Specific%20camera%20configurations)  and [here](https://docs.luxonis.com/projects/api/en/latest/tutorials/low-latency/)

Jenanaputra, 2024-03-27:
@Serafadam Hi, it works. Thanks.

Actually, the reason I want to compress the data is because I intend to fed it into rtabmap-ros. When I tried it out, I  faced two issues:

1. When I launch the rtabmap.launch.py using default configuration of camera (`rgb.i_low_bandwidth = False`), the 3D map came out, but it probably took like more than 5 minutes for the map came out. After the map came out, its very laggy.
2. When I launch it by setting the camera (`rgb.i_low_bandwidth = True`), I got warning on my terminal `[component_container-1] [194430103155211300] [192.168.8.170] [171.129] [VideoEncoder(10)] [warning] Arrived frame type (14) is not either NV12 or YUV400p (8-bit Gray)`. I also wait it for 5 minutes, but nothing came up. Isn't it supposed to be lighter?

Any idea how to solve the problems I encountered ?

Jenanaputra, 2024-04-13:
do you have any idea about this ? @Serafadam

Serafadam, 2024-04-22:
Hi @Jenanaputra, sorry for the delay, but I wanted to share that in the near future we will add RTABMap (and VIO) integration to depthai-core to mitigate some of the issues that can be happening with ROS, you can expect some early version releases in upcoming weeks.

 Regarding those warnings, could you post some more information according to [this report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)? Setting `rgb.i_low_bandwidth` shouldn't give such error messages.

Jenanaputra, 2024-05-07:
@Serafadam  Does that mean i need to open new issue ? or i can continue it on this existing issue but by following the report template ?

Jenanaputra, 2024-05-14:
Hi @Serafadam I have already created the new issue on #529 related to what I have faced lately, please kindly check it out.

closed 2024-05-26 by Jenanaputra