luxonis/depthai-ros #519: When launch the OAK-D LR camera, it stops other USB device from working

hangcui1201, 2024-03-28:
Hi,

We have made our OAK-D LR camera work with the depthai_ros_driver through the USB 3.1 cable. 

**However, we found whenever we launched the OAK-D LR camera using the depthai_ros_driver, it stopped our another USB device from working.** In our case, it is the Septentrio GNSS-INS device. It uses the USB 2.0 cable, USB-Over-Ethernet protocol. The heading information became N/A. Whenever we stopped launching the OAK-D LR camera, the heading information became available again. 

[image]

[image]

The GNSS-INS device was powered independently from the 12V power source. The OAK-D LR camera was powered through the USB 3.1 cable. We also used a USB 3.1 boost extension cable and we added an additional 5V power to that extension cable. **So we think such issue should not be due to the power supply.** 

We also tried to launch the OAK-D LR camera with streaming videos only from the center len. It still did not work. **So the issue should not be due to the USB bandwidth, either.** We tried different USB ports. The results were the same.  Whenever we launched the OAK-D LR camera using the depthai_ros_driver, it stopped our GNSS-INS USB device from working. 

We have no idea what happened. **Is it possible that whenever we launched the OAK-D LR camera, it used some USB protocol that stopped the GNSS-INS USB device from working?** Any other suggestions?

Thanks a lot for the help!

Best,
Hang

Serafadam, 2024-03-29:
Hi, could you post some more information following the [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)?

hangcui1201, 2024-04-01:
Hi Serafadam,

Sorry, we found it was the IMU initialization issue of the other USB GNSS/INS device.

Best,
Hang

Serafadam, 2024-04-04:
Thanks for the information, closing the issue.

closed 2024-04-04 by Serafadam