luxonis/depthai-ros #521: exit code -11, cmd when running stereo_inertial_node on RPI4

Ghada09, 2024-04-03:
pi@clover-5390:~/dai_ws/src/depthai-ros/depthai_examples/launch $ roslaunch stereo_inertial_node.launch 
... logging to /home/pi/.ros/log/def5fc84-f1b8-11ee-8767-23ac0e8d9fbf/roslaunch-clover-5390-16222.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://clover-5390.local:41415/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /oak/robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/detectionClassesCount: 80
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableRosBaseTimeUpdate: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/expTime: 20000
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/manualExposure: False
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/nnName: x
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/previewHeight: 416
 * /stereo_inertial_publisher/previewWidth: 416
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/resourceBaseFolder: /home/pi/dai_ws/s...
 * /stereo_inertial_publisher/rgbResolution: 1080p
 * /stereo_inertial_publisher/rgbScaleDinominator: 3
 * /stereo_inertial_publisher/rgbScaleNumerator: 2
 * /stereo_inertial_publisher/sensIso: 800
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

ROS_MASTER_URI=http://localhost:11311

process[oak_state_publisher-1]: started with pid [16255]
process[stereo_inertial_publisher-2]: started with pid [16256]
process[nodelet_manager-3]: started with pid [16257]
process[depth_image_convertion_nodelet-4]: started with pid [16258]
process[depth_image_to_rgb_pointcloud-5]: started with pid [16259]
[ INFO] [1712151888.360767771]: Initializing nodelet with 4 worker threads.
 nnPath ,, /home/pi/dai_ws/src/depthai-ros/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
1280 720 1280 720
Listing available devices...
Device Mx ID: 1844301041AA870E00
[1844301041AA870E00] [1.1.2] [1.568] [SpatialDetectionNetwork(9)] [warning] Network compiled for 6 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
Device USB status: SUPER
================================================================================REQUIRED process [stereo_inertial_publisher-2] has died!
process has died [pid 16256, exit code -11, cmd /home/pi/dai_ws/devel_isolated/depthai_examples/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/pi/.ros/log/def5fc84-f1b8-11ee-8767-23ac0e8d9fbf/stereo_inertial_publisher-2.log].
log file: /home/pi/.ros/log/def5fc84-f1b8-11ee-8767-23ac0e8d9fbf/stereo_inertial_publisher-2*.log
Initiating shutdown!
================================================================================
[depth_image_to_rgb_pointcloud-5] killing on exit
[depth_image_convertion_nodelet-4] killing on exit
[nodelet_manager-3] killing on exit
[stereo_inertial_publisher-2] killing on exit
[oak_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

Serafadam, 2024-04-04:
Hi, could you provide some more information according to [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)?

Ghada09, 2024-04-05:
thanks I already solved the issue by removing libboost that I installed manually

closed 2024-04-05 by Ghada09

rvhs9xsf6b-hue, 2026-10-03:
Nn