luxonis/depthai-ros #525: [BUG] Cameras on different subnets cannot be launched

jbarry-bdai, 2024-04-18:
Cameras on different subnets/LANs cannot be launched even when the IP is specified.  I believe this is because on lines 199-202 of [Camera.cpp](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/src/camera.cpp), a call is made to `getAllAvailableDevices` and an error is thrown if that doesn't return devices
```
                std::vector<dai::DeviceInfo> availableDevices = dai::Device::getAllAvailableDevices();
                if(availableDevices.size() == 0) {
                    throw std::runtime_error("No devices detected!");
                }
```
According to the [documentation](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe/#manually-specify-device-ip), this is expected behavior when the camera is on a different LAN.  Therefore, I think it needs to check if the user-provided IP works even if devices are not returned from `getAllAvailableDevices`.

**Minimal Reproducible Example**
1. Connect to an OAK-D camera over Ethernet but set your netmask to `255.255.0.0` (instead of `255.255.255.0`).
2. Confirm you can still ping the camera's IP address
3. Set `camera/i_ip` in the config file to the camera's IP address
4. Try to launch `ros2 launch depthai_ros_driver camera.launch.py` - it will not find the device even though it's pingable

However, if we use the depthai SDK directly, we can use the camera:
```
from depthai_sdk import OakCamera
# oak = OakCamera() <- This will fail because it's not on the same subnet
oak = OakCamera(device="<IP address>")  # This will succeed
color = oak.create_camera('color')
# Visualize the color camera to prove we're really connected
oak.visualize([color])
oak.start(blocking=True)
```

**Expected behavior**
When the camera IP is set, the launch file should be able to find the camera and launch successfully even when not on the same subnet.

Serafadam, 2024-04-22:
Hi, thanks for the report, a bugfix will be available for testing this week.

Serafadam, 2024-05-09:
Hi, sorry for the delay, could you test if [this branch](https://github.com/luxonis/depthai-ros/tree/ip_autodiscovery_fix_humble) works for you?