luxonis/depthai-ros #526: [Feature-Request] ROS Support for Sensor S5K33D

Thinkminator, 2024-04-23:
Hi, I have just bought a OAK-FFC-4P with 4 OAK-FFC ToF 33D recently, and would like to view TOFs from the sensors through ROS2 Humble, but I got error that the sensors S5K33D in the OAK-FFC ToF 33D are not supported
[image]
May I know is there any specific file or launch that I can run on, which support the modules , can either show TOFs or just normal color view?

MartinMotycka, 2024-05-09:
@Serafadam ^^

Serafadam, 2024-05-14:
Hi, sorry for the delay. Could you test if [this branch](https://github.com/luxonis/depthai-ros/tree/tof_support_humble) works for you?

closed 2025-01-03 by themarpe