luxonis/depthai-ros #527: [BUG] Directions for low bandwidth for POE cameras are not easy to find from error message

jbarry-bdai, 2024-04-26:
When launching with an POE OAK-D camera, if the low bandwidth settings are not enabled the driver helpfully tells me so:
```
PoE camera detected. Consider enabling low bandwidth for specific image topics (see readme).
```
However the README now redirects to [here](https://docs-beta.luxonis.com/software/ros/depthai-ros/intro/) and the documentation for the POE cameras here is linked from [here](https://docs-beta.luxonis.com/software/ros/depthai-ros/driver/#Depthai%20ROS%20Driver-Specific%20camera%20configurations).  (I know this is a trivial bug but I actually had quite a bit of trouble finding the parameters to set...)

MartinMotycka, 2024-05-20:
@Serafadam ^^

Serafadam, 2024-05-21:
Hi, warning message will be updated as part of https://github.com/luxonis/depthai-ros/tree/ip_autodiscovery_fix_humble