luxonis/depthai-ros #528: [BUG] {Max 250hz IMU publish rate with rotation vector rate active}

mjunsen123, 2024-05-09:
### Config
```
/oak:
  ros__parameters:
    camera:
      i_calibration_dump: false
      i_enable_imu: true
      i_enable_ir: false
      i_external_calibration_path: ''
      i_floodlight_brightness: 0
      i_ip: ''
      i_laser_dot_brightness: 800
      i_mx_id: ''
      i_nn_type: none
      i_pipeline_type: Depth
      i_usb_port_id: ''
      i_usb_speed: SUPER_PLUS
    diagnostic_updater:
      period: 1.0
    imu:
      i_acc_cov: 0.0
      i_acc_freq: 400
      i_batch_report_threshold: 1
      i_get_base_device_timestamp: false
      i_gyro_cov: 0.0
      i_gyro_freq: 400
      i_rot_cov: -1.0
      i_enable_rotation: true
      i_rot_freq: 400
      i_rotation_vector_type: ROTATION_VECTOR
      i_max_batch_reports: 10
      i_message_type: IMU
      i_sync_method: LINEAR_INTERPOLATE_ACCEL
      i_update_ros_base_time_on_ros_msg: false
    left:
      i_resolution: '400P'
      i_fps: 20.0
    right:
      i_resolution: '400P'
      i_fps: 20.0
    stereo:
      i_align_depth: false # switch between stereo - rgbd
      i_enable_distortion_correction: false
      i_output_disparity: false
      i_publish_left_rect: true
      i_publish_right_rect: true
      i_publish_synced_rect_pair: true
      i_publish_topic: false
      i_reverse_stereo_socket_order: true # try reverse
      i_publish_topic: false
      i_height: 400
      i_width: 640
    use_sim_time: false
```

### ROS2 topic hz /oak/imu/data output

```
average rate: 176.221
        min: 0.000s max: 0.059s std dev: 0.00878s window: 178
average rate: 180.299
        min: 0.000s max: 0.059s std dev: 0.00760s window: 363
average rate: 182.308
        min: 0.000s max: 0.059s std dev: 0.00707s window: 551
average rate: 183.709
        min: 0.000s max: 0.059s std dev: 0.00694s window: 739
average rate: 184.508
        min: 0.000s max: 0.059s std dev: 0.00681s window: 927
average rate: 185.414
        min: 0.000s max: 0.059s std dev: 0.00676s window: 1118
average rate: 186.261
        min: 0.000s max: 0.059s std dev: 0.00662s window: 1311
average rate: 186.460
        min: 0.000s max: 0.059s std dev: 0.00654s window: 1499
average rate: 186.781
        min: 0.000s max: 0.059s std dev: 0.00646s window: 1690
average rate: 186.978
        min: 0.000s max: 0.059s std dev: 0.00651s window: 1879
```

NOTE : If i set the frequency of acc, gyro and rot to 250 and it can achieve stable rate ~256.

MartinMotycka, 2024-05-20:
@Serafadam Check this, please.

Serafadam, 2024-05-21:
Hi @mjunsen123 , sorry for the delay, I discussed this issue with the firmware team and it seems this is caused by the IMU sensor processor as enabling sensor fusion degrades performance.

Serafadam, 2024-05-27:
Hi, unfortunately that is the case currently, this could change on RVC4 devices although it's hard to say now.

Serafadam, 2024-06-10:
Closing due to inactivity, please reopen if you have more questions.

closed 2024-06-10 by Serafadam