luxonis/depthai-ros #529: Got a particular warning and 3D map didn't show up when trying to launch the rtabmap.launch.py with low_bandwidth parameter is set to be true

Jenanaputra, 2024-05-14:
I want to launch the rtabmap.launch.py file with the low_bandwidth configuration set to true, so that when the 3D map is being generated, it is not laggy. 

But I got this warning :
`[component_container-1] [194430103155211300] [192.168.10.170] [9.068] [VideoEncoder(10)] [warning] Arrived frame type (14) is not either NV12 or YUV400p (8-bit Gray)`

I am using ROS2 Humble in ubuntu 22.04. I installed the depthai-ros by cloning it from the repository. 

Additionally, I also put some information here :

- The output of running `log_system_information.py` :
 
   [log_system_information.json](https://github.com/luxonis/depthai-ros/files/15303341/log_system_information.json)

- The output of `ldconfig -p | grep depthai` is :
`libdepthai-opencv.so (libc6,x86-64) => /usr/local/lib/libdepthai-opencv.so`
`libdepthai-core.so (libc6,x86-64) => /usr/local/lib/libdepthai-core.so`

- The version of the packages :
```
Package: ros-humble-depthai
Version: 2.24.0-1jammy.20240308.165504
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 21.1 MB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 4.2), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 12), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-humble-ros-workspace
Homepage: https://www.luxonis.com/
Download-Size: 10.9 MB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform

Package: ros-humble-depthai-ros
Version: 2.9.0-1jammy.20240415.142635
Priority: optional
Section: misc
Maintainer: sachin <sachin@luxonis.com>
Installed-Size: 43.0 kB
Depends: ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-depthai-descriptions, ros-humble-depthai-examples, ros-humble-depthai-filters, ros-humble-depthai-ros-driver, ros-humble-depthai-ros-msgs, ros-humble-ros-workspace
Download-Size: 6,292 B
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: The depthai-ros package

Package: ros-humble-depthai-bridge
Version: 2.9.0-1jammy.20240327.160327
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 475 kB
Depends: libc6 (>= 2.32), libgcc-s1 (>= 3.3.1), libopencv-calib3d4.5d (>= 4.5.4+dfsg), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgcodecs4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 11), libboost-dev, libopencv-dev, ros-humble-camera-info-manager, ros-humble-composition-interfaces, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-ros-msgs, ros-humble-image-transport, ros-humble-rclcpp, ros-humble-robot-state-publisher, ros-humble-ros-environment, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-stereo-msgs, ros-humble-tf2, ros-humble-tf2-geometry-msgs, ros-humble-tf2-ros, ros-humble-vision-msgs, ros-humble-xacro, ros-humble-ros-workspace
Download-Size: 129 kB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: The depthai_bridge package

Package: ros-humble-depthai-ros-msgs
Version: 2.9.0-1jammy.20240217.075403
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 1,723 kB
Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.3.1), libpython3.10 (>= 3.10.0), libstdc++6 (>= 11), ros-humble-fastcdr, ros-humble-builtin-interfaces, ros-humble-geometry-msgs, ros-humble-rclcpp, ros-humble-rosidl-default-generators, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 129 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: Package to keep interface independent of the driver

Package: ros-humble-depthai-ros-driver
Version: 2.9.0-1jammy.20240415.141623
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 2,811 kB
Depends: libc6 (>= 2.34), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 12), libtinyxml2-9 (>= 8.0.0), ros-humble-ament-cmake-auto, ros-humble-camera-calibration, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-depthai-descriptions, ros-humble-depthai-examples, ros-humble-depthai-ros-msgs, ros-humble-diagnostic-msgs, ros-humble-diagnostic-updater, ros-humble-image-pipeline, ros-humble-image-transport, ros-humble-image-transport-plugins, ros-humble-pluginlib, ros-humble-rclcpp, ros-humble-rclcpp-components, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-std-srvs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 621 kB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: Depthai ROS Monolithic node.
```

- The log when `rtabmap.launch.py` is launched :
[launch.log](https://github.com/luxonis/depthai-ros/files/15303399/launch.log)

- The pipeline graph:
[image]

Serafadam, 2024-05-17:
Hi, it seems that you have subpixel mode enabled (which sadly is not supported together with VideoEncoder). Could you share your parameters file?

Jenanaputra, 2024-05-18:
@Serafadam Hii !!
I looked again at my `rgbd.yaml` file, and you are absolutely right! . The` i_subpixel` parameter is enabled. As that parameter is enabled, it means that I need to make it to be disable , right ? 
 For additional information, I also share you my `rgbd.yaml` file :
```
 /oak:
  ros__parameters:
    camera:
      i_nn_type: none
    stereo:
      i_subpixel: true
```
The `camera.yaml` :
```
/oak:
  ros__parameters:
    camera:
      i_enable_imu: true
      i_enable_ir: true
      i_nn_type: spatial
      i_pipeline_type: RGBD
    nn:
      i_nn_config_path: depthai_ros_driver/mobilenet
```
and the `low_bandwidth.yaml` file :
```
/oak:
  ros__parameters:
    rgb:
      i_low_bandwidth: true
    left:
      i_low_bandwidth: true
    right:
      i_low_bandwidth: true
    stereo:
      i_low_bandwidth: true
```

Serafadam, 2024-05-21:
Hi, yes, [as per documentation](https://docs.luxonis.com/projects/api/en/latest/samples/VideoEncoder/disparity_encoding/?highlight=subpixel#disparity-encoding), unfortunately subpixel cannot be currently used with VideoEncoder.

Jenanaputra, 2024-05-26:
Thanks! It works right now

closed 2024-05-26 by Jenanaputra