luxonis/depthai-ros #530: [BUG] {Unable to connect to the camera with ros2 drivers}

makarandmandolkar, 2024-05-15:
I installed the ros2 drivers and tried to run the launch file, but unable to connect to the camera. Also, note that I am running this is docker.
I have no trouble running realsense cameras.
Please find the image for more details.
PS: The camera worked well in ros1.

[image]

Serafadam, 2024-05-17:
Hi, could you provide some more information according to [this template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)? Also, what command are you using to run the container?

Serafadam, 2024-06-17:
Closing due to inactivity, please reopen it if the issue still persists.

closed 2024-06-17 by Serafadam