luxonis/depthai-ros #534: [BUG] {Unable to get stable 30FPS for RGBD pipeline}

mjunsen123, 2024-05-24:
### Startup log:

> [component_container-1] [INFO] [1716546970.844394918] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[component_container-1] [INFO] [1716546970.864534402] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1716546970.864617096] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1716546970.881334717] [oak_state_publisher]: got segment oak
[component_container-1] [INFO] [1716546970.881431924] [oak_state_publisher]: got segment oak-d-base-frame
[component_container-1] [INFO] [1716546970.881441799] [oak_state_publisher]: got segment oak_imu_frame
[component_container-1] [INFO] [1716546970.881445986] [oak_state_publisher]: got segment oak_left_camera_frame
[component_container-1] [INFO] [1716546970.881450072] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[component_container-1] [INFO] [1716546970.881453733] [oak_state_publisher]: got segment oak_model_origin
[component_container-1] [INFO] [1716546970.881457480] [oak_state_publisher]: got segment oak_rgb_camera_frame
[component_container-1] [INFO] [1716546970.881461585] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[component_container-1] [INFO] [1716546970.881465270] [oak_state_publisher]: got segment oak_right_camera_frame
[component_container-1] [INFO] [1716546970.881468778] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak_state_publisher' in container 'oak_container'
[component_container-1] [INFO] [1716546970.884759779] [oak_container]: Load Library: /home/jyee/ros2_ws/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[component_container-1] [INFO] [1716546970.995873919] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1716546970.996002067] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1716546971.010616371] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [INFO] [1716546977.508486087] [oak]: Camera with MXID: 184430101152870E00 and Name: 1.1 connected!
[component_container-1] [INFO] [1716546977.512837047] [oak]: USB SPEED: SUPER
[component_container-1] [INFO] [1716546977.570541824] [oak]: Device type: OAK-D-PRO
[component_container-1] [INFO] [1716546977.580574291] [oak]: Pipeline type: RGBD
[component_container-1] [INFO] [1716546978.102017908] [oak]: Finished setting up pipeline.
[component_container-1] [INFO] [1716546978.861238018] [oak]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'

### Config

> /oak:
  ros__parameters:
    camera:
      i_enable_imu: true
      i_enable_ir: false
      i_pipeline_type: RGBD
      i_nn_type: none
    imu:
      i_acc_freq: 500
      i_batch_report_threshold: 5
      i_gyro_freq: 400
      i_max_batch_reports: 20
      i_sync_method: LINEAR_INTERPOLATE_GYRO
      i_message_type: IMU
    rgb:
      i_fps: 30.0
      i_publish_topic: true
      i_resolution: 1080P
      i_low_bandwidth: true
      i_low_bandwidth_quality: 50
      i_output_isp: false #change this accordingly
      i_width: 1280
      i_height: 720
      # i_isp_den: 3
      # i_isp_num: 2
    stereo:
      i_align_depth: true 
      i_publish_topic: true
      i_output_disparity: false
      i_fps: 30.0
      i_low_bandwidth: true
      i_low_bandwidth_quality: 50
      # i_reverse_stereo_socket_order: true

### FPS 

> average rate: 14.050
        min: 0.022s max: 0.367s std dev: 0.05225s window: 348
average rate: 14.223
        min: 0.022s max: 0.367s std dev: 0.05150s window: 367
average rate: 14.219
        min: 0.022s max: 0.367s std dev: 0.05126s window: 382
average rate: 14.051
        min: 0.022s max: 0.367s std dev: 0.05192s window: 393
average rate: 13.895
        min: 0.022s max: 0.367s std dev: 0.05304s window: 403
average rate: 14.017
        min: 0.022s max: 0.367s std dev: 0.05251s window: 421
average rate: 14.032
        min: 0.022s max: 0.367s std dev: 0.05195s window: 436

### Further Information
I also found that RVIZ:=true keep crashing in RGBD pipeline but working on Depth pipeline. In addtion, the FPS get even worse in ISP mode.

Serafadam, 2024-05-28:
Hi, what topic are you checking for FPS? Usually in ROS2 it's better to check hz of camera info topic rather than image due to serialization overhead in ros2 topic. Additionally, what hardware are you running on? Could you provide some more information as presented in [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)?

mjunsen123, 2024-06-10:
Hi @Serafadam Sorry for late reply. I can confirm that camera info topic achieve stable 30 fps. Thanks

closed 2024-06-10 by mjunsen123