luxonis/depthai-ros #536: [Feature-Request] Respawning composed node in rgbd_plc.launch.py, restarting from failed states.

TheMarksniper, 2024-05-28:
Why:
The composable nodes in rgbd_plc.launch.py (same goes for rtabmap.launch.py) do not have any way to get out of critical failure states, for example:
X_LINK_ERROR and ping missed messages which seemingly break the pipeline.
[image]
 It would be nice if there was a way to kill the nodes and restart the pipeline since ros2 does not support node killing like ros1 did.

What:
A parameter that would allow (composable)nodes to exit (and start up) gracefully after critical error.

How:
Maybe before resulting to straight up killing the nodes, a lifecycle nodes could be implemented and just the pipeline could be destroyed (in on_cleanup()) and restored (in on_configuration()). Otherwise, just exit() in critical failed state, and parameter for respawn in the launcher would be also doable.

Serafadam, 2024-05-31:
Hi, unfortunately Lifecycle nodes are not supported by image_transport package so unfortunately they cannot be used by camera drivers. There is a [watchdog/restart mechanism](https://github.com/luxonis/depthai-ros/blob/0144d7988ae9b981ae867b133934188628c8e950/depthai_ros_driver/src/param_handlers/camera_param_handler.cpp#L38) that you can enable via parameter, but in some edge cases it might fail.

TheMarksniper, 2024-06-03:
Thank you for that restart mechanism, i guess nothing more can be done about it now

closed 2024-06-03 by TheMarksniper