luxonis/depthai-ros #537: [BUG] Multiple cameras on same nodelet generate segfault/crash (ROS Noetic)

imwhocodes, 2024-05-29:
**Describe the bug**
If you try to load 2 cameras node on the same nodelet after some seconds ROS do crash

I saw multiple causes in the log, some of them are:
- double free()
- corrupted linked list
- malloc() failed in some way

All error seem caused by some concurrency problem on managing allocation / deallocations

**Minimal Reproducible Example**
My setup is based on this:
1. docker-image: osrf/ros:noetic-desktop-full
2. depthai-ros: apt install ros-noetic-depthai-ros

The minimum launch file to reproduce is:
```xml
<?xml version="1.0"?>
<launch>

  <node pkg="nodelet" type="nodelet" name="standalone_nodelet"  args="manager" output="screen"/>

  <group ns="OAKD_A">
    <node name="camera" pkg="nodelet" type="nodelet" output="screen" required="true" args="load depthai_ros_driver/Camera /standalone_nodelet">
      <rosparam param="camera_i_mx_id">SERIAL-CAMERA-A</rosparam>
    </node>
  </group>

  <group ns="OAKD_B">
    <node name="camera" pkg="nodelet" type="nodelet" output="screen" required="true" args="load depthai_ros_driver/Camera /standalone_nodelet">
      <rosparam param="camera_i_mx_id">SERIAL-CAMERA-B</rosparam>
    </node>
  </group>
  
</launch>
```

**Expected behavior**
You should be able to attach/load multiple camera-nodelet to the same nodelet-manager 

**Additional context**
With nodelet (if attached to the same manager) you don't pay extra cpu price when moving data from a node to the next of a given pipeline
This have a big impact when working with point-cloud (given they enormous message size) where from one node to the next you simply move a smart-pointer around without serialising and deserialising the data at each step
In my setup I'm acquiring multiple point cloud from multiple oak-d-pro-w, merging them and applying multiple steps of processing, being unable to exploit the memory sharing of the nodelet architecture is a big performance hit

Thanks,
Luca

imwhocodes, 2024-05-29:
I saw on the forum that the call to the various call to dai::Device and dai::Pipeline are not thread-safe

So I tried (very crudely) to mitigate it by getting sharing a lock between all Camera nodelet and by acquiring it for ANY function or callback present in Camera.cpp

As seen here:
https://github.com/luxonis/depthai-ros/compare/noetic...imwhocodes:depthai-ros:noetix-multicam-nodelet-fix

But either I'm doing something wrong or this don't help either
The nodelet-manager is still crashing after a couple of second of having both camera connected, with some new error sometimes

MartinMotycka, 2024-06-02:
@Serafadam Please check this asap. Thank you. M.

Serafadam, 2024-06-03:
@imwhocodes  Thanks for the report, it seems that the issue probably originates from [IMU interpolation](https://github.com/luxonis/depthai-ros/blob/noetic/depthai_bridge/include/depthai_bridge/ImuConverter.hpp), you can switch try checking if changing sync method to COPY helps in meantime.
```xml
        <rosparam param="imu_i_sync_method">COPY</rosparam>
```

imwhocodes, 2024-06-05:
> @imwhocodes Thanks for the report, it seems that the issue probably originates from [IMU interpolation](https://github.com/luxonis/depthai-ros/blob/noetic/depthai_bridge/include/depthai_bridge/ImuConverter.hpp), you can switch try checking if changing sync method to COPY helps in meantime.
> 
> ```
>         <rosparam param="imu_i_sync_method">COPY</rosparam>
> ```

Thanks, this worked out!