luxonis/depthai-ros #540: [BUG] depthai_bridge fails to build on Noetic: spdlog recompile with -fPIC unsupported relocation

sloretz, 2024-06-04:
**Check if issue already exists**

Did not find an existing issue

**Describe the bug**

depthai_bridge fails to build on the ROS Noetic buildfarm on all architectures, and has been doing so since May 31, 2024. Possibly caused by update to depthai? https://github.com/ros/rosdistro/pull/41439

The build failure appears to complain about symbols from spdlog being statically linked:

```
23:05:03 /usr/bin/ld: /opt/ros/noetic/lib/x86_64-linux-gnu/cmake/depthai/dependencies/lib/libspdlog.a(spdlog.cpp.o): relocation R_X86_64_TPOFF32 against `_ZGVZN6spdlog7details2os9thread_idEvE3tid' can not be used when making a shared object; recompile with -fPIC
23:05:03 /usr/bin/ld: /opt/ros/noetic/lib/x86_64-linux-gnu/libdepthai-core.a(CalibrationHandler.cpp.o): relocation R_X86_64_PC32 against symbol `_ZGVZN3dai7Logging11getInstanceEvE7logging' can not be used when making a shared object; recompile with -fPIC
23:05:03 /usr/bin/ld: final link failed: bad value
```

**Minimal Reproducible Example**

https://build.ros.org/job/Nbin_uF64__depthai_bridge__ubuntu_focal_amd64__binary/61/console

**Expected behavior**

The build succeeds on the Noetic buildfarm

**Screenshots**
N/A

**Pipeline Graph**

N/A

**Attach system log**

N/A

**Additional context**

* amd64: https://build.ros.org/job/Nbin_uF64__depthai_bridge__ubuntu_focal_amd64__binary/61/console
* armv8 https://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__depthai_bridge__ubuntu_focal_arm64__binary/65/console
* armhf https://build.ros.org/view/Nbin_ufhf_uFhf/job/Nbin_ufhf_uFhf__depthai_bridge__ubuntu_focal_armhf__binary/64/

Serafadam, 2024-06-05:
@sloretz Thanks for the report, it seems a small change has been omitted which removed arg for building shared libs (although shouldn't buildfarm set it by default anyway?).
To fix that, should I create a new tag under the same name, or bump the release version?

sloretz, 2024-06-05:
> To fix that, should I create a new tag under the same name, or bump the release version?

Please bump the release version.

chrismcclurg, 2024-06-11:
@sloretz Was there a workaround for this issue?

Serafadam, 2024-06-13:
New version of `depthai` with correct flags has been pushed to rosdistro.

gongbingyu, 2024-06-14:
> New version of `depthai` with correct flags has been committed to rosdistro.

How do I find it?

CharithReddy101, 2024-06-14:
> > New version of `depthai` with correct flags has been committed to rosdistro.
> 
> How do I find it?

Waiting for the same!!

anshu3012, 2024-06-14:
> New version of `depthai` with correct flags has been pushed to rosdistro.

Is it the version 2.26.1? The latest version available through apt is 2.26.0. It would be great if you could please outline steps to fix this issue. 

Thanks!

anshu3012, 2024-06-15:
I was able to build the package by installing v2.24 for depthai-core. 
To build : 
1. Instead of running the dependency script from the website, make a local script with the following:
`set -e

echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
sudo udevadm control --reload-rules && sudo udevadm trigger
cd /tmp
git clone --recursive https://github.com/luxonis/depthai-core.git --branch v2.24.0
cmake -Hdepthai-core -Bdepthai-core/build -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=/usr/local
cmake --build depthai-core/build --target install
cd /tmp
rm -r depthai-core
`
2. Clone v2.9 of depthai-ros 
3. catkin build

gongbingyu, 2024-06-17:
> @gongbingyu @CharithReddy101 @chrismcclurg @sloretz
> 
> I was able to build the package by installing v2.24 for depthai-core. Not ideal but seems to work. I will outline the steps hoping it helps while the issue is resolved.
> 
> 1. Instead of running the [dependency script from the website](https://docs.luxonis.com/software/ros/depthai-ros/build/), make a local script with the following:
> 
> ```
> set -e
> 
> echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
> sudo udevadm control --reload-rules && sudo udevadm trigger
> cd /tmp
> git clone --recursive https://github.com/luxonis/depthai-core.git --branch v2.24.0
> cmake -Hdepthai-core -Bdepthai-core/build -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=/usr/local
> cmake --build depthai-core/build --target install
> cd /tmp
> rm -r depthai-core
> ```
> 
> 2. clone v2.9 of depthai-ros
> 3. catkin build

Do I need to remove my previous installation of depthai-core. If so, how.

tetov, 2024-06-17:
@anshu3012 
> Do I need to remove my previous installation of depthai-core. If so, how.

```
sudo dpkg -r --force-depends ros-noetic-depthai
```

You can remove it with this.

rosdep will reinstall it though, but you can get away from that using

```
rosdep install --from-paths src --ignore-src -y --skip-keys=depthai
``

ZenoRobotics, 2024-06-17:
I wasted a lot of time this weekend trying to figure out why ros-noetic-depthia-ros disappeared and why I was getting such a strange error after reflashing my microSD card and reloaded depthai ros. Chased a lot of red herrings along the way, then repeating the reflash/rebuild process several times. It would be nice if this information wasn't buried here, but was somehow sent to the user. You need to change your instruction page at: https://docs.luxonis.com/software/ros/depthai-ros/ . This issue rendered my use of the Oak camera useless, so I then went back to my RS D435i. Ugh!!!

tetov, 2024-06-17:
The bug was fixed here: https://github.com/ros/rosdistro/pull/41594

Now it needs to make it's way through the ROS release pipeline from testing to stable.

@Serafadam: Thanks for the quick response! Next time, please give a hint which project's CMake file to fix ;).

gongbingyu, 2024-06-18:
Will there be any intention to release ros-noetic-depthai-ros?

gongbingyu, 2024-06-18:
I also specifically agree with his proposal (@ZenoRobotics)  to update the instructions [in]( https://docs.luxonis.com/software/ros/depthai-ros/) the documentation. It is also suggested to add a specific step for removing the depthai-core installer (both source builds, binary builds, and scripted builds), which would be especially helpful for beginners.

tetov, 2024-06-18:
Until new ros-noetic-depthai and ros-noetic-depthai-ros makes its way to the stable ros apt repo you can just change

```
deb http://packages.ros.org/ros/ubuntu focal main
```

in `/etc/apt/sources.list.d/ros-latest.list` to
 
```
#deb http://packages.ros.org/ros/ubuntu focal main
deb http://packages.ros.org/ros-testing/ubuntu focal main
```

After that you can run:

```bash
sudo apt update && sudo apt install ros-noetic-depthai-ros
```
```

sloretz, 2024-06-20:
depthai packages are back in ROS Noetic

* https://repositories.ros.org/status_page/ros_noetic_default.html?q=depthai
* https://discourse.ros.org/t/new-packages-for-noetic-2024-06-20/38278

closed 2024-06-20 by sloretz