luxonis/depthai-ros #542: [BUG] Humble ros_bridge does spatial detection missing fields

kikass13, 2024-06-08:
**Describe the bug**
Humble implementation of the ros msg converter does not provide a score value for spatial detections

**Expected behavior**
The score value should be set into the ros msg

**Additional context**
[CLICK ME - The Line in code where the fix should be added](https://github.com/luxonis/depthai-ros/blob/humble/depthai_bridge/src/SpatialDetectionConverter.cpp#L69)

something like this should be added in the `IS_HUMBLE` ifdef

```
opDetectionMsg.detections[i].results[0].score = inNetData->detections[i].confidence;
```

which results in a ros2 topic echo like this:

```---
header:
  stamp:
    sec: 1717864050
    nanosec: 427740811
  frame_id: oak_rgb_camera_optical_frame
detections:
- results:
  - class_id: '0'
    score: 8.31591796875
  bbox:
    center:
      position:
        x: 78.0
        y: 250.0
      theta: 0.0
    size_x: 200.0
    size_y: 252.0
  position:
    x: -0.19737829267978668
    y: -0.07760991901159286
    z: 0.5516734719276428
  is_tracking: false
  tracking_id: ''
---
```

jide07, 2024-07-25:
Hello @kikass13 i have no comment of the bug I just have a question on how do I run the bridge files 
I want to do SpatialDetection using the yolov4_spatial_publisher.cpp example which I am able to run with the launch file,
Now I want to convert the topic been sent from the node and I believe that's what TrackSpatialDetectionConverter.cpp is for
however I can figure how to run it, can you please explain how I can?

sagarvl96, 2024-07-25:
Any idea on how to publish the score in the meantime?

Serafadam, 2024-07-29:
Hi, thanks for the report, the fix is available [here](https://github.com/luxonis/depthai-ros/tree/spatial_detection_fields_humble) and will be merged soon

sagarvl96, 2024-08-07:
Hello @Serafadam!
Was the fix merged to the humble branch?