luxonis/depthai-ros #543: [BUG] Missing depthai-ros packages on Noetic

ptoupas, 2024-06-13:
Hi,

I am having some issue installing depthai-ros on Ubuntu 22.04. It seems like there has been some recent update and some packages of the ros-noetic-depthai-* are missing and cannot be installed. I can only install some of them (but the main ones seems to be missing like depthai-ros for example or the driver)
I had previously installed them at the same OS some time ago but it seems like I cant do this anymore.

This issue is also probably related to issue #540

Serafadam, 2024-06-24:
Hi, yes, it the new release of depthai broke builds for depthai-ros (this is fixed now), but according to the latest release post they should be back (and they aren't, respective issue created in rosdistro https://github.com/ros/rosdistro/issues/41775)

ptoupas, 2024-07-02:
Thanks a lot. It seems to be fixed now!

closed 2024-07-02 by ptoupas