luxonis/depthai-ros #551: [BUG] WLS filtered image all black

flabrosse, 2024-06-28:
Ubuntu 22.04, ROS humble, OAK-D S2.

I'm trying `example_wls_filter.launch.py` from the `depthai_filters` package.  All I get is a black image.  Nothing that looks like an error in the output.  All the other images are there, but the filtered one is black, but published at nearly 20Hz.

Without filtering the camera is close to useless with rtabmap as there is way too much noise.  What am I missing to make the wls filtering work?

flabrosse, 2024-07-15:
I installed everything from sources, to get the latest version of everything and now nothing gets published on the topic `/wls_filtered`. Any ideas?

flabrosse, 2024-07-15:
Could this be related:
```
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'rectify_color_node' of type 'image_proc::RectifyNode' in container 'oak_container': Could not find requested resource in ament index
```

Serafadam, 2024-07-16:
Hi, @flabrosse  thanks for the report and sorry for the delay, could you test if this [branch works for](https://github.com/luxonis/depthai-ros/tree/wls_fix_humble) you? Also, even though WLS filter "smooths" the depth readings it is also susceptible to distortions so I am not sure if it will be good enough for usage with RTABMap

flabrosse, 2024-07-16:
Same.  Nothing published on `/wls_filtered`.

Noise is definitely an issue.  I see many people reporting this, in particular with more noise from the ROS driver.  I am still playing with the settings, and also hope that it'll work better outside.