luxonis/depthai-ros #552: rtabmap.launch Did not receive data since 5 seconds!

whaudrms, 2024-06-28:
Hello, I am having trouble running depthai-ros-driver/rtabmap.launch. 
It seems to be the same problem as #266. How do I resize the image?

```
SUMMARY
========

PARAMETERS
 * /oak/camera_i_base_frame: oak-d_frame
 * /oak/camera_i_cam_pitch: 0.0
 * /oak/camera_i_cam_pos_x: 0.0
 * /oak/camera_i_cam_pos_y: 0.0
 * /oak/camera_i_cam_pos_z: 0.0
 * /oak/camera_i_cam_roll: 0.0
 * /oak/camera_i_cam_yaw: 0.0
 * /oak/camera_i_imu_from_descr: False
 * /oak/camera_i_nn_type: none
 * /oak/camera_i_parent_frame: oak-d-base-frame
 * /oak/camera_i_publish_tf_from_calibration: False
 * /oak/robot_description: <?xml version="1....
 * /oak/stereo_i_subpixel: True
 * /rgbd_odometry/approx_sync: True
 * /rgbd_odometry/approx_sync_max_interval: 0.02
 * /rgbd_odometry/frame_id: oak
 * /rgbd_odometry/subscribe_depth: True
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /rtabmap/Rtabmap/DetectionRate: 3.5
 * /rtabmap/frame_id: oak

NODES
  /
    oak (nodelet/nodelet)
    oak_nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rgbd_odometry (rtabmap_odom/rgbd_odometry)
    rtabmap (rtabmap_slam/rtabmap)
    rtabmap_viz (rtabmap_viz/rtabmap_viz)

ROS_MASTER_URI=http://localhost:11311

process[oak_state_publisher-1]: started with pid [18058]
process[oak_nodelet_manager-2]: started with pid [18059]
process[oak-3]: started with pid [18060]
process[rgbd_odometry-4]: started with pid [18061]
process[rtabmap-5]: started with pid [18062]
process[rtabmap_viz-6]: started with pid [18067]
[ INFO] [1719594593.359192573]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1719594593.359832014]: waitForService: Service [/oak_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1719594593.366278997]: Initializing nodelet with 24 worker threads.
[ INFO] [1719594593.380180162]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[ INFO] [1719594593.425088919]: No ip/mxid/usb_id specified, connecting to the next available device.
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1719594596.260823296]: Camera with MXID: 1844301071F8C71200 and Name: 1.9.3 connected!
[ INFO] [1719594596.261556273]: USB SPEED: HIGH
[ INFO] [1719594596.311904298]: Device type: OAK-D-PRO
[ INFO] [1719594596.316344235]: Pipeline type: RGBD
[ INFO] [1719594597.186352595]: Finished setting up pipeline.
[ WARN] [1719594597.772343558]: /rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rgbd_odometry subscribed to (approx sync, max interval=0.020000s):
   /oak/rgb/image_raw \
   /oak/stereo/image_raw \
   /oak/rgb/camera_info
[ WARN] [1719594597.842605201]: /rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=10 and 10 respectively).
/rtabmap subscribed to (approx sync):
   /odom \
   /oak/rgb/image_raw \
   /oak/stereo/image_raw \
   /oak/rgb/camera_info
[ WARN] [1719594599.272825465]: /rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=10 and 10 respectively).
/rtabmap_viz subscribed to:
   /odom
[ INFO] [1719594599.432144438]: Camera ready!

```
The results of rostopic echo are as follows:
```
subscribed to [/oak/rgb/image_raw]
average rate: 12.376
	min: 0.037s max: 0.091s std dev: 0.01480s window: 11
```

I read #531 and changed rtabmap.launch
```
<?xml version="1.0"?>
<launch>
    <arg name="name" default="oak" />
    <arg name="params_file" default="$(find depthai_ros_driver)/config/rgbd.yaml" />

    <include file="$(find depthai_ros_driver)/launch/camera.launch">
        <arg name="name" value="$(arg name)"/>
        <arg name="params_file" value="$(arg params_file)"/>
    </include>
    <node type="rgbd_odometry" name="rgbd_odometry" pkg="rtabmap_odom">
        <remap from="rgb/image" to="$(arg name)/rgb/image_raw"/>
        <remap from="rgb/camera_info" to="$(arg name)/rgb/camera_info"/>
        <remap from="depth/image" to="$(arg name)/stereo/image_raw"/>
        <param name="frame_id" type="string" value="$(arg name)"/>

        <rosparam param="subscribe_depth">True</rosparam>
        <rosparam param="approx_sync">True</rosparam>
        <rosparam param="approx_sync_max_interval">0.02</rosparam>
        
    </node>
    <node type="rtabmap" name="rtabmap" pkg="rtabmap_slam">
        <remap from="rgb/image" to="$(arg name)/rgb/image_raw"/>
        <remap from="rgb/camera_info" to="$(arg name)/rgb/camera_info"/>
        <remap from="depth/image" to="$(arg name)/stereo/image_raw"/>
        <rosparam param="Rtabmap/DetectionRate">3.5</rosparam>
        <param name="frame_id" type="string" value="$(arg name)"/>

    </node>
    <node type="rtabmap_viz" name="rtabmap_viz" pkg="rtabmap_viz">
        <remap from="rgb/image" to="$(arg name)/rgb/image_raw"/>
        <remap from="rgb/camera_info" to="$(arg name)/rgb/camera_info"/>
        <remap from="depth/image" to="$(arg name)/stereo/image_raw"/>
    </node>

</launch>
```

Could you please help me? Thank you for reading.

closed 2024-06-29 by whaudrms

gongbingyu, 2024-08-19:
I also have the same problem

Serafadam, 2024-08-20:
Hi, setting autoexposure/focus to fixed values can help with RGBD odometry, as well as setting autoreset for Odometry via Parameter interface

gongbingyu, 2024-08-20:
> 嗨，将自动曝光/对焦设置为固定值可以帮助 RGBD 里程计，以及通过参数界面设置里程计的自动重置
Thank you very much for your reply. Could you please tell me where to set it specifically?

Serafadam, 2024-08-20:
Autoexposure/focus settings can be set via parameter interface, you can find list of parameters [here](https://docs.luxonis.com/software/ros/depthai-ros/driver/#DepthAI%20ROS%20Driver-List%20of%20parameters) (for ROS2, ROS1 has a little bit different syntax that is explained in the documentation). For RTABMap parameters, you can refer to [this list](https://github.com/introlab/rtabmap/blob/6ce85b4e55adc2903a6f08dd25fa0c894b34c9cd/corelib/src/Parameters.cpp#L240)