luxonis/depthai-ros #554: Camera calibration interpretation

abaeyens-imod, 2024-07-01:
**Check if issue already exists**
[Here's the same issue on the Luxonis forum](https://discuss.luxonis.com/d/4706-oak-d-pro-wide-ros-camera-info-calibration-interpretation/3).

**Describe the bug**
Applying the calibration published in the CameraInfo message does not deliver sensible results. Recalibrating the camera using standard tools resulted in a very different calibration which does work, but preferably we would like to use what gets published on the camera_info topics.

**Minimal Reproducible Example**
[Here's an MRE, including a bag file](https://drive.google.com/file/d/16iPKlz2XGj4kxqKiVL9WojD369DRzDQU/view?usp=sharing). It uses a marker detection library to generate a set of 3D points from the right camera image and then projects these back onto image space for all three cameras.

Key code used for projecting 3D points to image coordinates using the Luxonis calibration:
```c++
void projectMarkersToImageCoordinates(
    const std::vector<stag_ros::STagMarker> markers, 
    const sensor_msgs::CameraInfo & camera_info,
    const cv::Point3d & camera_position,
    std::vector<cv::Point2d> & uvs) {
  // Ensure output is empty
  uvs.clear();
  // Nothing to project => exit
  if (markers.empty()) return;
  // Prepare input
  std::vector<cv::Point3d> object_points;
  for (const stag_ros::STagMarker& m : markers) {
    object_points.push_back(cv::Point3d(
      m.pose.position.x - camera_position.x,
      m.pose.position.y - camera_position.y,
      m.pose.position.z - camera_position.z));
  }
  cv::Mat r(3, 3, CV_64F);
  for (int i = 0; i < 9; ++i) r.at<double>(i) = camera_info.R[i];
  cv::Mat rvec;
  cv::Rodrigues(r, rvec);
  cv::Mat tvec = cv::Mat::zeros(cv::Size(3, 1), CV_64F);
  cv::Mat k(3, 3, CV_64F);
  for (int i = 0; i < 9; ++i) k.at<double>(i) = camera_info.K[i];
  // Assume distortion model follows OpenCV format
  cv::Mat d(camera_info.D.size(), 1, CV_64F);
  for (int i = 0; i < camera_info.D.size(); ++i) d.at<double>(i) = camera_info.D[i];
  // The actual 3D => 2D projection
  cv::projectPoints(object_points, rvec, tvec, k, d, uvs);
}
```

**Expected behavior**
The centers of the red circles align with the fiducial marker centers in all three images in the screenshot below.

**Observed behavior**
The centers align in the image of the right camera, indicating that the projections of the made node (see the linked MRE code) match those of the marker detection library. This is good. However, the centers don't align in the images of the left and the rgb camera, indicating a severe calibration (interpretation) issue.

**Screenshots**
[image]

**Pipeline Graph**
Standard pipeline `RGBStereo`.

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)
 - Which OS/OS version are you using? **Ubuntu 20.04/Focal container on Ubuntu 22.04/Jammy**
 - Which ROS version are you using? **ROS 1**
 - Which ROS distribution are you using ? **Noetic**
 - Is `depthai-ros` built from source or installed from apt? **Occurs with both**
 - Is `depthai/depthai-core` library installed from rosdep or manually? For rosdep install, check if `ros-<rosdistro>-depthai` package is installed, manual install can be checked with `ldconfig -p | grep depthai`. **ros-noetic-depthai is installed**
 - Please include versions of following packages - `apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver`

**Versions:**
| package | version |
| -- | -- |
| ros-noetic-depthai | 2.24.0-2focal.20240308.133037 |
| ros-noetic-depthai-ros | 2.9.0-1focal.20240402.150922 |
| ros-noetic-depthai-bridge | 2.9.0-1focal.20240308.134342 |
| ros-noetic-depthai-ros-msgs | 2.9.0-1focal.20240125.202559 |
| ros-noetic-depthai-ros-driver | 2.9.0-1focal.20240402.145653 |

 - To get additional logs, set `DEPTHAI_DEBUG=1` and paste the logs, either from command line or from latest log in `~/.ros/log`

**Additional context**
Thanks for taking your time to look into this.

Shivam7Sharma, 2024-07-02:
Did you use the Luxonis ROS package for calibrating? You can also get the calibration data from the depthai SDK using the calibrate.py file. It will save a file of calibration data if used correctly.

abaeyens-imod, 2024-07-03:
> Did you use the Luxonis ROS package for calibrating?

This is with the factory calibrations published on the `*camera_info` topics. I'll edit the initial post to clarify this.

Shivam7Sharma, 2024-07-03:
I am not associated with Luxonis. I am just trying to help you. Are you using a FFC board? If you calibrate using the file I suggested, then that code will flash the device memory, I think, and the ROS messages will show the updated calibration data.

abaeyens-imod, 2024-07-05:
I do appreciate your input; we can calibrate the camera, yet would prefer to use the factory calibration as recalibrating a camera adds a considerable cost to each camera. However, unlike with the Intel RealSense cameras, the factory calibration simply doesn't work out for us.
We so far observed the issue with the OAK-D Pro (IMX version), OAK-D Pro Wide (IMX + OV version) and the OAK-D S2 (IMX version).

abaeyens-imod, 2024-11-04:
Hi all, concerning the calibration, I just re-tested and this issue still occurs. [Here you find an updated MRE to reproduce this behavior yourself](https://drive.google.com/file/d/1dyX4vNG2zXFVwRbpq2rP8GVBcPodfT0i/view?usp=sharing) with one of your Luxonis cameras. Be sure to let know if you have any questions (if desired we can have a video call).

In RQt you can visualize the left, right and rgb images with markers (red circles), and we observe the following:
- right image: circles perfectly concentric (good)
- left image: not all circles concentric (bad, indicates calibration (interpretation) issue), example:
   [image]
- rgb image: similar story as for the left image, though the error is usually larger

`ros-noetic-depthai-ros` version: `2.10.3-1focal.20241015.133934`

@Serafadam I'd appreciate if you could have a second look at this.

Serafadam, 2024-11-08:
Regarding video call, please send me your schedule for the next week at adam.serafin@luxonis.com

abaeyens-imod, 2024-11-21:
Thanks for having the call yesterday. Something I'd like to add, which may help you in narrowing down the issue, is that we also observe unexpected behavior of the reported focal length.

When we run our OAK-D Wide RGB camera in 12MP mode and a 2/3 ISP downscale, the `*/rgb/camera_info` topic reports `K: [1605.0543212890625, 0.0, 1310.5240478515625, 0.0, 1604.3280029296875, 1010.4443359375, 0.0, 0.0, 1.0]` (I've attached the full messages below), i.e. a focal length of about **1604 px**. However, when running it in 4K mode and a 2/3 ISP downscale, it reports `K: [1528.623291015625, 0.0, 1248.1181640625, 0.0, 1527.9315185546875, 722.3280029296875, 0.0, 0.0, 1.0]`, i.e. a focal length of about **1528 px**. The reported focal lengths for these two settings should be similar as the latter image is just a crop of the former image, which was checked visually by overlaying an image of each mode over each other (both shown at scale 1:1, of course) which showed a perfect match. Also, when we ran a calibration on each of the modes we obtained practically the same focal length for both.

`CameraInfo` message when running ik 12MP mode and a 2/3 ISP downscale:
```
---
header: 
  seq: 9
  stamp: 
    secs: 1732181054
    nsecs: 935425474
  frame_id: "oak_rgb_camera_optical_frame"
height: 2016
width: 2688
distortion_model: "rational_polynomial"
D: [10.915542602539062, 11.508681297302246, 0.00045586744090542197, -0.00014918891247361898, -2.244452714920044, 10.899931907653809, 14.94647216796875, -1.498024821281433]
K: [1605.0543212890625, 0.0, 1310.5240478515625, 0.0, 1604.3280029296875, 1010.4443359375, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [1605.0543212890625, 0.0, 1310.5240478515625, 0.0, 0.0, 1604.3280029296875, 1010.4443359375, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```
`CameraInfo` message when running ik 4K mode and a 2/3 ISP downscale:
```
---
header: 
  seq: 7
  stamp: 
    secs: 1732181129
    nsecs: 434931008
  frame_id: "oak_rgb_camera_optical_frame"
height: 1440
width: 2560
distortion_model: "rational_polynomial"
D: [10.915542602539062, 11.508681297302246, 0.00045586744090542197, -0.00014918891247361898, -2.244452714920044, 10.899931907653809, 14.94647216796875, -1.498024821281433]
K: [1528.623291015625, 0.0, 1248.1181640625, 0.0, 1527.9315185546875, 722.3280029296875, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [1528.623291015625, 0.0, 1248.1181640625, 0.0, 0.0, 1527.9315185546875, 722.3280029296875, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```