luxonis/depthai-ros #560: [BUG] Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

trocolit2, 2024-07-14:
Dear all, 

I am facing an issue with running the ROS2 driver for the Humble version.  The driver software was installed via ROS2 Humble repository, which has the deptai_ros_driver version 2.26.
The is the OAK-D-LR, with Bootloader version 0.028.
The OS is the Ubuntu 22.04. 

After I ran the launch from the deptai_ros_driver, I got the following result:
```
$ ros2 launch depthai_ros_driver camera.launch.py 
[INFO] [launch]: All log files can be found below /home/ros2/.ros/log/2024-07-14-16-12-04-268345-thinkpad-70120
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [70134]
[component_container-1] [INFO] [1720959124.896794941] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[component_container-1] [INFO] [1720959124.902123062] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1720959124.902184528] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1720959124.909118012] [oak_state_publisher]: got segment oak
[component_container-1] [INFO] [1720959124.909140517] [oak_state_publisher]: got segment oak-d-base-frame
[component_container-1] [INFO] [1720959124.909147900] [oak_state_publisher]: got segment oak_imu_frame
[component_container-1] [INFO] [1720959124.909153298] [oak_state_publisher]: got segment oak_left_camera_frame
[component_container-1] [INFO] [1720959124.909158666] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[component_container-1] [INFO] [1720959124.909163688] [oak_state_publisher]: got segment oak_model_origin
[component_container-1] [INFO] [1720959124.909168640] [oak_state_publisher]: got segment oak_rgb_camera_frame
[component_container-1] [INFO] [1720959124.909173957] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[component_container-1] [INFO] [1720959124.909179045] [oak_state_publisher]: got segment oak_right_camera_frame
[component_container-1] [INFO] [1720959124.909184054] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-1] [INFO] [1720959124.910413550] [oak_container]: Load Library: /opt/ros/humble/lib/libdepthai_ros_driver.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak_state_publisher' in container 'oak_container'
[component_container-1] [INFO] [1720959124.947646296] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1720959124.947682938] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1720959124.955290283] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [ERROR] [1720959132.608889113] [oak]: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
[component_container-1] [INFO] [1720959132.609003490] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [2024-07-14 16:12:15.691] [depthai] [warning] Skipping X_LINK_BOOTLOADER device with name "1.7" ()
[component_container-1] [ERROR] [1720959136.707464666] [oak]: No available devices
[component_container-1] [INFO] [1720959136.707616192] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [ERROR] [1720959140.775311768] [oak]: No available devices
[component_container-1] [INFO] [1720959140.775454914] [oak]: No ip/mxid specified, connecting to the next available device.
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)

```
I suspected the udev rules were not correct, however, I cloned the repository from the depthai-python, and tested the [utility](https://github.com/luxonis/depthai-python/tree/main/utilities) python scripts _device_manager.py_ and the _cam_test.py_. From these scripts, I could access the camera information and acquire data. It proves the udev is correct. 

[image]

[image]

Terminal Output from the _cam_test.py_

```
./cam_test.py 
DepthAI version: 2.27.0.0
DepthAI path: /home/ros2/.local/lib/python3.10/site-packages/depthai.cpython-310-x86_64-linux-gnu.so
Enabled cameras:
   cama : color
   camb : color
   camc : color
CAM:  cama
CAM:  camb
CAM:  camc
[19443010616C742700] [1.7] [1.096] [ColorCamera(7)] [warning] Unsupported resolution set for detected camera AR0234, needs 1200_P, defaulting to it
[19443010616C742700] [1.7] [1.097] [ColorCamera(5)] [warning] Unsupported resolution set for detected camera AR0234, needs 1200_P, defaulting to it
[19443010616C742700] [1.7] [1.097] [ColorCamera(3)] [warning] Unsupported resolution set for detected camera AR0234, needs 1200_P, defaulting to it
Connected cameras:
 -socket CAM_A : AR0234 1920 x 1200 focus:fixed - COLOR
 -socket CAM_B : AR0234 1920 x 1200 focus:fixed - COLOR
 -socket CAM_C : AR0234 1920 x 1200 focus:fixed - COLOR
USB speed: SUPER
IR drivers: []
Cam:      cama          camb          camc     [host | capture timestamp]
FPS:  18.42| 18.00  18.17| 18.00  18.67| 18.37

```

I dug the source code the line where the error rises, and I found this [line](https://github.com/luxonis/depthai-core/blob/90487456cb92d15bcf0129a2f54c332250f1df31/src/xlink/XLinkConnection.cpp#L472)

I played with the bootupTimeout (WAIT_FOR_BOOTUP_TIMEOUT) values, but increasing or decreasing the time did not affect the error.

I also monitored the **lsusb** to see the device boot and when the device is not running, the values are:
``` Bus 001 Device 043: ID 03e7:f63c Intel Luxonis Bootloader ```
When I run the ROS driver, the devices change name to 
``` Bus 002 Device 019: ID 03e7:f63b Intel Myriad VPU [Movidius Neural Compute Stick] ```
and stay for a few seconds, until come back to the initial USB name. 

Could you help me? Please!

In case you more information, please do not hesitate to ask.

Serafadam, 2024-07-15:
Hi, could you please post more logs following the [bug template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)? You can skip running stereo_inertial_node as this one won't support OAK-D-LR

trocolit2, 2024-07-16:
**Bug Description:**

The camera OAK-D-LR presenting the issue during initialization with the camera launch from the _depthai_ros_driver_.
The driver start the device, but after the bootloader initialization, the driver can not find the device. 
The error:` [oak]: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND`
After the booting error,  the driver skips the camera connection:`  [warning] Skipping X_LINK_BOOTLOADER device with name "1.7" ()`. The camera is ignored by the driver, which can not connect the device anymore. 

**Minimal Reproducible Example**
To reproduce the error, please following the steps:

1. Install the ROS2 Humble framework
2. Install the depthai_ros_driver and its dependencies 
ros-humble-depthai                                2.26.1-1jammy.20240612.131023
ros-humble-depthai-bridge                         2.9.0-1jammy.20240612.132958
ros-humble-depthai-descriptions                   2.9.0-1jammy.20240530.095931
ros-humble-depthai-examples                       2.9.0-1jammy.20240612.134331
ros-humble-depthai-ros-driver                     2.9.0-1jammy.20240627.145557
ros-humble-depthai-ros-msgs                       2.9.0-1jammy.20240523.230218
3. Connect the camera OAK-D-LR, bootloader version 0.028, via USB 3.
```
$lusb
 Bus 001 Device 024: ID 03e7:f63c Intel Luxonis Bootloader
```
4. Set the ROS2 Humble source. `. install/setup.bash`
5. Run the depthai_ros_driver camera launch: `ros2 launch depthai_ros_driver camera.launch.py`, the output
```
$ ros2 launch depthai_ros_driver camera.launch.py 
[INFO] [launch]: All log files can be found below /home/ros2/.ros/log/2024-07-16-10-10-35-469979-thinkpad-364
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [378]
[component_container-1] [INFO] [1721110235.858445227] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[component_container-1] [INFO] [1721110235.863551060] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1721110235.863579507] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-1] [INFO] [1721110235.869022040] [oak_state_publisher]: got segment oak
[component_container-1] [INFO] [1721110235.869040687] [oak_state_publisher]: got segment oak-d-base-frame
[component_container-1] [INFO] [1721110235.869044891] [oak_state_publisher]: got segment oak_imu_frame
[component_container-1] [INFO] [1721110235.869047947] [oak_state_publisher]: got segment oak_left_camera_frame
[component_container-1] [INFO] [1721110235.869050868] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[component_container-1] [INFO] [1721110235.869053590] [oak_state_publisher]: got segment oak_model_origin
[component_container-1] [INFO] [1721110235.869056257] [oak_state_publisher]: got segment oak_rgb_camera_frame
[component_container-1] [INFO] [1721110235.869059074] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[component_container-1] [INFO] [1721110235.869061823] [oak_state_publisher]: got segment oak_right_camera_frame
[component_container-1] [INFO] [1721110235.869064539] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak_state_publisher' in container 'oak_container'
[component_container-1] [INFO] [1721110235.869978600] [oak_container]: Load Library: /home/ros2/dev/fast_usv_gimbal/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[component_container-1] [INFO] [1721110235.908728814] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1721110235.908796527] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1721110235.916522009] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [ERROR] [1721110253.565013351] [oak]: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
[component_container-1] [INFO] [1721110253.565149140] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [2024-07-16 10:10:56.664] [depthai] [warning] Skipping X_LINK_BOOTLOADER device with name "1.9" ()
[component_container-1] [ERROR] [1721110257.701019261] [oak]: No available devices
[component_container-1] [INFO] [1721110257.701173208] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [ERROR] [1721110261.800907928] [oak]: No available devices
[component_container-1] [INFO] [1721110261.801063424] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [ERROR] [1721110265.937104683] [oak]: No available devices
[component_container-1] [INFO] [1721110265.937274773] [oak]: No ip/mxid specified, connecting to the next available device.
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```
**Other Tests**

When I run the `cam_test.py` from the repository [depthai-python](https://github.com/luxonis/depthai-python)
I got the the correct behavior. Below are the my steps:

1. Clone the repo from [depthai-python](https://github.com/luxonis/depthai-python)
2. In the folder _depthai-python/utilities_, install the requirements
[... cut ...]
  Installed via rosdep, 
  ros-humble-depthai                                2.26.1-1jammy.20240612.131023
  ros-humble-depthai-bridge                         2.9.0-1jammy.20240612.132958
  ros-humble-depthai-descriptions                   2.9.0-1jammy.20240530.095931
  ros-humble-depthai-examples                       2.9.0-1jammy.20240612.134331
  ros-humble-depthai-ros-driver                     2.9.0-1jammy.20240627.145557
  ros-humble-depthai-ros-msgs                       2.9.0-1jammy.20240523.230218

Serafadam, 2024-07-17:
Hi, just to be sure, could you run 
``` bash
echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
sudo udevadm control --reload-rules && sudo udevadm trigger
```
and replug the device after that?

trocolit2, 2024-07-17:
Hi, @Serafadam 

Thanks for your reply. 
I did it already.  I think the access is correct, otherwise, I will not be able to get the information from the device, like the camera and images (posted https://github.com/luxonis/depthai-ros/issues/560#issue-2407423284).

**Adittional information**

I tested with the docker image(luxonis/depthai:latest), and I was able to connect the camera.

```
$docker run --privileged -v /dev/bus/usb:/dev/bus/usb --device-cgroup-rule='c 189:* rmw' -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix --network host --rm -i -t luxonis/depthai:latest python3 /depthai/depthai_demo.py
Using depthai module from:  /root/.local/lib/python3.9/site-packages/depthai.cpython-39-x86_64-linux-gnu.so
Depthai version installed:  2.24.0.0
/depthai/depthai_demo.py:589: DeprecationWarning: sipPyTypeDict() is deprecated, the extension module should use sipPyTypeDictRef() instead
  class WorkerSignals(QObject):
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Setting up demo...
Available devices:
[0] 19443010616C742700 [X_LINK_BOOTLOADER]
USB Connection speed: UsbSpeed.SUPER
Downloading /root/.cache/blobconverter/mobilenet-ssd_openvino_2021.4_6shave.blob...

Done
[19443010616C742700] [1.9] [21.037] [ColorCamera(6)] [warning] Unsupported resolution set for detected camera AR0234, needs 1200_P, defaulting to it
[19443010616C742700] [1.9] [21.038] [MonoCamera(0)] [error] AR0234 only supports THE_1200_P resolution, defaulting to it

```

GUI Screenshot
[image]

trocolit2, 2024-07-19:
Dear @Serafadam 

I tested the camera with the PoE and it t works with the deptai_ros_driver launch.
But when I connect the camera via USB, the issue persists. 

Did you manage to reproduce the issue?

Regards.

Serafadam, 2024-07-19:
Hi, unfortunately I haven't been able to replicate that issue, but could you test if you can connect via either MXID or USB port name directly?
This can be done by setting either following parameters:
`camera.i_mx_id: 19443010616C742700`
or 
`camera.i_usb_port_id: 1.1.4.2`

trocolit2, 2024-07-21:
Hi @Serafadam 

I tested with your suggestion, related the _mxid_ and the _usb_port_id_, and it make no difference.

To help with the issue replication,  I can generate an docker image with my base system environment, and share with you. Is it work for you?

Regards.